MarlinFirmware/Marlin #8121: GRID_MAX_POINTS_X 5 and ABL_BILINEAR_SUBDIVISION in bilinear autolevel causing nozzle to run straight down into the bed.

jadewolf05, 2017-10-27:
### Bug Report
- Description:
 When using GRID_MAX_POINTS_X 5 and ABL_BILINEAR_SUBDIVISION, when printing after autolevel (g29), the nozzel is forced straight down into the bed.  I tried on 1.1.5, 1.1.6, and the latest bugfix version.
- Expected behaviour: 
after G29, nozzel should move into position to start printing.
- Actual behaviour: after G29, nozzel is sent straight down into the bed.
- Steps to reproduce:
the issue is seen without actually starting a print as well if you just run G29.
set GRID_MAX_POINTS_X 5
enable ABL_BILINEAR_SUBDIVISION
home all with g28
issue g29
the output after g29 shows a few points on the bottom right of the subdivided grid as "=======" instead of a value.

example:
SENDING:G29
Bilinear Leveling Grid:
      0      1      2      3      4
 0 +2.055 +1.225 +0.242 -0.615 -1.475
 1 +1.982 +1.085 +0.175 -0.765 -1.612
 2 +1.992 +1.077 +0.037 -0.818 -1.805
 3 +1.957 +1.050 +0.120 -0.870 -1.862
 4 +2.162 +1.245 +0.227 -0.788 -1.730
Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6        7        8        9       10       11       12
 0 +2.05500 +1.78398 +1.51296 +1.22500 +0.90417 +0.56639 +0.24250 -0.05250 -0.33361 -0.61500 -0.90148 -1.18824 -1.47500
 1 +2.02778 +1.74657 +1.46536 +1.17343 +0.86142 +0.53869 +0.22259 -0.08041 -0.37679 -0.66861 -0.95504 -1.23692 -1.51880
 2 +2.00056 +1.70915 +1.41775 +1.12185 +0.81867 +0.51100 +0.20269 -0.10832 -0.41997 -0.72222 -1.00861 -1.28560 -1.56259
 3 +1.98250 +1.68380 +1.38509 +1.08500 +0.78370 +0.48102 +0.17500 -0.13954 -0.45741 -0.76500 -1.05435 -1.33343 -1.61250
 4 +1.98139 +1.68019 +1.37899 +1.07343 +0.75943 +0.44108 +0.12620 -0.18246 -0.48765 -0.78972 -1.08776 -1.38267 -1.67759
 5 +1.98944 +1.68864 +1.38783 +1.07657 +0.74294 +0.39886 +0.06963 -0.22869 -0.51215 -0.80361 -1.11334 -1.43107 -1.74880
 6 +1.99250 +1.69213 +1.39176 +1.07750 +0.73324 +0.37509 +0.03750 -0.25630 -0.52954 -0.81750 -1.13685 -1.47093 -1.80500
 7 +1.97528 +1.67442 +1.37356 +1.06157 +0.72639 +0.38008 +0.04778 -0.25328 -0.54034 -0.84000 -1.16533 -1.50327 -1.84120
 8 +1.95306 +1.65175 +1.35044 +1.04343 +0.72633 +0.40353 +0.08250 -0.23165 -0.54402 -0.86250 -1.19174 -1.52707 -1.86241
 9 +1.95750 +1.65583 +1.35417 +1.05000 +0.74389 +0.43528 +0.12000 -0.20546 -0.53759 -0.87000 -1.20065 -1.53157  =======
10 +2.00806 +1.70617 +1.40429 +1.09852 +0.78856 +0.47471 +0.15398 -0.17791 -0.51668 -0.85250 -1.18110 -1.50676  =======
11 +2.08528 +1.78327 +1.48127 +1.17176 +0.85085 +0.52245 +0.19074 -0.14581 -0.48565 -0.82000 -1.14407 -1.46264  =======
12 +2.16250 +1.86037 +1.55824 +1.24500 +0.91315 +0.57019 +0.22750 -0.11370 -0.45463 -0.78750 -1.10704 -1.41852  =======

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1423148/configuration.zip)

jadewolf05, 2017-10-27:
I don't konw if the "======" is causing the issue, but when I switched to GRID_MAX_POINTS_X 4, I no longer get the "======" in the grid and printing resumes per normal.  Below is the grid output with it set to GRID_MAX_POINTS_X 4.

Bilinear Leveling Grid:
      0      1      2      3
 0 +1.000 +0.420 -0.217 -0.865
 1 +0.897 +0.325 -0.315 -0.980
 2 +0.888 +0.255 -0.420 -1.140
 3 +1.040 +0.352 -0.348 -1.102
Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6        7        8        9
 0 +1.00000 +0.80880 +0.61759 +0.42000 +0.21213 -0.00213 -0.21750 -0.43259 -0.64880 -0.86500
 1 +0.96241 +0.77304 +0.58368 +0.38741 +0.18018 -0.03395 -0.24972 -0.46594 -0.68380 -0.90167
 2 +0.92481 +0.73729 +0.54976 +0.35481 +0.14823 -0.06576 -0.28194 -0.49929 -0.71881 -0.93833
 3 +0.89750 +0.70917 +0.52083 +0.32500 +0.11759 -0.09731 -0.31500 -0.53481 -0.75741 -0.98000
 4 +0.88130 +0.68770 +0.49409 +0.29361 +0.08283 -0.13484 -0.35602 -0.58077 -0.80904 -1.03731
 5 +0.87537 +0.67386 +0.47235 +0.26500 +0.04937 -0.17209 -0.39787 -0.62891 -0.86427 -1.09963
 6 +0.88750 +0.67824 +0.46898 +0.25500 +0.03481 -0.19009 -0.42000 -0.65667 -0.89833 -1.14000
 7 +0.92630 +0.71045 +0.49460 +0.27509 +0.05132 -0.17610 -0.40898 -0.64973 -0.89593 -1.14213
 8 +0.98315 +0.76087 +0.53859 +0.31380 +0.08672 -0.14287 -0.37824 -0.62241 -0.87236 -1.12231
 9 +1.04000 +0.81130 +0.58259 +0.35250 +0.12213 -0.10963 -0.34750 -0.59509 -0.84880 -1.10250

thinkyhead, 2018-02-09:
Still a problem in the current `bugfix-1.1.x` branch?

boelle, 2019-02-19:
@jadewolf05 Still a problem in the current `bugfix-2.0` branch?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

boelle, 2019-03-16:
i use bi-linear and i have never seen this issue, after G29 nozzle stays above the last probe spot

closed 2019-03-17 by p3p