MarlinFirmware/Marlin #8124: G33 Auto and others problem on a Delta

cdedwards, 2017-10-27:
Anyone else having trouble with G33 A1, and or G33 P# where # is > 1?

G33 P7 used to probe my bed from center outwards in 3 points in 7 locations. it now only probes the bed at 0,0 (bed center) 7 times.
G33 A1 I had never used, but I don't currently see how it would be useful. As it only homes 7 times

I've read through the G33.cpp and I know there have been changes made. I dont see anything glaring at the moment though.

Replacing G33.cpp with the old G33.cpp causes the expected behaviours to work. other than G33 A which is a new procedure.

READ: Tuning Ty
READ: .      c:+0.00   x:+0.00   y:+0.00   z:+0.00
READ: .               yz:+0.00  zx:+0.00  xy:+0.00
READ: Tuning Tz
READ: .      c:+0.00   x:+0.00   y:+0.00   z:+0.00
READ: .               yz:+0.00  zx:+0.00  xy:+0.00
READ: echo:busy: processing
busy: processing
SENT: M105
READ:   H_FACTOR: :+0.00   R_FACTOR: :0.00   A_FACTOR: :+0.00

ashthespy, 2017-10-28:
Can confirm that `G33 P#` with `# >1` doesn't work on the latest `bugfix-1.1.x` branch. 
You beat me to opening an issue ;-)

LVD-AC, 2017-10-28:
Yep a problem here, I will look into this next week

LVD-AC, 2017-10-28:
See #8136 and #8137 for fix

ashthespy, 2017-10-28:
That was a quick week! ;-) Thanks for the quick fix.
Will test and report tomorrow.

LVD-AC, 2017-10-28:
I had 1 hour to spare this night with wintertime ;)

closed 2017-10-29 by cdedwards