MarlinFirmware/Marlin #8125: problem in LCD 2004

ahmedpowera, 2017-10-27:
hi
i have problem in this version , when i do compile source code , the software give me good but the LCD not show any data why
i have LCD 2004
thank you

Tannoo, 2017-10-27:
Don't know what version and you didn't post your config files.

Likely the display used was not specified in `Configuration.h`.

ahmedpowera, 2017-10-27:
this code
#define REPRAP_DISCOUNT_SMART_CONTROLLER
now it show bad symbole in lcd
why

ahmedpowera, 2017-10-27:
[image]

Tannoo, 2017-10-28:
Idk. Follow the issue template.

Tannoo, 2017-10-28:
Could be a bad display, bad cables / connectors, bad breakout board, or bad controller.

If sofware... bad config is most likely.

Tannoo, 2017-10-28:
A not yet supported controller is another option.

ahmedpowera, 2017-10-28:
i was solve this problem thank you
but the heat extruder and fun extruder not work
not heating

closed 2018-02-09 by thinkyhead