MarlinFirmware/Marlin #8127: I can't extract filament until I home all.

ruggb, 2017-10-27:
When did this change? Is there a good reason for it?
I want to remove my filament but I can no longer do that without first homing all. Not a reason I can see to prevent that. Apparently you applied the homing rule to ALL steppers.

But I can heat up the hotend or hotbed to anything I want regardless of where the hotend is positioned.
If you stopped the everything until a home all was done, it would make more sense.

Tannoo, 2017-10-28:
#7814 and #7816

GMagician, 2017-10-28:
Theoretically that option must be explicitly enabled, by default it's not.
I will check the fix since it should stop only XYZ, not E.
Do you move it by LCD or not?
Because gcode is blocked by such option and, as it's clear here, maybe it should let E? axes pass through it.

ruggb, 2017-10-30:
Obviously, temp needs to be reached b4 E will move. If it should only stop XYZ, then a fix to the fix is needed.
If you are going to stop E also then you should stop bed and hotend heating along with it.

GMagician, 2017-10-30:
First of all, do you have NO_MOTION_BEFORE_HOMING enabled?
Because otherwise we have to check somewhere else.

P.S. Original intent was to stop XYZ movements (and thats done on LCD menu commands) but it block all movements from gcode side, that can be changed

GMagician, 2017-10-30:
I checked code...G0 and G1 should be allowed if no X, Y, Z axes are involved.
Can you tell me what are the steps to reproduce it with repetier?

GMagician, 2017-10-30:
Created [PR]

GMagician, 2017-10-31:
@thinkyhead change my original [PR]. Note that committed code solves your issue but the option doesn't work as expected since no XYZ is allowed to move until all are homed.

Tannoo, 2017-10-31:
XYZ are ok not to move... as long as it allows E to move.

GMagician, 2017-10-31:
Well maybe my English is not good, I'll try to say the same thing with other words
NO_MOTION_BEFORE_HOMING should prevent X or Y or Z to movements when at least one of them is not homed. When all three axes are home then they can move.
E is always allowed to move

fiveangle, 2017-11-04:
Discussion at #8189 about this

GMagician, 2017-11-04:
This can be closed, solution adopted by @thinkyhead will solve this issue and give more possibility on single axis when partial home is done

Tannoo, 2017-11-04:
So, I am wondering if `NO_MOTION_BEFORE_HOMING` should be enabled by default? It used to be.

Tannoo, 2017-11-04:
And, shouldn't it be in `Configuration_adv.h` instead of `Configuration.h`?

fiveangle, 2017-11-04:
> This can be closed, solution adopted by thinkyhead will solve this issue and give more possibility on single axis when partial home is done

@GMagician - I don't see any solution posted here or on #8189.  If there is a solution, tell @ruggb so they can try it :)

> So, I am wondering if NO_MOTION_BEFORE_HOMING should be enabled by default? It used to be.

Actually, this is new behavior as pointed out by @marcio-ao if i remember correctly.  One was always able to move up in Z for example before homing, even with safe homing, and home after stepper-deactivate, etc.  As I understand it, `NO_MOTION_BEFORE_HOMING` enforces homing prior to any user movement.

If history shows people adopting it I agree it would be a good default, but for now it's so new, I don't know what frustrations that might induce.  Many people are used to moving their Z up from the bed after turning on, so it might be best to wait and see how popular `NO_MOTION_BEFORE_HOMING` becomes.

GMagician, 2017-11-04:
@ruggb and @fiveangle:
My fix has been aborted in favour of the @thinkyhead one (#8176 and #8177)
fiveangle remember s correctly, it was a @marcio-ao request, this option is just a little bit changed by thinkyhead. Previously to move X, Y or Z you had to home all three axes, now when one is home that axis can be moved

GMagician, 2017-11-06:
Ok, let's try to clear all:
The only difference between my original idea and what is done now is:
original: you can always move E but not X or Y or Z until all are (XYZ) homed
current: you can always move E but not X if not homed or Y if not homed or Z if not homed

last is already merged by @thinkyhead

GMagician, 2017-11-06:
my original intent forgot E and so also E couldn't be moved

ruggb, 2017-11-06:
Go it - sounds good
Maybe this is a problem. 1b09733 - I guess this is 1.1.6 BF, right?
If I am manually running the extruder:
Before I home all, I must wait till the extruder stops b4 it will execute a home.
If I home first, it is the same. I can't manually move any axis until the extrude movement is finished.

I haven't tried to print, Is it different - manual vs printing execution?

GMagician, 2017-11-06:
That is a 1.1.x bugfix but I don't know what it solve
For other question I'm not able to answer.

ruggb, 2017-11-06:
I was just assuming that since 1.1.x is at 6 theBF must apply to 6

GMagician, 2017-11-06:
1.1.x branch is current "stable" version 1.1.6
1.1.x bugfix branch is for latest marlin version (1.1.x branch with bugfixes)

GMagician, 2017-11-07:
Please close report if you tested and issue has been fixed

fiveangle, 2017-11-07:
Homing status should not prevent E from movement so if that's what's occurring, it's broken and needs to be fixed.

However, make sure that nothing else is preventing E movement (min extrusion temp, etc).

Please try latest bugfix-1.1.x branch for your testing: https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x

GMagician, 2017-11-07:
old code (bad code) prevented E from moving and was the reason of this thread. That was not the original intent bug a simple "bug".
Fix on bugfix tree has fixed it, now E can always move (protected by cold extrusion)

ruggb, 2017-11-08:
OK this bug is fixed but the above issue seems to be related, but I will open another issue for it. thanks

closed 2017-11-08 by ruggb