MarlinFirmware/Marlin #8131: MKS SBase compilation

tig33r, 2017-10-28:
Anybody can tell me what am I doing wrong?

I want to compile marlin 2.0 for mks sbase (with reprap discount full graphic) using atom with platformio for windows. I cloned bugfix repo with git, set mks sbase and other things in config file and then tried compile with pio build (re-arm). Result file firmware.bin in .pioenvs has 98 Kb. After flashing board I dont have communication by usb and nothing on display.

In compilation log i have two warnings:

Warning! Library `CMSIS-LPC1768` has not been found in PlatformIO Registry.
You can ignore this message, if `CMSIS-LPC1768` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.

.piolibdeps\U8glib-ARM_ID1691\src\lib\u8g_rot.c:48:27: warning: initialization from incompatible pointer type [-Wincompatible-poin
ter-types]
u8g_dev_t u8g_dev_rot = { u8g_dev_rot_dummy_fn, NULL, NULL };
^~~~~~~~~~~~~~~~~~~~
.piolibdeps\U8glib-ARM_ID1691\src\lib\u8g_rot.c:48:27: note: (near initialization for 'u8g_dev_rot.dev_fn')

Spawn32, 2017-10-28:
Set you serial port to -1 and usb should work, there is currently a problem to communicate with the sd-card over usb, but should be fixed soon.
I dont have a reprap discount full graphic so couldn't test it, but s hopefully it should work :)

tig33r, 2017-10-28:
Thanks, serialport = -1 is working good.

Glod76, 2017-10-29:
Tig33r how would I compile and flash it to my mks sbase? I'm also using the RepRap full graphics display. did you get it working with the display? if you did how is the printer running?

tig33r, 2017-10-29:
How to compile:
1: Clone https://github.com/p3p/Marlin.git and checkout branch bugfix-2.0.x
2: Install PlatformIO IDE
3: Open project in PlatformIO and build for LPC1768 (click bottom left of screen and scroll to bottom of options)
4: Copy firmware (/.pioenvs/LPC1768/firmware.bin) to an sd card and put into Mks Sbase

Display still does not working, only communicaton by usb. If you asking about printer running... I don't know it yet. In pronterface seems like thermistor pins for bed and extruder are mismatched but i did not have much time to test it. I will check it with octoprint as host.

Additionally seems like after last commits we have error in ultralcd file.

LOG:

platformio run --environment LPC1768
|-- <CMSIS-LPC1768> v0.0.0
Compiling .pioenvs\LPC1768\src\src\lcd\ultralcd.o
Compiling .pioenvs\LPC1768\src\src\libs\stopwatch.o
Compiling .pioenvs\LPC1768\src\src\libs\vector_3.o
Compiling .pioenvs\LPC1768\src\src\module\configuration_store.o
In file included from c:\marlin\marlin\src\hal\HAL_LPC1768/HAL.h:57:0,
from c:\marlin\marlin\src\hal\hal.h:47,

from Marlin\src\lcd\../inc/MarlinConfig.h:34,
from Marlin\src\lcd\ultralcd.cpp:23:
Marlin\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h: In function 'void ST7920_SWSPI_SND_8BIT(uint8_t)':
c:\marlin\marlin\src\pins\pins_MKS_SBASE.h:158:28: error: 'P0_14' was not declared in this scope
#define LCD_PINS_D4      P0_14  // EXP1.5
^
c:\marlin\marlin\src\hal\HAL_LPC1768/fastio.h:46:99: note: in definition of macro 'LPC_GPIO'
#define LPC_GPIO(port)          ((volatile LPC_GPIO_TypeDef *)(LPC_GPIO0_BASE + LPC_PORT_OFFSET * port))
^~~~
c:\marlin\marlin\src\hal\HAL_LPC1768/fastio.h:57:40: note: in expansion of macro 'WRITE_PIN_SET'
#define WRITE_PIN(IO, v)        ((v) ? WRITE_PIN_SET(IO) : WRITE_PIN_CLR(IO))
^~~~~~~~~~~~~
c:\marlin\marlin\src\hal\HAL_LPC1768/fastio.h:70:23: note: in expansion of macro 'WRITE_PIN'
#define _WRITE(IO, v) WRITE_PIN(IO, v)
^~~~~~~~~
c:\marlin\marlin\src\hal\HAL_LPC1768/fastio.h:99:23: note: in expansion of macro '_WRITE'
#define WRITE(IO, v)  _WRITE(IO, v)
^~~~~~
Marlin\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:101:5: note: in expansion of macro 'WRITE'
WRITE(ST7920_CLK_PIN, LOW);        ST7920_DELAY_1; \
^~~~~
Marlin\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:30:25: note: in expansion of macro 'LCD_PINS_D4'
#define ST7920_CLK_PIN  LCD_PINS_D4
^~~~~~~~~~~
Marlin\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:101:11: note: in expansion of macro 'ST7920_CLK_PIN'
WRITE(ST7920_CLK_PIN, LOW);        ST7920_DELAY_1; \

^~~~~~~~~~~~~~
Marlin\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:107:5: note: in expansion of macro 'ST7920_SND_BIT'
ST7920_SND_BIT; // 1
^~~~~~~~~~~~~~
Marlin\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h: In function 'uint8_t u8g_dev_rrd_st7920_128x64_fn(u8g_t*, u8g_dev_t*, uint8_t, v
oid*)':
c:\marlin\marlin\src\pins\pins_MKS_SBASE.h:158:28: error: 'P0_14' was not declared in this scope
#define LCD_PINS_D4      P0_14  // EXP1.5
^
c:\marlin\marlin\src\hal\HAL_LPC1768/fastio.h:46:99: note: in definition of macro 'LPC_GPIO'
#define LPC_GPIO(port)          ((volatile LPC_GPIO_TypeDef *)(LPC_GPIO0_BASE + LPC_PORT_OFFSET * port))
^~~~
c:\marlin\marlin\src\hal\HAL_LPC1768/fastio.h:79:25: note: in expansion of macro 'SET_DIR_OUTPUT'
#define _SET_OUTPUT(IO) SET_DIR_OUTPUT(IO)
^~~~~~~~~~~~~~
c:\marlin\marlin\src\hal\HAL_LPC1768/fastio.h:109:29: note: in expansion of macro '_SET_OUTPUT'
#define SET_OUTPUT(IO)  do{ _SET_OUTPUT(IO); _WRITE(IO, LOW); }while(0)
^~~~~~~~~~~
c:\marlin\marlin\src\hal\HAL_LPC1768/fastio.h:120:28: note: in expansion of macro 'SET_OUTPUT'
#define OUT_WRITE(IO, v) { SET_OUTPUT(IO); WRITE(IO, v); }
^~~~~~~~~~
Marlin\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:140:9: note: in expansion of macro 'OUT_WRITE'
OUT_WRITE(ST7920_CLK_PIN, HIGH);
^~~~~~~~~
Marlin\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:30:25: note: in expansion of macro 'LCD_PINS_D4'

#define ST7920_CLK_PIN  LCD_PINS_D4
^~~~~~~~~~~
Marlin\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:140:19: note: in expansion of macro 'ST7920_CLK_PIN'
OUT_WRITE(ST7920_CLK_PIN, HIGH);
^~~~~~~~~~~~~~
c:\marlin\marlin\src\pins\pins_MKS_SBASE.h:158:28: error: 'P0_14' was not declared in this scope
#define LCD_PINS_D4      P0_14  // EXP1.5
^
c:\marlin\marlin\src\hal\HAL_LPC1768/fastio.h:46:99: note: in definition of macro 'LPC_GPIO'
#define LPC_GPIO(port)          ((volatile LPC_GPIO_TypeDef *)(LPC_GPIO0_BASE + LPC_PORT_OFFSET * port))
^~~~
c:\marlin\marlin\src\hal\HAL_LPC1768/fastio.h:57:40: note: in expansion of macro 'WRITE_PIN_SET'
#define WRITE_PIN(IO, v)        ((v) ? WRITE_PIN_SET(IO) : WRITE_PIN_CLR(IO))
^~~~~~~~~~~~~
c:\marlin\marlin\src\hal\HAL_LPC1768/fastio.h:70:23: note: in expansion of macro 'WRITE_PIN'
#define _WRITE(IO, v) WRITE_PIN(IO, v)
^~~~~~~~~
c:\marlin\marlin\src\hal\HAL_LPC1768/fastio.h:99:23: note: in expansion of macro '_WRITE'
#define WRITE(IO, v)  _WRITE(IO, v)
^~~~~~
c:\marlin\marlin\src\hal\HAL_LPC1768/fastio.h:120:44: note: in expansion of macro 'WRITE'
#define OUT_WRITE(IO, v) { SET_OUTPUT(IO); WRITE(IO, v); }
^~~~~
Marlin\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:140:9: note: in expansion of macro 'OUT_WRITE'

OUT_WRITE(ST7920_CLK_PIN, HIGH);
^~~~~~~~~
Marlin\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:30:25: note: in expansion of macro 'LCD_PINS_D4'
#define ST7920_CLK_PIN  LCD_PINS_D4
^~~~~~~~~~~
Marlin\src\lcd\dogm/ultralcd_st7920_u8glib_rrd.h:140:19: note: in expansion of macro 'ST7920_CLK_PIN'
OUT_WRITE(ST7920_CLK_PIN, HIGH);
^~~~~~~~~~~~~~
Compiling .pioenvs\LPC1768\src\src\module\delta.o
Compiling .pioenvs\LPC1768\src\src\module\endstops.o
*** [.pioenvs\LPC1768\src\src\lcd\ultralcd.o] Error 1
 [ERROR] Took 6.05 seconds

Glod76, 2017-10-29:
and to go back to smoothie just put the smoothieware firmware back on the card and reboot and its back to normal? so flashing is the same as I've been doing? and to update just compile firmware put on card and cycle power?

tig33r, 2017-10-29:
Yup

Glod76, 2017-10-29:
wow nice they made it easy like smoothieware. no chance of bricking the board doing this is there?

tig33r, 2017-10-29:
Dunno

Glod76, 2017-10-29:
do you still have sd card access on the mks through windows?

tig33r, 2017-10-29:
Yes.

Glod76, 2017-10-29:
Awesome. I'm going to be following this very closely. Smoothieware is so limited in features. Thanks for your help.

p3p, 2017-10-29:
> 1: Clone https://github.com/p3p/Marlin.git and checkout branch bugfix-2.0.x

That fork hasn't been kept upto date in a while, (since my work was merged into an official branch) you should be using https://github.com/MarlinFirmware/Marlin.git.

tig33r, 2017-10-29:
> That fork hasn't been kept upto date in a while, (since my work was merged into an official branch) you should be using https://github.com/MarlinFirmware/Marlin.git.

Yea of course! Sorry, my bad. I copied wrong old link.

Spawn32, 2017-10-29:
@tig33r  you are sure you can communicate and see whats on you sd card trough windows and usb now ?
and you have used the latest release from main github, not from @p3p github ?

Reason i ask is that i can not get my sd card top work, have to put it in my pc to re program it :/

Spawn32, 2017-10-29:
oh well, it's something in my config's, used a plain copy of github and just edit'ed the config to set the correct board and now i have sd card again, strange :/

tig33r, 2017-10-29:
@Spawn32 @Glod76 Do you have correct assigned pins for bed and heater in default mks sbase config (i have version 1.3). In my case they are swapped. Heater shows temp for bed and vice versa.

closed 2017-10-31 by tig33r

userosos, 2017-10-31:
Do How do to fix mine bug?

fiveangle, 2017-12-14:
FYI: Example MKS/Sbase/Configuration.h still fails:

```
The command "use_example_configs Mks/Sbase" exited with 0.
26.26s$ build_marlin_pio ${TRAVIS_BUILD_DIR} ${TEST_PLATFORM}
Warning! Library `CMSIS-LPC1768` has not been found in PlatformIO Registry.
You can ignore this message, if `CMSIS-LPC1768` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Marlin/src/sd/cardreader.cpp: In member function 'void CardReader::initsd()':
Marlin/src/sd/cardreader.cpp:245:29: error: 'SDSS' was not declared in this scope
if (!card.init(SPI_SPEED, SDSS)
^~~~
*** [.pioenvs/LPC1768/src/src/sd/cardreader.o] Error 1
========================== [ERROR] Took 26.08 seconds ==========================
================================== [SUMMARY] ==================================
Environment megaatmega2560          	[SKIP]
Environment megaatmega1280          	[SKIP]
Environment anet10                  	[SKIP]
Environment DUE                     	[SKIP]
Environment LPC1768                 	[ERROR]
Environment LPC1768_debug_and_upload	[SKIP]
Environment rambo                   	[SKIP]
Environment sanguino_atmega644p     	[SKIP]
Environment STM32F1                 	[SKIP]
Environment teensy20                	[SKIP]
Environment teensy35                	[SKIP]
========================== [ERROR] Took 26.08 seconds ==========================
The command "build_marlin_pio ${TRAVIS_BUILD_DIR} ${TEST_PLATFORM}" exited with 1.
0.01s$ restore_configs
```

Spawn32, 2017-12-14:
SDSS seems to be missing in the pins file, late here now, will have a look and add tomorrow :)

Spawn32, 2017-12-14:
@fiveangle 
Could you try : 

#define SDSS			   P0_6

Should be correct, if so i will commit tomorrow :)

fiveangle, 2017-12-14:
Zero-padded allowed compile: P0_0
Don't know if it will work as I don't have this board but example config compiles now.
This fix submitted as part of #8783 (or #8784)

Thanks,

-=dave

Glod76, 2017-12-16:
anyone have any luck getting this running on there mks sbase with the reprap full graphics display?

Glod76, 2017-12-18:
also is there any better way to compile the firmware platformio in atom isnt working properly?

tig33r, 2017-12-19:
Microsoft visual code

fiveangle, 2017-12-19:
also is there any better way to compile the firmware platformio in atom isnt working properly?

Sublime Text 3 + DevIoT is easiest and cross-platform:

https://github.com/MarlinFirmware/Marlin/issues/8796#issuecomment-352200857