MarlinFirmware/Marlin #8134: [FR] Improve dangerous thermal runaway implementation

dskw, 2017-10-28:
(Marlin 1.1.6)
I ran a lot of times into the thermal runaway detection due to an open window causing the components to not heat up in the expected time.
One time in the middle of a print which is the reason for my suggestion:

When a thermal runaway is detected during print the printer just freezes. This can even happen while the nozzle on top of the printed PLA / PETG and since the nozzle is still above 200° it can cause dangerous burns to the printed part, maybe even start a fire! I had burned parts after detecting the print stopped due to a runaway with the printpart being black already.

So my suggestion:
When marlin detects a thermal runaway first do at least a 10mm z-lift before halting the whole system, to avoid a possible fire!

studiodyne, 2017-10-28:
take care about therma protection , sometimes , the first layer , heat strongly the bed , and a lot of times , the crash comes from the bed and not the nozzle , because of the bed high temp heated by the high temp of the first layer 
In second comes the bad wiring and nude cable of thermistors , it's why thermocouple are strongly recommended
And for fire , 200 degres if not enough , the fire will never come from the print material but only by the components of the hotends , in a little area 
and to finish
When i want to reach 350 degres with 60w reprapme cartridges model it's very difficult , fire have not a lot of chances to start ..

But I'm ok , to moves the carriage before crashing the print , I'm really in favor of this

fiveangle, 2017-10-28:
> maybe even start a fire!

Uh, no... no it can't under any normal circumstances, unless you're using filament that catches fire during normal use temerature ranges :)

It certainly can deform the part surface due to stationary nozzle cooling down right over it, but the assumption is the part is incomplete and cannot be restarted, so is a write off anyway.  If there is thermal runaway, second-guessing the reason isn't a good idea.  Most important is to attempt to shutdown power to heaters ASAP.  Anything over that is hedging bets, perhaps wrongly.  It could be that wires are getting stretched and moving higher up makes it worse.  Likely ? Probalby not, but when is unexpected thermal runaway likely at all ?

Until there is some form of print progress storage / print resume, whatever value in "saving" the ruined print from being deformed from the hot nozzle as it cools down is not much value and no real additional safety (although I will completley admit that it could have a perceived improvement in safety 😄)

emartinez167, 2017-10-29:
If saving the failed part so you can print the remaining and then glue them together is an option for some people, I am voting +1 for this. It has happened to me many times and I wished the nozzle had not deformed the failed part where it stopped while cooling.

Regards,

Ernesto

fiveangle, 2017-10-29:
Hrmm... so would you try to measure the part with calipers and then just guess at what layer it was at and roll the dice ?  I'm just trying to envision how recovery would work.  I guess you could measure the progress with calipers, guess at the new start height, ensure several full bottom layers in your slicer, then slather it with a thick layer of glue so it will pick up the 100-350 micron space left from the partial printed layer, then go to town ?

Oh, actually... better still, you could toss a sheet of 400 grit onto the bed and sand the top of the part upside-down flat so it will mate perfectly with the new full base of the upper part half.

It's a total hack and would be ugly as sin, but if it's just a functional part, I suppose it could be useful.  You're starting to sway me 😄.

emartinez167, 2017-10-29:
Well, I have uses this method more than once. It is particularly useful with long prints! I hate to waste time and filament.

Tannoo, 2017-10-29:
>The hot extruder tip can certainly deform the part surface due to stationary nozzle cooling down right over it, but the assumption is the part is incomplete and cannot be restarted, so is a write off anyway.

Definately. I've had this exact thing happen when printing from my tablet that decided to perform an update and restart... the printer stopped moving once the buffer was empty.. all was good, it just sat there keeping temps waiting for the next command.

No fire ever happened. Bad part, yes.

Tannoo, 2017-10-29:
Well, not EXACT... but nozzle sitting on the part just melting the plastic away.

mikko70, 2017-10-30:
I hope feature when some error happen, or user cancel printing. Marlin drive head couple millimeter away from item (z-axis).

thinkyhead, 2017-10-31:
> an open window causing the components to not heat up in the expected time.

These settings should helpfully kill the printer in that case:

```cpp
  #define WATCH_TEMP_PERIOD 20                // Seconds
  #define WATCH_TEMP_INCREASE 2               // Degrees Celsius
```

> When marlin detects a thermal runaway first do at least a 10mm z-lift before halting the whole system

Not a bad idea.

MathijsG, 2017-10-31:
@fiveangle About you saying: "Until there is some form of print progress storage / print resume, whatever value in "saving" the ruined print from being deformed from the hot nozzle as it cools down is not much value and no real additional safety (although I will completley admit that it could have a perceived improvement in safety"

I can write my settings to EEPROM in Marlin. Isn't there some space left to write the stopped layer to this EEPROM after a failed print? It would be great if we would know the exact layer it stopped, so we can slice the remaining part accordingly. Any thoughts?

fiveangle, 2017-10-31:
Since thermal runaway does not remove power, saving layer progress so print can be restarted would be possible, but what is the real value ?

If your printer encounters thermal runaway, it needs to be fixed.  This isn't a routine operation and just restarting the print doesn't solve the potentially dangerous condition that caused runaway in the first place.  And if you use a heated bed, it's been shutoff so there's the issue of the print possibly coming off the bed once cooled.

There are so many things that could make this go wrong, is it really worth addressing for such a rare situation like thermal runaway ?

MathijsG, 2017-10-31:
In this case it's about thermal runaway, but sometimes you hear about randomly stopping, in either way it seems to me as a handy feature to at least know until what layer the model was printed. For example you could have printed something, which stopped mid-end printing, and wanting to resume on another printer or whatever, I often read people trying out with trial and error to find this specific layer, which can be quite hard. I do have some uncompleted models, if I knew at what layer it had stopped (irregardless of the stopping issue) I could have finished them instead of wasting precious filament.

AnHardt, 2017-10-31:
Marlin has no knowledge about layers. That's a concept only used by slicers. Marlin does only know about height.

fiveangle, 2017-10-31:
> In this case it's about thermal runaway, but sometimes you hear about randomly stopping

I do not hear about these "randomly stopping" :)

The only opportunity to restart a print is if Marlin hasn't crashed, power hasn't been cut, we know why we had to stop, and can resolve the reason (fillament runout sensor triggered is the only one I am aware of).  

For power-out there's been plenty of discussions about it if you search, the latest one here:

https://github.com/MarlinFirmware/Marlin/issues/8037

Bor thermal runaway, the "know why we had to stop" really isn't satisfied, making planning for a resume pretty weak.

If someone wants to resume one of these failed prints, it's probably just as effective (e.g. not very) to try and determine the Z reached by measuring the height of the partial print, then (after fixing your printer), crafting a slice of the print that begins at that height and cross your fingers.  No FW changes required.

dskw, 2017-11-16:
Oh man, I should have enabled the mail notifications for github. Sorry everyone that I answer so late after creating the ticket!

So the alert was not triggered by a window but a malformed fan duct that cooled down the nozzle.
So thanks @thinkyhead for the tip but luckily I won't need it :)

Anyeay yes all of you are right that it most likely will not cause a fire, though I am not certain how those special Wood-PLA types behave like under those circumstances.

In any case just looking at the discussion that has started here I think there is at least.. let's call it a "desire" for such a feature and with C-defines this could even be completely excluded from the firmware when deactivated, so I assume a minimum impact to the current firmware size. I haven't had a look at the code but I hope such an optional behavior also does not involve big rewrites of the thermal runaway handling as it is right now, but I could be wrong of course.

gaskins1992, 2019-04-28:
Has anyone tried to implement this somehow? Id really like to see this as an option if possible.

moham96, 2021-03-19:
HI, any new about this?
I have setup `POWER_LOSS_RECOVERY` and it works fine when power is lost, it would be great if thermal runaway can initiate a power loss routine(ie. save the progress to the sdcard and use `POWER_LOSS_ZRAISE` and `POWER_LOSS_RETRACT_LEN` to save the print.

I run into a thermal runway occasionally because i power the printer from a buck converter, running the power loss  recovery routing after a thermal runway would save me big time