MarlinFirmware/Marlin #8138: TMC2660 drivers?

alexxy, 2017-10-28:
Hi all!

I have a Azteeg X5 GT board with TMC2660 (Bigfoot BSD2660), and seems like Marlin has support for some types of TMC drivers. How to enable TMC2660?

Spawn32, 2017-10-29:
Did you get the TMC2660's working ?
I am designing a new controller board that uses them, so would be nice to know :)

alexxy, 2017-10-29:
Currently no. Seems like we need new lib to get them working =)

Spawn32, 2017-10-29:
To bad, have a look at the bottom of https://github.com/MarlinFirmware/Marlin/pull/8063#issuecomment-340261846 to see what i am working on :)

teemuatlut, 2017-10-29:
Yea the drivers need a new library for Marlin to be able to communicate with them. I could expand my TMC2130 or TMC2208 libraries to support them but I don't have the drivers to test. Nor really a board that would support the layout.

alexxy, 2017-10-29:
@teemuatlut if you extend you're lib i'd be glad to test =) Currently i have working Marlin on Azteeg X5 GT (standart drivers work, but i have 2660 ones) #8139

Spawn32, 2017-10-29:
And i can test when i get my board ready, probably be about 2 weeks...
Have a bootloader sorted, but will need a HAL and TMC2660 drivers :/

alexxy, 2017-10-29:
@Spawn32 stm32 based board? I actualy have an idea to create stm32 board with 6-8 steppers =)

Spawn32, 2017-10-29:
STM32F7 and 6x TMC2660 :)

alexxy, 2017-10-29:
Its a monster that can even run linux

teemuatlut, 2017-10-29:
I'm not too keen on developing for something I don't actually have. While it may take me from a couple of days to a week to get the library running, it'll be months or years of supporting it. Old and new features that need testing, bug fixes, interacting with users and helping them solve problems. I don't really want to put myself in a position where my best is only a guess.
I'll try contacting Panucatt and see if they'd be interested in providing samples so that I may bring Marlin/Arduino support for their drivers.

alexxy, 2017-10-29:
Actualy seems like [duet3d](https://www.duet3d.com/) board uses same drivers and very similar setup

Spawn32, 2017-10-29:
Yes, the Duet3d uses the TMC2660's...
[DuetWifi_Schematic_v1.02.pdf](https://github.com/MarlinFirmware/Marlin/files/1425456/DuetWifi_Schematic_v1.02.pdf)

Roxy-3D, 2017-10-29:
> I'll try contacting Panucatt and see if they'd be interested in providing samples so that I may bring Marlin/Arduino support for their drivers.

@teemuatlut   Roy has been very helpful trying to get an official 32-bit branch of Marlin going.   When we were discussing how to seed the Marlin developers with 32-bit boards, it kind of felt like his preference was to use the Azteeg X5 GT 32bit Motion Controller as the reference platform.   (It may be I'm just imagining that.  But that is what I think.)    

With the Azteeg X5 GT 32bit Motion Controller you can plug in the Bigfoot BSD2660 sub modules.

It says they are "Based on TMC2660. SPI setup and configuration including current limit and microstepping. 3A 30V. For 3D printing and CNC applications. Super quiet and smooth. 1/256 microstepping. "

Part of the reason I was pushing for the Re-ARM as the reference platform is because it would keep costs down.   But we are now at the point where just one developer having an Azteeg X5 GT can make good progress.    If you contact Roy, please include a link to this post so he knows I'm supportive.    I should get around to sending him an updated status report also.

alexxy, 2017-10-29:
Yeah, Azteeg X5 GT is interesting board. I already done pinout for marlin #8139 . Only thing missing is 2660 steppers.

teemuatlut, 2017-10-30:
Quick update. I got a response from Roy at Panucatt and he was happy to provide me with the drivers, so now I'll need to create a new driver library to support the TMC2660. It'll likely be ready (as can be) before my drivers get here but you can test them before I can.

@Roxy-3D Roy asked me about the progress with LPC HAL and I gave him a brief summary but also said that you'd be contacting soon with a more in depth update about it. Hope you don't mind =)

Roxy-3D, 2017-10-30:
> I gave him a brief summary but also said that you'd be contacting soon with a more in depth update about it. Hope you don't mind =)

Not a problem!   I'll try to do that in the next day or two...

Spawn32, 2017-10-30:
@Roxy-3D 
Nice, is just a incredible job you guy's do here. thank you :)

Spawn32, 2017-11-14:
@teemuatlut 
Any news on TMC2660 ? 😃

teemuatlut, 2017-11-15:
I think the hardware is on its' way so I could theoretically start with the library. I just have a bunch of other stuff too that I should be doing.

alexxy, 2017-12-05:
Hi!
Any news about this drivers?

alexborro, 2017-12-05:
Someone using this driver could tell if the lack of StealthChop mode makes
any difference. I mean, do you miss it?

teemuatlut, 2017-12-05:
@alexxy I've started working on a universal TMC driver library that would support 5 driver models. This is quite a big task and requires a completely new architecture inside the library to do it in a way that makes sense. That said, I did take a look at the TMC2660 communication information and it seems to differ a bit from the other drivers. Makes me think that they're a slightly older driver design.

alexxy, 2017-12-06:
@cool are there some branch to test?

Spawn32, 2017-12-06:
Good work @teemuatlut looking forward to testing it :)

teemuatlut, 2017-12-06:
@alexxy No not yet. I'll first try to get the library examples to work on TMC2660 as well as 2130, 2208, 2224 and 5130.
The final Marlin integration on a basic level is actually quite simple and shouldn't take much time.

alexxy, 2017-12-19:
@teemuatlut hows youre universal tmc lib going?

teemuatlut, 2017-12-19:
TMC5130 and TMC2130 compile the Simple example. I've been busy with other tasks for a while now but the new library is on the agenda over the christmas holidays because I can't do much CAD work on the laptop.
I've pretty much decided that TMC2660 has to be an entity on its' own because of a drastically different register structure. TMC2208 needs some fundamental changes to work properly.

alfredanil, 2017-12-19:
@teemuatlut TMC 5130 has an integrated velocity ramp generator, do you plan to implement the ramp generator function as well? apart from standard trapezoidal ramp it also has a six point ramp function, which seems pretty cool.  To test TMC2660 there is a readily available shield for Uno called TOS100 for Arduino and it can be purchased from Watterott Electronic for about 25 Euro. TMC2660 was also used in T-Bone cape for beagle bone (project available on GitHub) tried and tested lib available for all functions.

teemuatlut, 2017-12-19:
I do have full support for Trinamic's TRAMS board in the works but it's been pushed back ever since the summer. But IIRC it did do movements and homing and other basic functions.
Regarding the TMC2660, I have the bigfoot drivers from Panucatt and I've also got an Azteeg X5 GT to go along with them so testing is not a problem. I've already been able to communicate them and set the TOFF to have some movement. Mostly it's a matter of writing the methods in a more user friendly way.

teemuatlut, 2017-12-23:
[Some progress](https://github.com/teemuatlut/TMCStepper/blob/master/src/TMCStepper.h#L554).
Obviously not finished but you should be able to get the driver moving with `toff(4); cs(10);`.

alexxy, 2017-12-26:
@teemuatlut cool.  Seems basic movement works for me

alexxy, 2018-01-29:
Hi!

Seems like STM32F7 HAL added driver for TMC2660. Was this planned as generic driver for such type of TMC drivers?

jsondag, 2018-03-17:
I'm currently running an azteeg xt GT with the TMC 2660 big foot drivers. Running smoothie currently. Would like to try marlin 2. Does anyone have this working well, and is there any documentation on how to get it going. I'm struggling to find any.

thinkyhead, 2018-03-18:
@jsondag — There's no real documentation at this time. Basically, use these options in `Configuration_adv.h` (bugfix branch)…
```cpp
/**
 * Enable this section if you have TMC26X motor drivers.
 * You will need to import the TMC26XStepper library into the Arduino IDE for this
 * (https://github.com/trinamic/TMC26XStepper.git)
 */
//#define HAVE_TMC26X
#if ENABLED(HAVE_TMC26X)  // Choose your axes here. This is mandatory!
  . . .
#endif // HAVE_TMC26X
```

jsondag, 2018-03-18:
Thanks. Is there anywhere about how to compile for the azteeg? I saw something about using platformio but wasn’t quite sure how to go about that.

thinkyhead, 2018-03-18:
Which Azteeg? I'm assuming you mean the **Azteeg X5 GT**.

Easiest way to get PlatformIO:
- Download and install [Visual Studio Code](https://code.visualstudio.com/download) (free)
- From within Code, use the plugin manager to install PlatformIO IDE
[image]

Then you can use the Tasks menu to build…
[image]

[image]

…or upload…
[image]

Use the PlaformIO Home button  [image] to show the PlatformIO Home tab if you need to install additional board support.

alexxy, 2018-03-24:
Ghm.. Does 2660 drivers works with TMC26XStepper? it seems like different driver.

Spawn32, 2018-03-24:
@alexxy 
Yes TMC26XStepper works for the 2660, in my F7 hal i have a slightly modified version, mostly since i use the STM32Generic and it needs the newer version of the I2C, doesn't like the Marlin version :)

Right now i dont have a working stepper anymore on my F7 Borg board, Timers has been very modified the last month's and its broken, don even now where to begin, LOL

But, it will get sorted, and i will update the HW rev on the board ( a few blunders) an make a new batch + ad it to gihub :)

really hope @teemuatlut finds time to make a lib for the 2660 soon :)

alexxy, 2018-03-24:
Ghm... Current TMC26XStepper doesnt compile on LPC (however i'll try to fix it)

thinkyhead, 2018-03-29:
@alexxy — If you discover that changes to the library are needed, check to see if the original author will apply the changes to their library on GitHub so everyone can benefit. If they're unresponsive, then send us the updated library and we'll just host it under the `MarlinFirmware` organization or let @teemuatlut host it with his other `TMCxxxxStepper` libraries.

teemuatlut, 2018-04-01:
https://github.com/teemuatlut/Marlin/tree/bf2_tmc2660
https://github.com/teemuatlut/TMCStepper/blob/master/src/TMCStepper.h#L475

This is what I've been working on for the past week.

alexxy, 2018-04-02:
Cool =)

alexxy, 2018-04-03:
@teemuatlut with some minor changes seems works on Azteeg X5 GT. But steppers are quite hot...

Spawn32, 2018-04-03:
Looking forward to test this as soon as i can get my timers fixed again, lol :)

alexxy, 2018-04-03:
Seems like current control doesnt work,  as well as TMC_DEBUG and MONITOR_DRIVER_STATUS

teemuatlut, 2018-04-03:
I made a complete rewrite last week with no access to the hardware. I haven't tested anything yet.
The additional features ~~likely~~ definitely don't work yet.

teemuatlut, 2018-04-03:
> But steppers are quite hot

It's possible the current calculation formula needs some attention. Or the SPI command was wrong. Or...

Did you remember to change the sense resistor value?

alexxy, 2018-04-03:
No. I didnt change it.

alexxy, 2018-04-03:
Smoothie uses that value
```
motor_driver_control.beta.sense_resistor     100              # set the sense resistor used

```

alexxy, 2018-04-03:
I changed value of sense resistor to 100, stepper still quickly became very hot....

teemuatlut, 2018-04-03:
No. Use 0.1 for Panucatt BSD2660.

Driver | R_sense | Max current
-------:|:---------|:--------------
Watterott TMC2130 | 0.11 | 1.77A
Watterott TMC2208 | 0.11 | 1.77A
Panucatt BSD2660 | 0.100 | 2.19A
UltiMachine Einsy | 0.22 | 0.96A
UltiMachine Archim2 | 0.22 | 0.96A

teemuatlut, 2018-04-03:
I pushed a fix for the TMC2660 current equation.
But I'm not sure if it fixes your overheating issue. I would need to see the actual bit pattern for the register to verify it's being properly configured.

teemuatlut, 2018-04-03:
Where are you using the drivers? I'll add a `M122 R` option to see the raw patterns for the drivers

alexxy, 2018-04-03:
They are plugged into Azteeg X5 GT. How can I help you?

alexxy, 2018-04-03:
Testing you're new equation. Seems not heating much @800mA rms current

teemuatlut, 2018-04-03:
Update and upload your `bf2_tmc2660` branch and post your `M122 R` output. Remember to turn on `TMC_DEBUG`.

alexxy, 2018-04-03:
```
SENDING:M122 R
Raw registers: 
stepperX.sgcsconf=      13stepperY.sgcsconf=      13stepperZ.sgcsconf=      13stepperE0.sgcsconf=      13ok
```

alexxy, 2018-04-03:
doesnt build

```
Compiling .pioenvs/LPC1768/src/src/gcode/feature/trinamic/M906.o
Marlin/src/gcode/feature/trinamic/M122.cpp:47:40: error: macro "SERIAL_ECHO_F" requires 2 arguments, but only 1 given
SERIAL_ECHO_F(stepperX.SGCSCONF());
^
Marlin/src/gcode/feature/trinamic/M122.cpp:53:40: error: macro "SERIAL_ECHO_F" requires 2 arguments, but only 1 given
SERIAL_ECHO_F(stepperY.SGCSCONF());
^
Marlin/src/gcode/feature/trinamic/M122.cpp:59:40: error: macro "SERIAL_ECHO_F" requires 2 arguments, but only 1 given
SERIAL_ECHO_F(stepperZ.SGCSCONF());
^
Marlin/src/gcode/feature/trinamic/M122.cpp:65:41: error: macro "SERIAL_ECHO_F" requires 2 arguments, but only 1 given
SERIAL_ECHO_F(stepperE0.SGCSCONF());
^
Compiling .pioenvs/LPC1768/src/src/gcode/feature/trinamic/M911-M915.o
Compiling .pioenvs/LPC1768/src/src/gcode/gcode.o
Marlin/src/gcode/feature/trinamic/M122.cpp: In static member function 'static void GcodeSuite::M122()':
Marlin/src/gcode/feature/trinamic/M122.cpp:47:7: error: 'SERIAL_ECHO_F' was not declared in this scope
SERIAL_ECHO_F(stepperX.SGCSCONF());
^~~~~~~~~~~~~
*** [.pioenvs/LPC1768/src/src/gcode/feature/trinamic/M122.o] Error 1

```

teemuatlut, 2018-04-03:
Change them to `SERIAL_ECHO`, or comment out `SERIAL_PORT_2`.

alexxy, 2018-04-03:
Ok.
Some output (i changed current)
```
T:25> M122 R
SENDING:M122 R
Raw registers: 
stepperX.sgcsconf=0x      13=19
stepperY.sgcsconf=0x       E=14
stepperZ.sgcsconf=0x      10=16
stepperE0.sgcsconf=0x      13=19
stepperE1.sgcsconf=0x      13=19ok
T:25> M501  
SENDING:M501
V54 stored settings retrieved (673 bytes; crc 31784)
  G21    ; Units in mm
  M149 C ; Units in Celsius
  M200 D3.000000
  M200 T1 D3.000000
  M200 D0      
  M92 X80.000000 Y80.000000 Z4000.000000 E500.000000
  M203 X300.000000 Y300.000000 Z5.000000 E25.000000
  M201 X3000 Y3000 Z100 E10000
  M204 P3000.000000 R3000.000000 T3000.000000
  M205 S0.000000 T0.000000 B20000 X10.000000 Y10.000000 Z0.300000 E5.000000
  M206 X0.000000 Y0.000000 Z0.000000
  M218 T1 X0.000000 Y0.000000
  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
  M301 E0 P22.200001 I1.080000 D114.000000
  M301 E1 P22.200001 I1.080000 D114.000000
  M304 P10.000000 I0.023000 D305.399994
  M250 C140    
  M906 X800    
  M906 Y600    
  M906 Z700    
  M906 T0 E800 
  M906 T1 E800 
T:25> 
T:25> M906 Y900
SENDING:M906 Y900
T:25> M500
SENDING:M500
Settings Stored (673 bytes; crc 18577)
T:25> 
T:25> 
T:25> M122 R   
SENDING:M122 R
Raw registers: 
stepperX.sgcsconf=0x      13=19
stepperY.sgcsconf=0x      15=21
stepperZ.sgcsconf=0x      10=16
stepperE0.sgcsconf=0x      13=19
stepperE1.sgcsconf=0x      13=19ok
T:25> 
T:25> 
T:25> 
T:25> 
T:25> M501
SENDING:M501
V54 stored settings retrieved (673 bytes; crc 18577)
  G21    ; Units in mm
  M149 C ; Units in Celsius
  M200 D3.000000
  M200 T1 D3.000000
  M200 D0      
  M92 X80.000000 Y80.000000 Z4000.000000 E500.000000
  M203 X300.000000 Y300.000000 Z5.000000 E25.000000
  M201 X3000 Y3000 Z100 E10000
  M204 P3000.000000 R3000.000000 T3000.000000
  M205 S0.000000 T0.000000 B20000 X10.000000 Y10.000000 Z0.300000 E5.000000
  M206 X0.000000 Y0.000000 Z0.000000
  M218 T1 X0.000000 Y0.000000
  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
  M301 E0 P22.200001 I1.080000 D114.000000
  M301 E1 P22.200001 I1.080000 D114.000000
  M304 P10.000000 I0.023000 D305.399994
  M250 C140    
  M906 X800    
  M906 Y900    
  M906 Z700    
  M906 T0 E800 
  M906 T1 E800 

```

Interesting that setting current to 600mA gives E in M122 R

[117 more replies]

closed 2020-09-11 by sjasonsmith