MarlinFirmware/Marlin #8140: Filament change loading Bug

Nuttavoot, 2017-10-29:
### Bug Report
- Description: When using Filament change, the filament load length is shorter than the specify value. 
- Expected behaviour: Filament change load lenght 600mm
- Actual behaviour: Filament change load only about 50mm
- Steps to reproduce:
  - Preheat hot end
  - Select Filament change
  - Wait until unload done
  - Insert filament when prompt

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1424845/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/1424846/Configuration_adv.zip)

AnHardt, 2017-10-29:
`FILAMENT_CHANGE_UNLOAD_FEEDRATE` too high?

Nuttavoot, 2017-10-29:
No problem with filament unload, re-load is the problem.

AnHardt, 2017-10-29:
FILAMENT_CHANGE_LOAD_FEEDRATE too high?
Also check the other values near that. (Configuration_adv.h)

tcm0116, 2017-10-30:
The 50 mm is from your `ADVANCED_PAUSE_EXTRUDE_LENGTH` value. `M600` should first load the `FILAMENT_CHANGE_LOAD_LENGTH` at the `FILAMENT_CHANGE_LOAD_FEEDRATE`, and then the `ADVANCED_PAUSE_EXTRUDE_LENGTH` at the `ADVANCED_PAUSE_EXTRUDE_FEEDRATE`. The idea is to get the filament to the hot end quickly, but then slow down for the last bit. In your case, it appears to be skipping the first part of the load operation.

On your LCD, are you seeing the "Wait for filament load" or "Loading..." message followed by the "Wait for filament extrude" or "Extruding..." message, or just the extruding message when it's loading the filament?

Can you try manually running `M600` from the terminal with the L parameter? For example, `M600 L500` should cause it to load 500 + `ADVANCED_PAUSE_EXTRUDE_LENGTH ` mm after the unload.

Nuttavoot, 2017-10-30:
Yes, skipping the load operation that is what I am thinking. Will try to follow your advice, thanks.

Nuttavoot, 2017-10-30:
Got it sorted out, the problem are these 2 settings in "configuration.h"; 

#define PREVENT_LENGTHY_EXTRUDE
#define EXTRUDE_MAXLENGTH 1250

I have to increase "EXTRUDE_MAXLENGTH" from 1000 to 1250 for it to work properly. The magic number 1250 came from summing up FILAMENT_CHANGE_UNLOAD_LENGTH + FILAMENT_CHANGE_LOAD_LENGTH + ADVANCED_PAUSE_EXTRUDE_LENGTH
and in my case it is 600+600+50 = 1250

tcm0116, 2017-10-30:
I believe that's a bug. Each one of those movements should be considered unique so far as `EXTRUDE_MAXLENGTH` is concerned.

Nuttavoot, 2017-10-30:
OK, reopen it.

tcm0116, 2017-11-10:
@Nuttavoot Do you know if you have volumetric extrusion enabled when you were having this issue?

Nuttavoot, 2017-11-10:
@tcm0116 No, I have not done any changes to it.

tcm0116, 2017-11-10:
I did some testing last night, but I was unable to replicate this using the
latest bugfix-2.0.x. I did find some issues associated with volumetric
extrusion and flown rate, so I've submitted some fixes for those.

epoxi, 2017-11-21:
I could verify this on 1.1.6. The load filamet lenght is ommited. 

I have a bowden setup with 700mm tube (plus 50mm hotend), with the EXTRUDE_MAXLENGTH defined to 725mm.

tcm0116, 2017-11-21:
@epoxi can you try using bugfix-1.1.x to see if the problem remains?

closed 2018-02-09 by thinkyhead

Hesi-Re, 2020-01-29:
This problem continues with Marlin 2.0.2. Change filament option in LCD works well (650 unload /600 load) , but with M600 printer unloads correctly but only loads about 550 mm of filament.