MarlinFirmware/Marlin #8145: Mixing Extruder- I can't change esteps

sopjohn, 2017-10-29:
### Bug Report 

I have a small experience in 3d printing with single nozzle.
Recently i purchased a Diamond Hotend.
I am not sure if it is hardware or firmware problem but i can't change esteps only when i am using  the MIXING_EXTRUDER
I am flashing the board (with EEPROM Clear always first)with new differents values every time.
Executing M501 i saw the new values but the stepper extrude always the same X ammount.
Changing and storing esteps from LCD Screen produce the same problem,the screen has the new values but the stepper move always the same X distance.
Even with this condition i am printing (not correct) with many messages about invalid extruder
If i disable  the MIXING_EXTRUDER and choose #define EXTRUDERS 3 
i can change esteps without any problem.

Marlin 1.1.6
Rumba + from China
 DRV8825

Sorry for my English
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1425180/Configuration.zip)

------------------------------------------------------------

thinkyhead, 2017-10-31:
I'm curious, when you tell the E0 stepper to pull in 100mm of filament, does it pull in exactly 100mm of filament? This code will reset the mixing extruder and use the first stepper only:
```gcode
G28
G1 X100 Y100
G1 Z30
M163 S0 P1
M163 S1 P0
M163 S2 P0
M109 S200
G92 E0
G1 E100 F600
```
And then, if you reduce your E steps by 50% (using the LCD or with `M92 Exxx`) does the same G-code still pull in 100mm of filament?

sopjohn, 2017-10-31:
Hi,thank you for your reply and help
 I followed your instructions and everything works fine for E0 stepper
Changing your code for stepper E1 and E2 i encounter the same problem, the steppers pull a  permanent X amount of filament.
`G28
G1 X100 Y100
G1 Z30
M163 S0 P0
M163 S1 P1
M163 S2 P0
M109 S200
G92 E0
G1 E100 F300`
No matter what  values i have entered  for  E1 and E2 via flashing the board or via LCD.
Today for first time i tried to change the values for steppers using this Gcode
`M92 T0 Exxx
M500`
Everything worked well,when i tried 
`M92 T1 Exxx
 or 
M92 T2 Exxx`
i received from Repetier Host the message : 
echo:M92 Invalid extruder 1 and echo:M92 Invalid extruder 2
So i did a test,i added a very big value in E0 steps.
With
``G28
G1 X100 Y100
G1 Z30
M163 S0 P0
M163 S1 P1
M163 S2 P0
M109 S200
G92 E0
G1 E100 F300``
the E1 stepper pull  the very big ammount which is set for E0

sopjohn, 2017-11-25:
So is a bug here or wrong configuration?

thinkyhead, 2017-11-26:
Try the latest `bugfix-1.1.x` code and see if we've managed to fix this yet.

sopjohn, 2017-12-07:
The problem still remains

thinkyhead, 2017-12-07:
In that case, I'll check it out!

> `M92 T1 Exxx`

All the E steppers in a mixing extruder have to match, currently. But I will look at modifying that. I'd like it to work when `DISTINCT_E_FACTORS` is enabled.

sopjohn, 2017-12-10:
Thank you for your work in this project!
Marlin is my favorite firmware, i will wait for the changes.

sopjohn, 2018-02-01:
Reminder

sopjohn, 2018-09-22:
Which part of Marlin need changes?
I have a small knowledge of VB.
I know is a total different languages but i will like to try.

InsanityAutomation, 2018-09-22:
Do or do not, there is no try! Probably need to follow the stepper motion back from where the planner sends it commands. I haven't looked much at that segment of the code though, so maybe a chance of @thinkyhead giving better direction when he's back from world Maker faire NYC.

thinkyhead, 2018-09-22:
I'm not sure it can or should be put in at that point (`normalize_mix`) because none of the E steps-per-mm values is an authoritative one. So it's not clear how you'd scale each E stepper relative to the others. Does the 200 steps-per-mm get scaled up relative to 50, or does 50 get scaled down relative to 200?

Instead we want to find the point in the code where the E steps for the planner are figured from the E millimeters. In other words, `_populate_block` needs the modification. But currently the mixing steppers are considered as sub-units of E0 or E1, depending on which of the two mixing extruders you're addressing (up to two are currently allowed). So there would also need to be a general modification of how extruders are counted and how the per-axis settings are applied for the mixing extruder.

It's a non-trivial change, and will require some careful attention to the way `_populate_block` works and how the "longest E axis move" is reckoned, taking all the individual E stepper moves into account.

thinkyhead, 2018-09-22:
```cpp
  const float esteps_float = de * e_factor[extruder];
  const uint32_t esteps = ABS(esteps_float) + 0.5f;
  . . .
  block->steps[E_AXIS] = esteps;
  . . .
  #if ENABLED(MIXING_EXTRUDER)
    for (uint8_t i = 0; i < MIXING_STEPPERS; i++)
      block->mix_steps[i] = mixing_factor[i] * (
        #if ENABLED(LIN_ADVANCE)
          esteps
        #else
          block->step_event_count
        #endif
      );
  #endif
  . . .
```

AnHardt, 2018-09-22:
Ha. This could be the source of the  "too much extrusion when moving" issue (#11682 continued in #11310).
@e-Mole 
Please try to replace, in the block, above `block->step_event_count` with `esteps`.
Else as soon as E is not the leading axis a too high sum of steps seems to be distributed here.

e-Mole, 2018-09-23:
After changing` block->step_event_count` with `esteps`:
`//xy movement from 0,0 to 0, -100
T0 (E0 100% E1 0%)
G1 X0 Y-100 E100 F2000 (extrude 106 mm from E0 - strange!)
G1 X0 Y0 E0 F2000 (retract 106 mm from E0 - strange!)
G1 X0 Y-100 F2000 (extrude nothing) - here the behavior has changed!
G1 X0 Y0 F2000 (extrude nothing)`

AnHardt, 2018-09-23:
@e-Mole 
So we are at the right track. Let's continue in #11310.

@The_Interested
```cpp
      block->mix_steps[i] = mixing_factor[i] * esteps;
```
Let me have another try.
Where `block->mix_steps[i]` is the number of steps in that block for a MIXING_STEPPER, 
`mixing_factor[i]` is the fraction of 'esteps'  going to that MIXING_STEPPER.
The `mixing_factor[]` array must sum up to 1.0. This ensures `block->mix_steps[]` array sums up to `esteps`. 
`esteps` is calculated from  the 'E' parameter of a g-code and the 4th value in
```cpp
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4000, 500 }
```
or it's representation in RAM (EEPROM).

Adding different steps/mm values per MIXING_EXTRUDER needs at first a definition of that values. For example:
```diff
#if ENABLED(MIXING_EXTRUDER)
  #define MIXING_STEPPERS 2        // Number of steppers in your mixing extruder
+ #define MIXING_STEPPERS_STEPS_PER_UNIT   { 80, 100 }
  #define MIXING_VIRTUAL_TOOLS 16  // Use the Virtual Tool method with M163 and M164
  //#define DIRECT_MIXING_IN_G1    // Allow ABCDHI mix factors in G1 movement commands
#endif
```
These definition has to be normatized. We want to have an `stepping_factor[]` array, again, summing up to 1.0.
Than we can set up the block with (1)
```cpp
      block->mix_steps[i] = mixing_factor[i] * stepping_factor[i] * esteps;
```  
As long `MIXING_STEPPERS_STEPS_PER_UNIT` cant be live adjusted by g-code, we can save `MIXING_STEPPERS` multiplications per block by adjusting (2)
```cpp
mixing_factor[i] *= stepping_factor[i];
```
in `normalize_mix()`.

Using `DISTINCT_E_FACTORS` instead of  `MIXING_STEPPERS_STEPS_PER_UNIT` allows only (1).  `stepping_factor[i]` must bee recalculated for every change in steps/mm.

sopjohn, 2018-10-02:
I tried to understand the code but without success.
I think the solution is easy.
Example:
User's data
-------------------------------------------------------------------------------------
; purple
M163 S0 P0.5
M163 S1 P0.5
M163 S2 P0
M164 S0
-------------------------------------------------------------------------------------
#define DEFAULT_AXIS_STEPS_PER_UNIT { 80, 80, 800, 124,145,300 }
--------------------------------------------------------------------------------------

Marlin have to make this calculation

E1steps = 124 * 0.5
E2steps = 145 * 0.5
E3steps = 300 * 0

AnHardt, 2019-08-27:
Even in 2.0 Mixing requires identical steps/mm for the extruders. All steppers take the steps/mm from E0. You may get around that by hand-correcting the mix and adjusting the flow.

AnHardt, 2019-08-28:
Feel free to implement and share it.
I ran out of time and had no *different* extruders, to test, at hand last year when it worked on it.

Maybe you can get some hints in the comments of https://github.com/MarlinFirmware/Marlin/pull/12098 or one of the issues from last summer.

AnHardt, 2019-08-28:
Still looking for a multi mixing hotend less bulky than a diamond.
Any suggestions?
Someone already tested https://de.aliexpress.com/item/32880595748.html?spm=a2g0o.detail.1000060.1.115b4d24t3Q5Dh&gps-id=pcDetailBottomMoreThisSeller&scm=1007.13339.99734.0&scm_id=1007.13339.99734.0&scm-url=1007.13339.99734.0&pvid=007caac9-9a7f-4c8c-92e1-561f1b7432f6
[image]

Or any news about hotends with more than 3 color channels? CMYBW would be perfect to get the full color range.

jt-github, 2019-08-31:
At some point, I'd love a hotend that just takes white and/or clear filament and then mixes with an inkjet printer like that gorgeous (and insanely expensive) HP Jet Fusion beast, allowing just about any color, with a second filament just for support material.  But sure, CMYKWS (+support) would be awesome.

By the way, if you have two extruder motors for a mixing end (like the Geeetech A10M), even if you're lucky and the extruder motors are perfectly calibrated to put out the exact same amount of material with the same E-Steps, you could still have tested "screwing them up individually" to help out poor schlubs like me.  ;)

I might just have to try digging into the code myself.  I prefer sticking with Marlin since that's what Geeetech suggests for their printers, but I'm going to try some other firmware like Klipper and see if it already allows individual stepper calibration.

All that said, you didn't answer my question: What's the point of even having `DISTINCT_E_FACTORS` and allowing different `DEFAULT_AXIS_STEPS_PER_UNIT`?

AnHardt, 2019-08-31:
You do know what `DISTINCT_E_FACTORS` is for.
I said: It's not yet implemented for mixing_extruders.

So either wait for someone having the need, the knowledge and the equipment to test - or do it you self, or pay someone to do it for you.

tommylux, 2020-11-27:
I can't seem to set my steps using M92.. I am using A20T..
[image]

I tried setting the number of extruders but get this:
[image]

I can update the steps in the LCD and they appear when I run M503, but they do not seem to apply, the same amount is extruded.
[image]

SulzRohr, 2020-12-14:
is there already a solution for this?

tommylux, 2021-01-11:
I think this issue is still outstanding. When trying to set the steppers in the LCD, 3 different values are accepted and stored, but the actual extruded amount on all 3 extruders equals one of the stored values.

TheThomasD, 2021-04-20:
I had a similar problem with my A30T I think. I tried to set different e-steps per motor, but only the first one responded to the settings. Took me a while to figure that out. Well, luckily my extruders seem to behave very alike, so my temporary solution to this problem is to set the flow rate in my slicer (Cura) to e.g. 104% for all extruders and leave the e-steps at the default. By doing so, Cura compensates for the incorrect e-step setting. As I use Repetier Server I can still change the flow rate (even during print). Nevertheless, I'd really like to have a better solution for this.

tommylux, 2021-06-27:
Is there an update to this as setting individual esteps per extruder not working. Thanks

tommylux, 2021-11-19:
Possible workaround, change the esteps during Extruder start code:
M92 E411

tommylux, 2021-12-24:
Tried to compile new Marlin and received this error:
"MIXING_EXTRUDER can't be used with DISTINCT_E_FACTORS. But you may use SINGLENOZZLE with DISTINCT_E_FACTORS."

Noticed it related to the linked article: [FR] Rewrite Mixing Extruder to allow Distinct E Factors #22495

I assume removing DISTINCT_E_FACTORS is essentially the same situation as it was in the older version - just you have handled the error better by not allowing you to set distinct E Factors for mixing extruders.

So I guess the question is, when might we be able to see the maths and a possible release to allow both MIXING_EXTRUDER and DISTINCT_E_FACTORS together?

In the meantime, I have been able to add the following to my ToolChange in Prusaslicer, (where E429 is the average of 3 extruders) given we don't know the mix is exactly.
```
";CurrentExtruder: {next_extruder}
{if next_extruder == 0}M92 E441
{elsif next_extruder == 1}M92 E425
{elsif next_extruder == 2}M92 E421
{else}M92 E429
{endif}
M104 S{temperature[next_extruder]}"
```

tommylux, 2022-03-12:
It isn't difficult to calculate what eSteps should be for a mixing extruder. Here is the math:

[INT]$RatioE0 = Get Ratio as a % for E0
[INT]$RatioE1 = Get Ratio as a % for E1
[INT]$RatioE2 = Get Ratio as a % for E2
[INT]$StepsE0 = Get Current Steps for E0
[INT]$StepsE1 = Get Current Steps for E0
[INT]$StepsE2 = Get Current Steps for E0

(($StepsE0 / 100) * $RatioE0) = Esteps to use for E0
(($StepsE1 / 100) * $RatioE1) = Esteps to use for E1
(($StepsE2 / 100) * $RatioE2) = Esteps to use for E2

As as workaround, I have noticed that the GCode for a filament is outputted after the GCode for a tool change in PrusaSlicer. Therefore, if you have static mixing on a filament, you can override the esteps for the average of that mix. For instance:
M165 A90 B5 C5
M92 E439

(441/100)*90 = 396.9
(441/100)*5 = 21.25
(441/100)*5 = 21.25
Average Esteps to use 439.2

tommylux, 2022-06-08:
I have upgraded one extruder which uses 139 steps and the other two extuders now uses 400 steps. When I try and enable DISTINCT_E_FACTORS, I get:
"MIXING_EXTRUDER can't be used with DISTINCT_E_FACTORS

This is because validation has been put in to prevent both of these options from being selected as there is no handling of distinct E-Steps. Additionally, when I try and set the following:
DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 139 , 400 , 400 }

I get:
Static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements

My only option is to upgrade the other 2 stepper assemblies

Tom