MarlinFirmware/Marlin #8146: Custom Logo show everytime before menu when click 'Select'

fred2088, 2017-10-29:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

### Bug Report
- Description: --- Custom Logo show everytime before menu when click 'Select'
- Expected behaviour: ---  show menu when click 'Select'
- Actual behaviour: ---Custom Logo show everytime before menu when click 'Select'
rc 1.1.6

thinkyhead, 2017-10-30:
Sounds like your Marlin is crashing. Does the host output also show the reboot message every time you click the button?

Roxy-3D, 2017-10-30:
Agreed about this sounding like Marlin is crashing.    Next time you build the firmware, can you cut and paste the build statistics?   It would be good to check if it is running out of memory:

Archiving built core (caching) in: C:\Tmp\arduino_cache_687559\core\core_arduino_avr_mega_cpu_atmega2560_0c812875ac70eb4a9b385d8fb077f54c.a
Sketch uses 211638 bytes (83%) of program storage space. Maximum is 253952 bytes.
Global variables use 5593 bytes (68%) of dynamic memory, leaving 2599 bytes for local variables. Maximum is 8192 bytes.

fred2088, 2017-10-31:
no reset, just show custom logo in between Click and menu, everything work as usual ...

thinkyhead, 2017-10-31:
Do you see the same issue when you remove your SD card?

thinkyhead, 2017-10-31:
The patch linked above should take care of the issue for you!

closed 2017-10-31 by thinkyhead

fred2088, 2017-11-12:
still not yet solve from bugfix-1.1.x

it's show the custom logo every time press "select"
and the marlin logo is gone away from startup

https://www.youtube.com/watch?v=HMwmDu2FBQc

Roxy-3D, 2017-11-12:
> still not yet solve from bugfix-1.1.x

Still missing requested information.

>>Next time you build the firmware, can you cut and paste the build statistics? It would be good to check if it is running out of memory:

```cpp
Archiving built core (caching) in: C:\Tmp\arduino_cache_687559\core\core_arduino_avr_mega_cpu_atmega2560_0c812875ac70eb4a9b385d8fb077f54c.a
Sketch uses 211638 bytes (83%) of program storage space. Maximum is 253952 bytes.
Global variables use 5593 bytes (68%) of dynamic memory, leaving 2599 bytes for local variables. Maximum is 8192 bytes.
```

fred2088, 2017-11-12:
sorry for bordering ...
Sketch uses 130684 bytes (51%) of program storage space. Maximum is 253952 bytes.
Global variables use 4674 bytes (57%) of dynamic memory, leaving 3518 bytes for

fred2088, 2017-11-26:
https://www.youtube.com/watch?v=HMwmDu2FBQc
still got the same result  with bugfix 1.1.x on 26/11/2017

Regards

AnHardt, 2017-11-26:
I'd have much more trust in what you say, if the info screen in your video would be a bit more like a Marlin info screen. Is that `https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x` or from an other site or with your own additions?
Try a stock Marlin bugfix-1.1.x from our site, add a custom logo and report back. We will not support any custom mess.
I consider posting this video in at least tree threads as unfriendly.

Regards (an by)

fred2088, 2017-11-27:
sorry for making such more threads that founding the thread is merge to another thread and don't know which thread to reply or follow, 

My step to test on every problem is
1. to download the lastest bugfix-1.1.x from github, with clean install
2. apply every changes I made from step zero...
3. no any single code from other website,

Thanks for for reminder, and will try again, may be my changes make the problem.
will try to do as per your instruction again..

Thanks for for unlimited support :)

fred2088, 2017-11-27:
got the problem solved !!!
it's my fault because I copy the ultralcd_impl_DOGM.h changed from 2017-05-04 of bugfix.1.1.x to the lastest bugfix.1.1.x

sorry for cause the problem, 
I need to change again the ultralcd_impl_DOGM.h from the begining also ...

one thing found is the neopixel is init before marlin logo and custom logo which cause few seconds  delay before showing up the logo ...