MarlinFirmware/Marlin #8147: G28 X/Y raises Z instead.

DarkAlchy, 2017-10-29:
### Bug Report
- Description: --- I use Pronterface and home X, or Y, or G28 X/Y and neither axis moves but Z goes up by 10.
- Expected behaviour: --- The axis I home needs to home towards its endstop.
- Actual behaviour: --- Z raises by 10 then stops.
- Steps to reproduce:
  - Do this - G28 X, or G28 Y
  
I have my eeprom off but did go ahead and do a M502 and M501.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1425309/Marlin.zip)

Roxy-3D, 2017-10-29:
If you have a Z-Probe...   The probe needs space to deploy.   You have this set at 10mm

```cpp
#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
```

DarkAlchy, 2017-10-29:
For now the probe is off and I haven't even made it to that point yet as I am just trying to get past the G28 X, or Y, issue.  Next up will be the probe/ABL section but the lack of G28 X/Y first.

Roxy-3D, 2017-10-29:
Can your printer to a simple G28 with no extra parameters?

DarkAlchy, 2017-10-29:
I am afraid to try because the Z has the ABL probe and no endstop (I upgraded my printer so upgrading it's firmware as well and I am going back to ABL).

DarkAlchy, 2017-10-29:
Before I try did I get it right that my Z will move down to home?

Roxy-3D, 2017-10-29:
So raise the nozzle to the middle of the Z travel...  Give the command, and manually trigger the Z probe.   And be ready to press the reset button.

DarkAlchy, 2017-10-29:
No, G28 did not work.  What it did was move all AXIS by 10.  X over to the right by 10, Y to the back by 10 and Z up by 10.

Roxy-3D, 2017-10-29:
What does M119 say when the nozzle is in the center of the bed?

DarkAlchy, 2017-10-29:
Well, that was interesting I must say.  All logic is now from false (RC3 I used) to true for endstops as M119 came back with triggered and when I pushed them in by hand came back with off so I knew the logic was wrong (no idea when that changed as my RC3 still says false).  Now a G28 works.  A G28 X made Z go up by 10 then X homed and it was done.  Z is still going up so what was happening is the endstop logic was reversed in a future version (that has been taken care of now) so it didn't move my axis because it thought it was already on the endstop (perfect) and what I was seeing is that G28 X, or Y, moves the Z axis up by 10 first then actually would home that axis.  Since it thought the axis was at its end didn't do that but Z had already moved up by 10.   I wonder why Z is being moved at all?

Roxy-3D, 2017-10-29:
Z has to move because it needs to deploy the Z-probe.   It needs clearance to do that.  

You have the Z-Probe turned on:
```cpp

// Enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to use the Z_MIN_PIN for your Z_MIN_PROBE.
// The Z_MIN_PIN will then be used for both Z-homing and probing.
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
```

DarkAlchy, 2017-10-29:
My Z probe never deploys though and Z should not be touched if I am telling it to move only a single axis that is not Z.  That raising Z to deploy the arm makes no sense when I am telling it to home X or Y only.

DarkAlchy, 2017-10-29:
I had #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN on so it was moving up by the amount #define Z_CLEARANCE_DEPLOY_PROBE 10 (now 35) but it is doing that no matter what axis I am telling it when it should, in fact, only do it when I say G28 Z or just G28 and never ever should it do that with a G28 X or a G28 Y.

Roxy-3D, 2017-10-29:
It needs to raise Z if just homing X or Y because it doesn't know if the nozzle is pressed against the bed.

And if you get the servo stuff setup right...  You might see the Z-Probe deploy.     Also...   Give it another G28...   The 2nd G28 should not raise the nozzle because at that point it knows it has clearance.

DarkAlchy, 2017-10-29:
Is there a way I can have it do the old behavior of not touching Z and if it scratches my glass with a G28 X/Y then so be it?

Roxy-3D, 2017-10-29:
> Is there a way I can have it do the old behavior 

Set all the Raise and clearance numbers to 0.     But that is a bad idea.  It only takes a second to raise the nozzle and get some clearance.   That behavior is in the firmware for a very good reason.

DarkAlchy, 2017-10-29:
I just did a G28 then followed that up immediately with a G28 X and Z went back up by 35 (due to changing it to 35).

thinkyhead, 2017-10-30:
What is your `Z_HOMING_HEIGHT` setting?

In general, when issues like this come up the first thing we ask is for you to attach your Configuration files. You can ZIP them up together and drop them on your next reply. Someone will then check your settings to see what might be leading to this issue.

If the answer isn't obvious the next thing we would do is have you enable `DEBUG_LEVELING_FEATURE` and then use `M111 S32` to turn on extra debugging. That way you get lost of extra output with `G28` that we can examine for anomalies.

Otherwise all we are doing is, as you can see, taking stabls in the dark.

thinkyhead, 2017-10-30:
In general, when issues like this come up the first thing we ask is for you to attach your Configuration files for us to examine. I see you did post them above along with your entire Marlin folder (certainly not needed). I'll take a look at your configs and see if there are any obvious causes.

If the answer isn't obvious the next thing we would do is have you enable `DEBUG_LEVELING_FEATURE` and then use `M111 S32` to turn on extra debugging. That way you get lost of extra output with `G28` that we can examine for anomalies.

Without following these standard procedures, all we end up doing is taking stabs in the dark.

thinkyhead, 2017-10-30:
Ok, so after a look at your configs I see nothing unusual there. Next, please use `M503` and post the output here.

After that, please enable `DEBUG_LEVELING_FEATURE`, re-flash your firmware, issue `M111 S248` (lots of debugging) and do `G28` once or twice. Copy the output from the host console and paste it in your next reply.

Thanks! From that we will have a much better idea where the bug or issue lies.

DarkAlchy, 2017-10-30:
>>> M503
SENDING:M503
echo:  G21    ; Units in mm
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X100.00 Y100.00 Z3200.00 E605.00
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z3.50 E20.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X1500 Y1500 Z150 E3000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1500.00 R3000.00 T1500.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E10.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P10.91 I1.40 D26.38
echo:Z-Probe Offset (mm):
echo:  M851 Z-10.15

DarkAlchy, 2017-10-30:
I just did it and this time it seemed to not want to mess up.

SENDING:M111 S248
echo:DEBUG:DRYRUN,COMMUNICATION,LEVELING
>>> gcode_G28
Machine Type: Cartesian
Probe: SERVO PROBE
Probe Offset X:-25 Y:-5 Z:-10.15 (Left-Front & Below Nozzle)
  current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
> endstops.enable(true)
Raise Z (before homing) to 4.00
>>> do_blocking_move_to(0.00, 0.00, 4.00)
<<< do_blocking_move_to
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move(X, -300.00, 0.00)
  current_position=(0.00, 0.00, 4.00) : sync_plan_position
<<< do_homing_move(X)
Move Away:
>>> do_homing_move(X, 5.00, 0.00)
  current_position=(0.00, 0.00, 4.00) : sync_plan_position
<<< do_homing_move(X)
Home 2 Slow:
>>> do_homing_move(X, -10.00, 25.00)
  current_position=(0.00, 0.00, 4.00) : sync_plan_position
<<< do_homing_move(X)
>>> set_axis_is_at_home(X)
For X axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 200.00
> home_offset[X] = 0.00
  current_position=(0.00, 0.00, 4.00) :
<<< set_axis_is_at_home(X)
  current_position=(0.00, 0.00, 4.00) : sync_plan_position
  current_position=(0.00, 0.00, 4.00) : > AFTER set_axis_is_at_home
<<< homeaxis(X)
  current_position=(0.00, 0.00, 4.00) : > homeX
>>> homeaxis(Y)
Home 1 Fast:
>>> do_homing_move(Y, 259.50, 0.00)
  current_position=(0.00, 0.00, 4.00) : sync_plan_position
<<< do_homing_move(Y)
Move Away:
>>> do_homing_move(Y, -5.00, 0.00)
  current_position=(0.00, 0.00, 4.00) : sync_plan_position
<<< do_homing_move(Y)
Home 2 Slow:
>>> do_homing_move(Y, 10.00, 25.00)
  current_position=(0.00, 0.00, 4.00) : sync_plan_position
<<< do_homing_move(Y)
>>> set_axis_is_at_home(Y)
For Y axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 173.00
> home_offset[Y] = 0.00
  current_position=(0.00, 173.00, 4.00) :
<<< set_axis_is_at_home(Y)
  current_position=(0.00, 173.00, 4.00) : sync_plan_position
  current_position=(0.00, 173.00, 4.00) : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
  current_position=(0.00, 173.00, 4.00) : > homeY
Z_SAFE_HOMING >>>
  current_position=(0.00, 173.00, 4.00) : sync_plan_position
  destination=(125.00, 91.00, 4.00) : Z_SAFE_HOMING
>>> do_blocking_move_to(125.00, 91.00, 4.00)
<<< do_blocking_move_to
>>> homeaxis(Z)
  current_position=(125.00, 91.00, 4.00) : set_probe_deployed
deploy: 1
do_probe_raise(10.00)
>>> do_blocking_move_to(125.00, 91.00, 20.15)
<<< do_blocking_move_to
>>> do_blocking_move_to(125.00, 91.00, 20.15)
<<< do_blocking_move_to
Home 1 Fast:
>>> do_homing_move(Z, -300.00, 0.00)
  current_position=(125.00, 91.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, 5.00, 0.00)
  current_position=(125.00, 91.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, -10.00, 1.00)
  current_position=(125.00, 91.00, 0.00) : sync_plan_position
<<< do_homing_move(Z)
>>> set_axis_is_at_home(Z)
For Z axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 200.00
*** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
> zprobe_zoffset = -10.15
> home_offset[Z] = 0.00
  current_position=(125.00, 91.00, 10.15) :
<<< set_axis_is_at_home(Z)
  current_position=(125.00, 91.00, 10.15) : sync_plan_position
  current_position=(125.00, 91.00, 10.15) : > AFTER set_axis_is_at_home
  current_position=(125.00, 91.00, 10.15) : set_probe_deployed
deploy: 0
do_probe_raise(10.00)
>>> do_blocking_move_to(125.00, 91.00, 20.15)
<<< do_blocking_move_to
>>> do_blocking_move_to(125.00, 91.00, 20.15)
<<< do_blocking_move_to
<<< homeaxis(Z)
<<< Z_SAFE_HOMING
  current_position=(125.00, 91.00, 20.15) : > (home_all || homeZ) > final
  current_position=(125.00, 91.00, 20.15) : sync_plan_position
  current_position=(125.00, 91.00, 20.15) : clean_up_after_endstop_or_probe_move
<<< gcode_G28

thinkyhead, 2017-10-30:
Everything looks okay. Your `Z_HOMING_HEIGHT` is what causes the initial 4mm raise at the start of `G28`, and that's recommended. The `Z_CLEARANCE_DEPLOY_PROBE` mentioned at the outset is actually irrelevant unless you're homing Z with the probe (and not a Z endstop), so sorry for any confusion there.

thinkyhead, 2017-10-30:
Everything looks okay. Your `Z_HOMING_HEIGHT` is what causes the initial 4mm raise at the start of `G28`, and that's recommended. The `Z_CLEARANCE_DEPLOY_PROBE` mentioned at the outset is relevant when you're homing Z with the probe (and not a Z endstop). I see that you are homing Z with your probe, so that raise is also correct.

thinkyhead, 2017-10-30:
If after more testing you can no longer reproduce the behavior, please close this issue.

DarkAlchy, 2017-10-30:
I will do that though this makes me wonder if in my configuration.h I missed something "Raise Z (before homing) to 4.00" because it has always been 10,15, or 35 depending on what I change it to so that 4mm is oddly placed and I never had it only ever go up by 4.  What is controlling that and I wonder why I have never seen it do just 4?

Roxy-3D, 2017-10-30:
Marlin gets its values from EEPROM (if that is enabled).   Did you do a M502 & M500 to load the EEPROM with values from your Configuration.h files ?

DarkAlchy, 2017-10-30:
My eeprom is off for years now and I did the M five hundred commands I was instructed to do earlier.

thinkyhead, 2017-10-31:
> Marlin gets its values from EEPROM

While that is true of the settins you can see when you issue `M503`, the `Z_CLEARANCE_DEPLOY_PROBE` and `Z_HOMING_HEIGHT` settings always come from your configuration, do not have corresponding slots in the EEPROM, and so are not directly affected by old values in EEPROM.

closed 2017-10-31 by thinkyhead