MarlinFirmware/Marlin #8150: Error:Heating failed on Dualsetup

RollingGecko, 2017-10-29:
------------------------------------------------------------
### Bug Report
- Description: When I tried to start my first print on a Dual-X-Carriage printer I get an error.
 
- Expected behaviour: Heating up and print
- Actual behaviour:  When I tried to start my first print on a Dual-X-Carriage printer I get an error:
**READ: Error:Heating failed, system stopped! Heater_ID: 1**
**READ: Error:Printer halted. kill() called!''
- Steps to reproduce:
Simplify3d
I tried to print out of Simplify 3d. Previous tests of manual heating a extrusion over the simplify machine control panel worked without any problems.
I updated to the newest commit in branch 1.1.x.
I updated to the newest commit in branch 1.1.x.
Attached you will find a zip with config files and an error log from simplify.
[Error_Report.zip](https://github.com/MarlinFirmware/Marlin/files/1425422/Error_Report.zip)

I did not expect to get that close before the finish line an error I can not fix by my self. Please help.

thinkyhead, 2017-10-31:
Please post the G-code that you're using. We should check and see if Simplify3D is trying to do things your machine can't handle, like heating both nozzles from cold at the same time. To mitigate the issue you can try increasing the `WATCH_TEMP_PERIOD` and/or decreasing the `WATCH_TEMP_INCREASE` value.

RollingGecko, 2017-10-31:
I'm so sorry to open this issue. Your mention of the second extruder reminded me that the second extruder hotend is not yet mounted. With a resistor in place of the second thermistor solved my own stupidness. 
Tnx.

closed 2017-10-31 by RollingGecko

KARTHICKROBO, 2017-11-01:
AT PRINTING EXTRUDED 2 TEMP GO DOWN SET VALUE 210 TO 150  ??? AND IT STOP PRINTING AND HEATING AGAIN 
ANT IT RESETS