MarlinFirmware/Marlin #8153: BL-Touch problem

EnricoSx, 2017-10-29:
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1425475/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/1425476/Configuration_adv.h.txt)

i have a Anycubic Kossel Linear Plus with Trigorilla Board ( in configuration.h is board n°33 in stock firmware), Marlin 1.1.6

i connect BL-Touch on Servo0 and Z_MIN, edit configuration.h but something is wrong.

- Hotend fan try start, but can't, after power on the 3d printer.
- M401 send nozzle up to home 
- BLTOUCH ROD at startup , go up and down, red light on
- M402, M280 P0 S10,  M280 P0 S90, M280 P0 S120, M280 P0 S160,    M280 P0 S60: do nothing on BLTOUCH
- If move bltouch with hand, and send M119, report open o triggered correctly

i think i PIN problem on configuration, can you help me?

Istruction my BL-Touch version
https://drive.google.com/file/d/0B_lX6Bwizwk0RzFhTTF2dHhSems/view?%20usp%20=%20condivisione

Roxy-3D, 2017-10-29:
> If move bltouch with hand, and send M119, report open o triggered correctly

This doesn't mean anything.   In the error state, the BL-Touch reports 'Triggered'.    That is probably a good thing because that keeps the nozzle from being driven into the bed.   

The real question is why you can't deploy and retract the BL-Touch probe.   You have BLTOUCH defined, so if you have it plugged in correctly to power and the Servo-0 connector...  It should deploy and retract (and give meaningful readings).

EnricoSx, 2017-10-29:
"Do you really have a RAMPS v1.3 board?" 
i don't know, i have a Trigorilla, in stock firmware "Marlin 1.1.0-RC8" board select i n° 33
M401 send up nozzle :\

Roxy-3D, 2017-10-29:
Well....   Switch the MOTHERBOARD to 

  #define MOTHERBOARD BOARD_RAMPS_14_EFB

and see if the BL-Touch starts deploying and retracting....

EnricoSx, 2017-10-29:
FAN is ok
All commands works correctly : D
but i normal when send M401 the nozzle go to home position?

EnricoSx, 2017-11-09:
sorry i found another issue, now
m280 p0 s10
m280 p0 s90
m280 p0 s120
m280 p0 s160
works, but
if i use G29 or M401, nozzle try over the home position :|

boelle, 2019-02-19:
@EnricoSx do you still have a problem with latest bugfix 2.0?

EnricoSx, 2019-02-20:
I sell my printer

closed 2019-02-20 by EnricoSx

zed-maker, 2020-02-05:
Try board #43 instead of #33