MarlinFirmware/Marlin #8154: extruder can't work

ahmedpowera, 2017-10-29:
hi
i have extruder with fan not work
i dont now why maybe the confugiration
i have this version framware

* Marlin 3D Printer Firmware
* Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin] 

thank you

thinkyhead, 2017-10-31:
Try asking for help here: http://forums.reprap.org/list.php?415

boelle, 2019-02-19:
@ahmedpowera use latest bugfix 2.0 and Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

boelle, 2019-03-12:
@thinkyhead i think we can close this one

Phr3d13, 2019-06-10:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle