MarlinFirmware/Marlin #8161: Strange looping behaviour when auto home selected (G28), debugging advice please?

lqbombjack, 2017-10-30:
- Description / Background:  Attempting to replace the stock firmware on a 'Tronxy X5' printer.  Motherboard appears to be a clone of that used on the Anet 10, albeit with a slightly different layout of a few connectors.  

Initial flash with firmware compiled using the Anet_10 board pins configuration worked through Pronterface, but the LCD was blank.  In the end I had to trace out the pins from the LCD to get it working as non of the alternatives in the Anet_10 pins file was appropriate for this printer (LCD mechanically appears to be reprap smart controller, but with a Full graphics LCD rather than the 20x4 display).  

Now that I have the menu working I've noticed some strange behaviour.  Performing a auto home from the LCD menu, results in the printer performing the process, then repeating it endlessly (well I assume so, reset it after about a minute).  The LCD menu becomes unresponsive during this process, and the only way to stop it is to reset the printer.

Controlling the printer through Pronterface a G28 request operates normally performing the homing just a single time.  However after this the LCD menu begins alternating between showing the info screen and the top level of the menu, probably about twice a second.  This continues until the timeout when the stepper motors relax, and then the LCD returns to the info screen and remains static and the menu works as normal again.

I've used the debugging G code M11 S62 to get some details.  I keep seeing 'echo:enqueueing "G28"' during the homing process if started through the LCD Menu.  Not so if commanded through Pronterface so that explains why it's not looping, but doesnt explain the LCD menu alternating.

Can anyone suggest some additional debugging ideas?  I've tried adding bits of code to echo messages to the serial port to try and indicate where we are.  It does appear to be calling 'menu_action_gcode()' a lot when the homing process is looping, but I don't appear to be able to echo the actual gcode being passed to the function, it's almost like the value passed is null.  That said it could just be my lack of familiarity with C/C++ having not used it much since college ~20 years ago.

Is the code multitasking using timers to jump between contexts? (Sorry I haven't dug deeply enough to understand this yet)  Perhaps the menu code is stuck in a condition where it's constantly enqueing the G28 command.  Though that still doesn't explain the strange LCD alternating behaviour :-/

Any pointers on things to try would be helpful.
Thanks.

[config-files.zip](https://github.com/MarlinFirmware/Marlin/files/1426444/config-files.zip)

thinkyhead, 2017-10-31:
Can you confirm that this behavior still shows up if you use the latest `bugfix-1.1.x` branch? There's been some refinement of the G-code parser and LCD code recently that might clear it up.

studiodyne, 2017-10-31:
it sounds like a bad switch endstop
If information not good , the firmware is frozen and wait a changing state on endstop pins

studiodyne, 2017-10-31:
try an old marlin , if someone have it here
It will be quick  to test it

lqbombjack, 2017-10-31:
Thanks for your replies.

I merged bufixes-1.1.x into the master branch on Sunday, so I was reasonably up to date.

I also tried checking out the tag for v1.1.5 as the original Chinese firmware would indicate v1.1.5-dev on startup, though it's anybodies guess whether that related back to the Marlin version numbering or their own internal versioning.

If trying an older version, how far back would be worth trying?  Are there any particular gotcha's worth avoiding?

studiodyne, 2017-10-31:
It's just because i don't know if you know languages ,or if you are just a maker
and the first help , is to know if it's software or not
But i see you have already worked on this issue and more
The only help i can give you , is
SERIAL ECHOING lol , overload all routines and analyse , where the programm goes and variables

closed 2018-02-06 by thinkyhead