MarlinFirmware/Marlin #8163: Printing thinking so much

mikko70, 2017-10-30:
My own build printer keep long waitings when print this part. I try check, if problem is Marlin, Cura or my printer.
Is small part between linear rail and linear screw , no take normally long time print. 
Can anybody confirm problem or no problem. (Everybody can use it, linear rail is installed stepper bracket and x movement happen between this part and linear rail)
 i use Marlin 1.16, 0.3mm nozzle, 1.75mm PLA and cura 3.03, freecad 0.17
note temp is high (245), because my sensor has error. Please change your own.

https://filebin.ca/3fa0n6FNXaWd/OT_pidsimple.gcode

thinkyhead, 2017-10-31:
The top print speed (120mm/s) might be a little high depending on your machine. Try re-slicing with a lower print speed. I'm not sure otherwise what might be leading to "long waitings." Do you mean that there are long pauses between some moves, or do you mean that everything just moves too slowly?

mikko70, 2017-10-31:
Yes, but  it do 1mm think 10sec, do couple mm and thinking again etc. until start do longer area.
I investigated it, marlin can't handle small rounds. Mabe (cura problem, i have read it can do smaller routes than printer can do.
I fixed problem after fix item dimensions bigger holes etc.  Maybe marlin need some error handling or warning message if it can't hande things?

mikko70, 2017-11-01:
Video show problem, normally and problematic
Normally:
https://1drv.ms/v/s!AnWG6o77egjrhJgJ6Qs3B-GNkVGjUw

problematic:
https://1drv.ms/v/s!AnWG6o77egjrhJUnWUf8Tu-LE1Ys7A

thinkyhead, 2017-11-02:
> Maybe marlin need some error handling or warning message if it can't hande things?

Too many diagnostics, too much processing, even slower performance. It seems to me that Marlin is telling you there is a problem in very plain terms. 😉 

I agree with your theory. Cura is creating lines far below the threshold that Marlin will do by default. However, you can try lowering `MIN_STEPS_PER_SEGMENT` from the default 6 to 0 and see if it eliminates the pauses.

mikko70, 2017-11-02:
Sorry, i have deleted old cad design.

But if you want test i used Freecad 0.17 and when design was ready, select all from image or make union and export to .OBJ file.
And this file drag in CURA
My fault bore was 1,5 mm i think this cause problems, before this i was part of sphere what caused similar problems.

where MIN_STEPS_PER_SEGMENT can found ?

mikko70, 2017-11-11:
i think it is same problem than
https://github.com/MarlinFirmware/Marlin/issues/3981

thinkyhead, 2017-11-14:
Can you provide any concrete information that can be used to narrow down the issue? Theories and references to 5-year-old issues are of little utility.

boelle, 2019-02-19:
@mikko70 do you still have an problem with latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead