MarlinFirmware/Marlin #8170: Dual X carriage homing seems broken

oysteinkrog, 2017-10-30:
I just updated my printer to the latest bugfix 1.1,(d9cb7be) and find that when I try to home now it sends T0 towards the max endstop (instead of the min).
I then tested a few revisions earlier (77cf425) (before your DUAL_ENDSTOP merge @thinkyhead) and it is fine.

thinkyhead, 2017-10-30:
Checking the dual endstops code I don't see any changes that would affect the direction of homing fir `DUAL_X_CARRIAGE`. If you can confirm that it works okay with commit 9850ba0 but that it is then broken with commit e792c2b then I'll look even closer than closer.

thinkyhead, 2017-10-31:
Meanwhile, also post your updated configurations files so I can try compiling with them.

oysteinkrog, 2017-10-31:
My config is here:
https://github.com/oysteinkrog/Marlin/tree/config_kmax
It's at the earlier/working rebase position. I'll test those when I get home later tonight.

thinkyhead, 2017-10-31:
Thanks! I'm off to bed, but will check back in about 8-10 hours.

mdtusz, 2017-11-07:
Any progress or result on this? I've encountered the same issue (using `bugfix-2.0.x`) that seems to be introduced by this commit: 723f2a77f6de40024508ea07fd7324e0987a2aa9

oysteinkrog, 2017-11-19:
Sorry about the radio silence here, I finally found time to test.
I can indeed confirm that 9850ba0 is ok and that e792c2b is broken.
I also tested latest master (bf6a181) and it is also broken.

oysteinkrog, 2018-01-09:
Ping @thinkyhead I tested this again (97b6eab) and it's still broken :/

oysteinkrog, 2018-01-10:
Just tested latest bugfix-2.0.x as well (9951657), same problem.

mdtusz, 2018-01-10:
If you try on either side of the commit I posted above does it work? That's where it's breaking for me.

oysteinkrog, 2018-01-10:
Yes I can confirm that 723f2a7 is the exact commit that breaks things in the 2.0 branch.

oysteinkrog, 2018-01-11:
@thinkyhead This is now reported by 3 different people, we need a confirmed bug label.

bruce356, 2018-01-12:
This may be well and truly be out of order by a total novice (user only) but when will you make an end and release a usable version of Marlin. This constant adding of new features and making changes that fuck up working existing versions of the firmware seems to be endless. This is not to say that I do not appreciate the efforts by you talented programmers volunteering your precious time for free, I do, but there must be a point when adding a constant flow of new stuff has to reach an end point and you release a fully working version to that point. Then continue to develop with say version 2.1 or 2.5.
In appreciation and regards - Bruce

mdtusz, 2018-01-12:
@bruce356 you truly are out of order. You are using a piece of _free software_ that has been developed voluntarily by the community - even the core maintainers are unpaid. If you are unsatisfied with using the tip of a development branch, then use a commit that runs stable with your printer, or fork the project and maintain your own version. The refactoring and changes to code is done in order to improve the codebase, and implement features that _users just like you_ desire. Keep your comments to yourself, or create a separate issue to air your grievances. Open source maintainers have a hard enough job without users complaining. You are not entitled to anything.

oysteinkrog, 2018-01-20:
Ping @thinkyhead @Roxy-3D @Bob-the-Kuhn This issue still needs proper labels, maybe we need more people to review issues/assign tags?
I feel qualified and would not mind having the power to assign labels :)

thinkyhead, 2018-01-20:
Over 900 issues in my queue at the moment, and today is devoted to bed leveling issues. I would like to get back to this issue soon and see what the dual endstops submission broke.

Do we know anyone with a Dual X machine who is also capable enough to do testing and drill down to the specific point in the code where the malfunction is happening? None of the core developers have access to a Dual X setup at this time, or you can be sure this would have been solved by now.

mdtusz, 2018-01-20:
Thanks! I've got access to a dualX machine and am willing to help out to test and verify - the commit that I linked to above seems to be the breaking one though. If you can comment on what you _suspect_ could be the issue based on the changes in that commit, I'll poke around as well to try and see what's up.

thinkyhead, 2018-01-20:
I think it could stem from changes [at this point](https://github.com/MarlinFirmware/Marlin/commit/723f2a77f6de40024508ea07fd7324e0987a2aa9#diff-a7c736c674fcde8ddf7bf0e86181e4c0R192). Let me post a patched branch for you to test out, and hopefully it fixes the main issue.

thinkyhead, 2018-01-20:
Ok, so download and install [`bugfix-1.1.x`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x) or [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) and give it a test. Hopefully with the patch applied your machine is no longer horribly broken.

oysteinkrog, 2018-01-21:
I tested bugfix-2.0.x and your changes has changed the behavior, but it is still not correct.
When I auto home now, T1 moves to the right (correct), but it does not stop when hitting the endstop. I'm checking my HW to make sure nothing is wrong.

oysteinkrog, 2018-01-21:
M119 correctly reports both x_min and x_max as not triggered/triggered when testing the switches.

oysteinkrog, 2018-01-21:
I'm looking at the code, and I'm wondering if this is the correct output from M119:
```
SENDING:M119
Reporting endstop status
x_min: open
x_max: open
y_min: open
z_min: open
```
The reason I'm wondering is that from an initial skim of the code I would expect to see x2_max in the output list when DUAL_X_CARRIAGE is enabled..

oysteinkrog, 2018-01-21:
I tested by manually adding defines for X2_MAX_PIN and X2_MAX_ENDSTOP_INVERTING, now M119 report includes x2_max, but unfortunately homing still does not work correctly.
T1 still moves to the right, hits the x max endstop but the x2 steppes does not stop moving.

oysteinkrog, 2018-01-21:
Tested without endstop interrupts feature, same behavior.

oysteinkrog, 2018-01-21:
Ok, I think I found the bug, submitting a PR.

thinkyhead, 2018-01-22:
> Ok, I think I found the bug, it's a simple typo, submitting a PR.

Always the simple typos are the ones that get us.

oysteinkrog, 2018-01-22:
@mdtusz: maybe you can test and confirm that this is fixed now?

oysteinkrog, 2018-01-22:
> Yep just rebased it and tested it out on our machines and it's working
> with no issues now. Thanks for putting in the fix @oysteinkrog
> <https://github.com/oysteinkrog>!

Thanks to @thinkyhead too for his fix :)

closed 2018-02-02 by thinkyhead