MarlinFirmware/Marlin #8172: [FR] Include board definitions for Anet A8 in the project.

MathijsG, 2017-10-30:
The Anet A8 is supported for quite a while already, since the changes of the fork of Marlin called Skynet3D were implemented back upstream. However the board definitions for these motherboards are still gathered from the old project Skynet3D on: https://github.com/SkyNet3D/anet-board

Can we somehow integrate this into the Marlin project? Now it's already getting quite confusing in the Anet A8 printer scene, many people are still referring to old Skynet, which is a much older fork of Marlin. And now we need to teach them that they need to go To Marlin instead of Skynet, while still needing the board definitions Skynet has/made.

TL;DR: Can we include or host the Anet board definitions in the Marlin project instead of relying on Skynet? Or can we remove this specific and hard to find definition from the Anet A8 sketch and just change it to the correct Arduino version (in the end the Anet is still based on an Arduino chip), so you could even flash the sketch with a stock Arduino IDE?

Any thoughts on this?

thinkyhead, 2017-10-30:
Already done.

closed 2017-10-30 by thinkyhead

MathijsG, 2017-10-30:
Can you elaborate on this somewhat more @thinkyhead ? I'm not talking about supporting the Anet printer, but talking about the Board definition for Arduino IDE. Or what has been solved already?

MathijsG, 2017-10-31:
Yesterday I tried to upload latest Marlin to my Anet, and instead of the full Skynet3D package I tried the stock IDE. And I couldn't upload it, because I still needed the Anet board definitions, which are hosted elsewhere. My proposal was to host them here, or maybe even include them in the downloads, for every printer we support? Now people still need to google around for their obscure board definitions. Why not make it more easy for people?

Don't know why it's marked solved, because seems to me it isn't.

benlye, 2017-10-31:
There shouldn't be too much Googling required - the pins_ANET.h file already contains this information (which is still accurate):

```
/**
 * The standard Arduino IDE extension (board manager) for this board
 * is located at https://github.com/SkyNet3D/anet-board.
 *
 * Installation instructions are on that page.
 *
 * After copying the files to the appropriate location, restart Arduino and
 * you'll see "Anet V1.0" and "Anet V1.0 (Optiboot)" in the boards list.
 *
 * "Anet V1.0" uses the bootloader that was installed on the board when
 * it shipped from the factory.
 *
 * "Anet V1.0 (Optiboot)" frees up another 3K of FLASH.  You'll need to burn
 * a new bootloader to the board to be able to automatically download a
 * compiled image.
 *
 */
```
Providing Arduino board extensions seems out-of-scope for Marlin to me, but others may disagree.

MathijsG, 2017-10-31:
I think this external dependency can easily be solved by hosting these definitions ourselves, in the project. In Anet's case we can easily include them since they are also using Github. It get's quite confusing for beginner hobbyists in Anet's case, to talk them out of Skynet3D, which is an old dying fork, and give them Marlin instead, while still needing some parts from the Skynet3D project. They could easily end up flashing Skynet3D instead.

For the sake of ease of use it would be nice, in my opinion, to host these definitions here. Maybe even have an all in one IDE package which also contains all needed libraries already, like Skynet3D did. I could provide this easily, but it's up what people in this project want. Just remember, many of these Marlin supported printers attract beginner hobbyists, who often get confused in the beginning already.