MarlinFirmware/Marlin #8180: PlatformIO Compile Error TMC2130 SPI.h FORCE_INLINE does not name a type

ss89, 2017-10-31:
### Compile Error
When I compile with PlatformIO on Branch 1.1.5 I get an error:
```
Marlin/SPI.h:33:5: error: 'FORCE_INLINE' does not name a type
```
i get this error for every usage of FORCE_INLINE in the TMC2130 library THOUGH i don't have it configured on that printer at all.
[image]

Config:
[Archiv.zip](https://github.com/MarlinFirmware/Marlin/files/1429754/Archiv.zip)

platformio.ini

> [platformio]
> src_dir = Marlin
> envs_dir = .pioenvs
> lib_dir = .piolib
> libdeps_dir = .piolibdeps
> env_default = megaatmega2560
> 
> [common]
> lib_deps = U8glib@1.19.1
> 
> [env:megaatmega2560]
> platform = atmelavr
> framework = arduino
> board = megaatmega2560
> build_flags = -I $BUILDSRC_DIR
> board_f_cpu = 16000000L
> lib_deps = ${common.lib_deps}

thinkyhead, 2017-10-31:
Replace `FORCE_INLINE` in that file with `__attribute__((always_inline)) inline` and it should clear up the error.

Where did you get your `TMC2130Stepper.cpp` file from? That's not a part of the Marlin distribution.

closed 2017-10-31 by thinkyhead

ss89, 2017-10-31:
It was from PlatformIOs library manager, so https://github.com/teemuatlut/TMC2130Stepper

ss89, 2017-10-31:
Now it says MISO_PIN, MOSI_PIN and SCK_PIN not declared. Doh!