MarlinFirmware/Marlin #8184: Build failure for MKS SBASE using bugfix-2.0.x

userosos, 2017-10-31:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
------------------------------------------------------------
### Compile Error
When I compile with the config I've been using for this project since bugfix-2.0.x was first published I get an error:

pio run -e LPC1768

```
[Tue Oct 31 12:10:28 2017] Processing LPC1768 (build_unflags: -Wall; build_flags: !python Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py; src_build_flags: -Wall; lib_deps: U8glib-ARM, CMSIS-LPC1768; extra_scripts: Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py; board_f_cpu: 100000000L; platform: nxplpc; lib_extra_dirs: frameworks; lib_ldf_mode: off; src_filter: +<src/*> -<src/config>)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Warning! Library `CMSIS-LPC1768` has not been found in PlatformIO Registry.
You can ignore this message, if `CMSIS-LPC1768` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Verbose mode can be enabled via `-v, --verbose` option
Collected 2 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <U8glib-ARM> v1.18.1
|-- <CMSIS-LPC1768> v0.0.0
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_AVR/HAL_AVR.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_AVR/HAL_spi_AVR.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_AVR/MarlinSerial.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_AVR/persistent_store_impl.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_AVR/servo_AVR.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_AVR/watchdog_AVR.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/HAL_Due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/HAL_spi_Due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/HAL_timers_Due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/InterruptVectors_Due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/MarlinSerial_Due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/Servo_Due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/persistent_store_impl.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_DUE/watchdog_Due.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_timers.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HardwareSerial.o
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/LPC1768_Servo.o
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:40:0,
from Marlin/src/HAL/HAL_LPC1768/HAL.cpp:23:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:330:8: error: #error "SDSORT_CACHE_VFATS must be 2 or greater!"
#error "SDSORT_CACHE_VFATS must be 2 or greater!"
^~~~~
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:40:0,
from Marlin/src/HAL/HAL_LPC1768/HAL_timers.cpp:31:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:330:8: error: #error "SDSORT_CACHE_VFATS must be 2 or greater!"
#error "SDSORT_CACHE_VFATS must be 2 or greater!"
^~~~~
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:40:0,
from Marlin/src/HAL/HAL_LPC1768/HardwareSerial.cpp:25:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:330:8: error: #error "SDSORT_CACHE_VFATS must be 2 or greater!"
#error "SDSORT_CACHE_VFATS must be 2 or greater!"
^~~~~
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:40:0,
from Marlin/src/HAL/HAL_LPC1768/LPC1768_Servo.cpp:63:
Marlin/src/HAL/HAL_LPC1768/../../inc/SanityCheck.h:330:8: error: #error "SDSORT_CACHE_VFATS must be 2 or greater!"
#error "SDSORT_CACHE_VFATS must be 2 or greater!"
^~~~~
*** [.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL.o] Error 1
*** [.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_timers.o] Error 1
*** [.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HardwareSerial.o] Error 1
*** [.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/LPC1768_Servo.o] Error 1
```
//________________________________________________________________
[config.zip](https://github.com/MarlinFirmware/Marlin/files/1430170/config.zip)

maar1201gh, 2017-10-31:
Compile Error with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER defined

PS C:\Users\maar\Documents\GitHub\Marlin> platformio run -e LPC1768

[10/31/17 12:14:59] Processing LPC1768 (build_unflags: -Wall; build_flags: !python Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py; src_build_flags: -Wall; lib_deps: U8glib-ARM, CMSIS-LPC1768; extra_scripts: Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py; board_f_cpu: 100000000L; platform: nxplpc; lib_extra_dirs: frameworks; lib_ldf_mode: off; src_filter: +<src/*> -<src/config>)
-----------------------------------------------------------------------------------------------------------------------
Warning! Library `CMSIS-LPC1768` has not been found in PlatformIO Registry.
You can ignore this message, if `CMSIS-LPC1768` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Verbose mode can be enabled via `-v, --verbose` option
Collected 4 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <U8glib-ARM> v1.18.1
|-- <CMSIS-LPC1768> v0.0.0
Compiling .pioenvs\LPC1768\src\src\lcd\ultralcd.o
Compiling .pioenvs\LPC1768\src\src\module\motion.o
Compiling .pioenvs\LPC1768\src\src\module\planner.o
Compiling .pioenvs\LPC1768\src\src\module\planner_bezier.o
Compiling .pioenvs\LPC1768\src\src\module\printcounter.o
Compiling .pioenvs\LPC1768\src\src\module\probe.o
Compiling .pioenvs\LPC1768\src\src\module\scara.o
Compiling .pioenvs\LPC1768\src\src\module\stepper.o
In file included from Marlin\src\lcd\../inc/../pins/pins.h:306:0,
from Marlin\src\lcd\../inc/MarlinConfig.h:35,
from Marlin\src\lcd\ultralcd.cpp:23:
Marlin\src\lcd\../inc/../pins/pins_MKS_SBASE.h:158:28: error: 'P0_14' was not declared in this scope
#define LCD_PINS_D4      P0_14  // EXP1.5
^
Marlin\src\lcd\ultralcd_impl_HD44780.h:200:47: note: in expansion of macro 'LCD_PINS_D4'
LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7
^~~~~~~~~~~
*** [.pioenvs\LPC1768\src\src\lcd\ultralcd.o] Error 1
============================================== [ERROR] Took 3.17 seconds ==============================================

====================================================== [SUMMARY] ======================================================
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment teensy20                    [SKIP]
Environment rambo                       [SKIP]
Environment anet10                      [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment DUE                         [SKIP]
Environment teensy35                    [SKIP]
Environment LPC1768                     [ERROR]
Environment LPC1768_debug_and_upload    [SKIP]
============================================== [ERROR] Took 3.18 seconds ==============================================

alexxy, 2017-10-31:
Seems like some pins missing in pinmap

Spawn32, 2017-10-31:
@maar1201gh seems like we forgot to change a pin that we discovered was wrong in the pins_MKS_SBASE.h
van you cange #define LCD_PINS_D4      P0_14  // EXP1.5 to #define LCD_PINS_D4      P0_15  // EXP1.5 and see if it works ? i only have a MKS TFT display myself so haven't tested this...

tig33r, 2017-10-31:
Already mentioned this bug in #8131 in addition pins for heater and bed temp sensor are mismatched (mks sbase 1.3).

Spawn32, 2017-10-31:
tig33r, i cant seem to tag you in here...
Whya are they wrong, MKS doesn't say what thermistors port to use for heatbed and hotend any ware ?
i use TH1(P0_23) for my hotend and TH2 (P0_24) for my heatbed, had it the same way i smoothie, you can just swap the cables or re define it in pins_MKS_SBASE.h, it's just a personal choice :)

Spawn32, 2017-10-31:
@maar1201gh let me know if it works and i will update the file on github...

tig33r, 2017-10-31:
Ofc I can swap cables. In original smoothie sequence for thermistors is same as power output pins. BED - E1 - E2 - FAN. I don't think is a major bug but maybe we can consider adjusting this to original arrangement. In addition in internet we can find some pinouts pictures and they are showing BED - E1 - E2 - FAN arrangement (the most popular is tutorial from instructables http://www.instructables.com/id/Configuring-MKS-Sbase-V12-32-bit-Controller-Basics/). Maybe this could evade a lot of questions from beginner users i case of transitions from smoothie to marlin.

Spawn32, 2017-10-31:
Ok, seems more logic, will update that when i get a answer from @maar1201gh  :)

maar1201gh, 2017-10-31:
Spawn32, Yes, it's work. Thanks

userosos, 2017-10-31:
Do How do to fix mine bug?

tig33r, 2017-10-31:
> Ok, seems more logic, will update that when i get a answer from @maar1201gh :)
As I have same display and same error when try compiling with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER as @maar1201gh I can check it but late evening.

Spawn32, 2017-10-31:
@userosos 
I think one of the dev's has to look in to you fault, don think it has anything to do with pin mapping etc..

@tcm0116 any ider ?

tcm0116, 2017-10-31:
@userosos I think this was broken by #8168. I think if you enable `SDCARD_SORT_ALPHA` in Configuration_adv.h, you should be able to compile.

tcm0116, 2017-10-31:
@Spawn32 we got so caught up on P0_14 on J7, I didn't even look for other instances of P0_14 in the pins file. I've got some more changes recommended by @p3p queued up, so I'll fix the pins file for MKS SBASE as well.

Spawn32, 2017-10-31:
@tcm0116  ah, i already fixed it :)

https://github.com/MarlinFirmware/Marlin/pull/8188

tig33r, 2017-10-31:
Preliminary looks like REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER works good with P0_15 instead P0_14.

tig33r, 2017-10-31:
SD Card is not detected, both this one in sbase board and this one in display.

thinkyhead, 2017-10-31:
Will have that fixed shortly @tcm0116. Actually had those lines in but then -oops-.

closed 2017-10-31 by thinkyhead