MarlinFirmware/Marlin #8186: SHOW_CUSTOM_BOOTSCREEN hides Marlin logo

AnHardt, 2017-10-31:
The current code (and also the state before this patches #8181 #8182) is far from what we wanted when we introduced SHOW_CUSTOM_BOOTSCREEN. We wanted an **additional** logo - **not a replacement** of the Marlin logo.
In the time before the bitmap of the Marlin logo was simply replaced. Only the replaced logo was shown.
To emphasise: "This is Marlin", we wanted to provide a simple way to show a additional custom logo without not showing the Marlin logo.

If that is not wanted anymore, please remove the option for SHOW_CUSTOM_BOOTSCREEN and go back to replacing the picture. We don't want to make it too easy to hide - "This is Marlin".

thinkyhead, 2017-10-31:
Got broken? See if this helps:
```diff
  #if ENABLED(SHOW_BOOTSCREEN)
    #if ENABLED(DOGLCD)                           // On DOGM the first bootscreen is already drawn
      #if ENABLED(SHOW_CUSTOM_BOOTSCREEN)
        safe_delay(CUSTOM_BOOTSCREEN_TIMEOUT);    // Custom boot screen pause
        lcd_bootscreen();                         // Show Marlin boot screen
      #endif
      safe_delay(BOOTSCREEN_TIMEOUT);             // Pause
-   #elif ENABLED(ULTRA_LCD)
+   #endif
+   #if ENABLED(ULTRA_LCD)
      lcd_bootscreen();
      #if DISABLED(SDSUPPORT)
        lcd_init();
      #endif
    #endif
  #endif
```

Tannoo, 2017-11-01:
This is what I have with 2.0.x and it works just fine:

```
  #if ENABLED(SHOW_BOOTSCREEN)
    #if ENABLED(DOGLCD)                           // On DOGM the first bootscreen is already drawn
      #if ENABLED(SHOW_CUSTOM_BOOTSCREEN)
        safe_delay(CUSTOM_BOOTSCREEN_TIMEOUT);    // Custom boot screen pause
        lcd_bootscreen();                         // Show Marlin boot screen
      #endif
      safe_delay(BOOTSCREEN_TIMEOUT);             // Pause
    #elif ENABLED(ULTRA_LCD)
      lcd_bootscreen();
      #if DISABLED(SDSUPPORT)
        lcd_init();
      #endif
    #endif
  #endif
```

thinkyhead, 2017-11-02:
Sounds like something may be amiss with your configuration @AnHardt but I will double-check when I get to the office where we have a CR-10 with graphical display.

AnHardt, 2017-11-02:
Sorry. Was misleaded by
https://github.com/thinkyhead/Marlin/blob/a516856df953667c30742f5517f6d052b10030eb/Marlin/src/lcd/ultralcd_impl_DOGM.h#L355-L363

closed 2017-11-02 by AnHardt

fred2088, 2017-11-26:
https://www.youtube.com/watch?v=HMwmDu2FBQc
still got the same result  with bugfix 1.1.x on 26/11/2017

Regards