MarlinFirmware/Marlin #8194: Error compiling 'PROPORTIONAL_FONT_RATIO' was not declared in this scope

knotbeerdan, 2017-10-31:
### Compile Error

'PROPORTIONAL_FONT_RATIO' was not declared in this scope
```
I am attempting to update from version 1.1.0 and I have made all the changes required in my configuration.h and configuration_adv.h files. The last error that I have is 'PROPORTIONAL_FONT_RATIO' was not declared in this scope.

The error is on the serial.cpp file. I saw another issue where they added #define PROPORTIONAL_FONT_RATIO 1.0

If I do the same then I can compile without any issues, though I wanted to know if there was something else wrong.

I am using Arduino IDE Version 1.8.5(latest) and have tried 1.8.3 with the same results.
```

psavva, 2017-10-31:
To which version are you using to?
Can you please post your config files?

AnHardt, 2017-10-31:
Don't copy old config files into newer versions of Marlin!

 https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration_adv.h#L1266-L1274

thinkyhead, 2017-11-01:
Add this line to your configuration so you can get a different error on the next compile…
```cpp
#define PROPORTIONAL_FONT_RATIO 1.0
```

closed 2017-11-01 by thinkyhead