MarlinFirmware/Marlin #8195: [FR] Re-ARM UART to USB host bridge

Grogyan, 2017-10-31:
Re-ARM has a UART brekout that is otherwise unused, and with the issues relating to long, and unshielded, ribbon cables for external SD card comms.
Suggest as an alternative, that gcode loading and running be done from an thumb drive connected to a UART to USB converter connected to this port.

AnHardt, 2017-10-31:
You'll need a host in between of the both usb-devices. How about a Raspberry PI? :-(

Grogyan, 2017-10-31:
Just a USB host controller connected to the UART is all the that should be needed to communicate with a flash drive

Grogyan, 2017-11-01:
Such break out boards a somewhat common.  
Here is one [USB Host Breakout](http://www.hobbytronics.co.uk/usb-host-board-v2) that I found

boelle, 2019-02-19:
is this a marlin problem??

sjasonsmith, 2020-09-11:
This is an old request, and seems to be superseded by the popularity of serial-connected TFT displays which have their own SD and USB connections.

In addition, the referenced module seems to be no longer available.

I am going to close this as no longer relevant, but please reply with further explanation if you disagree.

closed 2020-09-11 by sjasonsmith