MarlinFirmware/Marlin #8198: Is there any chance of adding the ultratronics pro v1.0 in the future?

multi1970, 2017-11-01:
//file Configuration.h Release:origin/bugfix-2.0.x
//section machine
#define MOTHERBOARD BOARD ????

teemuatlut, 2017-11-01:
Sure could, I just did something similar with UltiMachine concerning their Archim2 board. So a developer needs to get their hands on one these boards and create the required files and definitions. I can try contacting RepRapWorld if you don't want to do it yourself.

In the mean time, you can try the [MK4duo Marlin by MagoKimbra](https://github.com/MKFirmware/MK4duo). It seems to have a board definition for UltraTronics.

multi1970, 2017-11-01:
I already tried to use the MK4duo but compiling gives me only a lot of errors.

multi1970, 2017-11-01:
And thank you for the quick respons.

thinkyhead, 2017-11-06:
I started that process, but it's way out of date. #5055

We'll have to re-create it now that 2.0 is starting to mature.

multi1970, 2017-11-08:
That would be great, the new features in 2.0 would really be a asset :-) .

ivanporro, 2017-12-12:
Hi @multi1970

Latest available MK4Duo does compile without error and it works on Ultratronics standard.
I did it myself yesterday using Arduino from reprapworld.com and MK4Duo configured on its website.
Ethernet however does not currently work on the Pro board with his firmware,nor it will in near future due to compilation issues (source: magokimbra himself on Dec, 2, 2017 ).
Hope it helps

multi1970, 2017-12-15:
Hi ivanporro, 
Thanks for letting me know, i will try it out this weekend. 
I have been working with a working repetier version (0.92.3 dev) but i'll keep it close hand whenever i have the need for it. 
I rather work with Marlin, so i will try to get it to work :-)

Thanks for the effort :-)

closed 2018-02-11 by thinkyhead

bassamalghram, 2019-08-04:
have you figure out a solution to the problem

multi1970, 2019-08-07:
Shortly after trying both marlin bugfix 2.0.bugfix i accidently mis oriented a stepper driver and blew up the board. Since a new board was to expensive i did not continued research on it. maybe in the near future i will start all over again, but it will not be on my top 3 list for a while. The last effort resulted in a working board with 12864 display and it did all the things it should until i decided to replace the stepper drivers with tmc 2208. One of them was in the wrong direction without me noticing so .... pfffffft and the board was gone... I still think the board has great potential i just can't afford one right now.