MarlinFirmware/Marlin #8201: Trouble with Marlin 1.1.6, RUMBA board and the RepRapDiscount Full Graphic Smart Controller

rjcuomo1, 2017-11-01:
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/1435144/Configuration.h.zip)
Hello, I posted this on the RepRap forum on Thingaverse:

> Hello, I recently put together a RepRapDiscount Full Graphic Smart Controller and RUMBA board combination. It's running Marlin 1.1.6. The LCD backlight comes on but the graphics do not show. I know they both work since they were displaying and functioning earlier. I recently made a few changes in the firmware but I'm not sure if it's related since it was mostly adjustment to current settings (changing step counts, fine tuning, etc.). The only new feature I turned on was the custom graphic option but after uploading and seeing the display not functioning properly, I reversed this change. Does anyone have any insight on why the graphics won't load or have ran into similar problems?
> 
> Thank you!

https://www.thingiverse.com/groups/reprap/forums/general/topic:23407

Another person who was running the same board and screen combination reported on the thread that they were having trouble with their BL Touch and LEDs but reverted back to Marlin1.1.5 and it fixed most of his (Dr_Ju_Ju) problems. 

The Thingiverse link goes to the topic and has all the steps I tried so far. I attached the configuration.h zip as well. 

Thank you for your time!

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

thinkyhead, 2017-11-01:
Do you see the same issue when using the `bugfix-1.1.x` branch? If so, we can proceed from there.

rjcuomo1, 2017-11-02:
I have not tried the bugfix-1.1.x branch. I am unsure of how to proceed with it since there isn't any instructions on how to apply it on the marlinfw.org site. I did download it and unzipped it. This is my first experience with anything this advance in Arduino.

thinkyhead, 2017-11-02:
For help learning how to build and install Marlin, search YouTube for videos on the topic. Thomas Sanladerer as a good video on the subject. If you need more help: https://forums.reprap.org/

rjcuomo1, 2017-11-03:
I have done some bench testing and already gotten positive results, though my printer isn't fully assembled so I can't put it through it's paces as of yet. Display is working, I'm getting temp readings, outputs seem to be working properly when input a meter on them. I won't be able to know if it fully works until it's assembled though. I directed the other person on thingiverse who is having issues and has the same hardware combination as I do to this thread. He might be willing and able to help test it further.

thinkyhead, 2017-11-04:
And your SD card on the LCD is working also? Maybe you can help someone with this issue… #8237

DrJuJu, 2017-11-18:
I'm not sure if it will help, but I now have a working printer based on the bug fix version (1.1.x downloaded on 4th Nov) & tailored to my D-Bot variant, with working display SD card etc...

I have attached my config files for you to reference, but note that these are for a CoreXY, D-bot, printer with a 'real' BLTouch Z axis sensor and an LED case light, E3D Titan Aero hotend etc., and I used Arduino 1.8.5 as the IDE (with all updates).

FYI, the current bug fix version on github is an update to what I've used, so not sure if it will work the same...

[Marlin-bugfix-1.1.x_DBot.zip](https://github.com/MarlinFirmware/Marlin/files/1484832/Marlin-bugfix-1.1.x_DBot.zip)

rjcuomo1, 2018-01-03:
The 1.1.6 bugfix worked. It still is working with 1.1.7 as well.

closed 2018-02-11 by thinkyhead