MarlinFirmware/Marlin #8202: [FR] ADC reference voltage

hasso77, 2017-11-01:
I have a custom circuit board for Pt100/Pt1000 sensors. How can I set ADC reference voltage to 1.1V, which is optimal for my board? (ArduinoMega2560 allow this setting)

thinkyhead, 2017-11-02:
This is unknown to me. But I think if you look at at [this link](https://www.google.com/search?q=arduino+ADC+reference+voltage) it should lead to the answer.

hasso77, 2017-11-02:
Ok.  I cannot find where ADC is initialized, BEFORE it can convert any data.

hasso77, 2017-11-02:
I think i found the answer:
To set reference voltage to 1.1V
I must set ADMUX.REFS0 to 0 and ADMUX.REFS1 to 1

I'm not familiar with programming for Arduino, but I think it's here:

File: temperature.cpp
---------------
  #define SET_ADMUX_ADCSRA(pin) ADMUX = _BV(REFS1) | (pin & 0x07); SBI(ADCSRA, ADSC)

hasso77, 2017-11-02:
I think that info will be useful:
The schematics of the board I want to use is in
AppNote from Microchip AN687 "Precision Temperature Sensing with RTD circuit"
I have a Honeywell Pt1000 sensor in ceramic cilindrcal case, which perfectly fit in place for thermictor.
I set an excitation current for sensor 0.3mA (recommended for this sensor to avoid self-heating) and gain to 1.5
Voltage at 0 degrees : 1000 Ohm * 0.3mA*1.5 = 0.45V.  ADCvalue = 0.45*1024/Vref(1.1V)=419
Voltage at 300 degrees : 2155 Ohm * 0.3mA*1.5 = 0.96975V. ADCvalue = .96975*1024/1.1=903
1 ADC step is about 0.6degrees with very good linearity

closed 2018-02-06 by thinkyhead