MarlinFirmware/Marlin #8203: First pull request - help wanted

stuartmp, 2017-11-01:
Hi All,

This is my first post.
I wanted to add another predefined material to the preheat options
I managed to modify the code and get it working.

However when I try and submit my pull request and the system runs its checks it tells me that

PREHEAT_3_TEMP_HOTEND, PREHEAT_3_TEMP_BED and PREHEAT_3_FAN_SPEED are not defined

However I can clearly see that they are

They are defined right under the preheat 1 and preheat 2 variables

...

#define PREHEAT_3_TEMP_HOTEND 250 
#define PREHEAT_3_TEMP_BED 80 
#define PREHEAT_3_FAN_SPEED 0// Value from 0 to 255

Please help me out or give me some guidance on how to solve the issue.

I did try replacing the variables in the configuration_store.cpp with static values and the code check out ok. 

So it appears to be just the fact that the system checker thinks the variables are not defined.

Not sure what to try next.

### Compile Error
When I compile with `FEATURE_X` I get an error:
Picked up JAVA_TOOL_OPTIONS: 

Loading configuration...
Xlib:  extension "RANDR" missing on display ":1.0".
Initializing packages...
Preparing boards...
Verifying...
sketch/configuration_store.cpp: In static member function 'static void MarlinSettings::reset()':
configuration_store.cpp:1254: error: 'PREHEAT_3_TEMP_HOTEND' was not declared in this scope
  lcd_preheat_hotend_temp[2] = PREHEAT_3_TEMP_HOTEND;
                               ^
configuration_store.cpp:1258: error: 'PREHEAT_3_TEMP_BED' was not declared in this scope
  lcd_preheat_bed_temp[2] = PREHEAT_3_TEMP_BED;
                            ^
configuration_store.cpp:1262: error: 'PREHEAT_3_FAN_SPEED' was not declared in this scope
  lcd_preheat_fan_speed[2] = PREHEAT_3_FAN_SPEED;
                             ^
exit status 1
'PREHEAT_3_TEMP_HOTEND' was not declared in this scope

thinkyhead, 2017-11-02:
Please give a link to your branch on Github. I'll download it, patch the error, and submit a PR to your branch.

stuartmp, 2017-11-02:
https://github.com/stuartmp/Marlin

Thank you scott.

thinkyhead, 2017-11-02:
Which branch are you working with?
[image]

thinkyhead, 2017-11-02:
Looks like you've been modifying the 1.1.x branch. Note that if you want to submit a PR in future, you'll need to apply all changes to a copy of the `bugfix-1.1.x` branch.

stuartmp, 2017-11-02:
I believe that it is 1.1x

stuartmp, 2017-11-02:
Ok. Thank you. I  did not know that as I am just starting out with github.com

thinkyhead, 2017-11-02:
It compiles fine for me. Sometimes with Arduino IDE it may help to close and re-open the project, if it doesn't notice your edits.

stuartmp, 2017-11-02:
Have a look under commit notes and you will see what I tried.

thinkyhead, 2017-11-02:
Enabling lcd to recompile and test some more…

thinkyhead, 2017-11-02:
Again, compiles ok.

stuartmp, 2017-11-02:
Yes it compiles ok in the Arduino IDE but I am unable to submit a pull request.

When I submit the pull request the system  runs a check. That fails.

If you look under the pull requests for 1.1.x you will see some about new pre heat option added.

thinkyhead, 2017-11-02:
> Yes it compiles ok in the Arduino IDE

For me too.

> I am unable to submit a pull request.

The presence of a compile error won't prevent submitting a PR. (I do it all the time.)

In any case, you won't be able to submit a PR with that branch, regardless. It will be rejected. You'll need to rebase the changes onto `bugfix-1.1.x` before submitting a PR.

thinkyhead, 2017-11-02:
The above commit references this issue, so we can see it undergoing Travis CI tests. That should reveal where any errors lie.

Side-note: There's no need to add a 3rd preheat option if you only tend to use two materials.

stuartmp, 2017-11-02:
Thank you Scott I will give that a try.

What do I do with the current PR. Do I Delete it some how?

stuartmp, 2017-11-02:
Yes its the Travis CI tests that fails it says that there is un declared variables when infact they are defined.

I use 3 material all the time. So its helpful for me to have this option

thinkyhead, 2017-11-02:
I found that it was quite easy to rebase it onto 1.1.x using `git cherry-pick` but I won't insist you go down that path. If you're curious, I first used `git rebase -i upstream/1.1.x` to squash all commits down to a single commit. Then I did `git reset --hard upstream/bugfix-1.1.x` to reset the branch to `bugfix-1.1.x`, and finally, I did a `git cherry-pick 5557f89` to take that single squashed commit and apply it onto the branch. The result, your changes applied to the latest `bugfix-1.1.x` code.

Since I've done all the work already, you can just create a new branch based on `bugfix-1.1.x` to start with, then download my branch at https://github.com/thinkyhead/Marlin/tree/stuart_fixup, copy the changed files into your branch, and create a shiny new commit.

Changed files are…

[image]

stuartmp, 2017-11-02:
Wow... Thank you Scott. I can see that your skills and knowledge of Github are right up there. I would love to lean some more but find it difficult to source good material online that doesn't just repeat the basics.

So what is the best way to copy the files back into my branch.

stuartmp, 2017-11-02:
Wow I see that you added labels to this item too. How did you do that. I tried but could not see how to.

thinkyhead, 2017-11-02:
As an admin I have label power.

The best way to copy files from my branch into yours is to use your computer's Finder or Explorer.

I assume you are using Github Desktop as your gateway to Github…?

stuartmp, 2017-11-02:
I have switch my branch on my Github desktop to bugfix-1.1.x and I have cloned your branch to my local pc under a directory marlin bug fix. What is the best way now to copy the files into my original repository? Do I just use file explorer and just copy those changed files that you show me above?

stuartmp, 2017-11-02:
Ok I will just copy then using file explorer.

thinkyhead, 2017-11-02:
If you're into puzzle-like tutorials and curious about Git, the thing that got me started in comprehending its quirks (though it is a dream compared to CVS or SVN) was this interactive tutorial: https://learngitbranching.js.org/ I had to do it many times to start getting the gist. In the end, I use a very small and specific subset of Git's full power, along with some helper scripts, and it lets me fly through my work now.

stuartmp, 2017-11-02:
Thank you for the notes and the tutorial link I will certainly looks into that.

OK I have the files as you dad it about. Do I just commit it now and Push it to the origin?

[image]

stuartmp, 2017-11-02:
Just to clear things up for me Scott. Did my attempt fail because I was trying to create a PR to the wrong branch or was there something else?

stuartmp, 2017-11-02:
I also just changed my default branch on GitHub online to bugfix-1.1.x I hope this was the right thing to do?

thinkyhead, 2018-02-06:
[image]

closed 2018-02-06 by thinkyhead