MarlinFirmware/Marlin #8207: M500 eeprom doesn't seem to work v2.0

cdedwards, 2017-11-02:
Ok so I 

M502  reload config.h values
M500  save to eeprom
M501  to look at values
everything looks good at this point.
M665 H333.89  which changes the value of the height of my delta.
M501  nothing has changed.

this was all working until a couple of days ago.

thinkyhead, 2017-11-02:
Everything seems to be in place in the code. Try again, and be careful to use `M503` — not `M501` — to view current settings. The `M501` and `M502` commands may be skipped for your test.

- `M665 H123.45`
```cpp
  inline void gcode_M665() {
    if (parser.seen('H')) {
      home_offset[Z_AXIS] = parser.value_linear_units() - DELTA_HEIGHT;
      update_software_endstops(Z_AXIS);
    }
    . . .
```

- `M503`
```cpp
      CONFIG_ECHO_START;
      SERIAL_ECHOPAIR("  M665 L", LINEAR_UNIT(delta_diagonal_rod));
      SERIAL_ECHOPAIR(" R", LINEAR_UNIT(delta_radius));
      SERIAL_ECHOPAIR(" H", LINEAR_UNIT(DELTA_HEIGHT + home_offset[Z_AXIS]));
      SERIAL_ECHOPAIR(" S", delta_segments_per_second);
      SERIAL_ECHOPAIR(" B", LINEAR_UNIT(delta_calibration_radius));
      SERIAL_ECHOPAIR(" X", LINEAR_UNIT(delta_tower_angle_trim[A_AXIS]));
      SERIAL_ECHOPAIR(" Y", LINEAR_UNIT(delta_tower_angle_trim[B_AXIS]));
      SERIAL_ECHOPAIR(" Z", LINEAR_UNIT(delta_tower_angle_trim[C_AXIS]));
      SERIAL_EOL();
```
- `M500`
```cpp
    #if ENABLED(DELTA)
      dummy = 0.0;
      EEPROM_WRITE(dummy);
      EEPROM_WRITE(dummy);
      dummy = DELTA_HEIGHT + home_offset[Z_AXIS];
      EEPROM_WRITE(dummy);
    #else
      EEPROM_WRITE(home_offset);
    #endif
```

cdedwards, 2017-11-02:
ok so I used M503

M503
M665 H333.91
M500
M503
value has changed..
M501 and I'm back to 333.89 for my height.
According to the wiki, M501 should read the values from eeprom
It's almost as if M502 and M501 are reversed or M501 is doing a factory reset similar to what M502 is doing.

karabas, 2017-11-02:
EEPROM works for me on Sbase. M500 creates file eeprom.dat
What format is it?

closed 2017-11-17 by cdedwards