MarlinFirmware/Marlin #8208: [FR] Build chamber temp monitoring and environmental control

rjcuomo1, 2017-11-02:
Hello,

I have a thermistor which monitors the temperature of the build chamber of my printer for printing ABS. Currently, I have my firmware set up to where it thinks there is a second extruder and the second extruder temp reading is my build chamber temp reading. I do plan on upgrading to multiple extruders in the future so I won't be able to use that work around since there would be an actual extruder using that pin and reading. Also, it would be nice for the graphics to say "build chamber temperature" or "BC temp" or a little cube graphic representing the reading. 

I may add heating and ventilation to the build chamber as well if it shows it can aid in printing. So being able to set that up and monitor the environmental conditions inside the build chamber would possibly  help keep complexity down and have better control by having everything unified under one firmware and board working together vs multiple independent systems trying to work together. Maybe use an environmental control setting set up a vent fan to run while routing something dirty or smelly or venting while running a home brew laser engraver. Some SLS 3D printers use heated chambers too.

Having a build chamber reading could be another fail safe in case something goes wrong and can shut down the printer and alert the user.

Thank you for your consideration!

thinkyhead, 2017-11-02:
Thank you for your request! This is on our list, just awaiting a submission from someone with the relevant hardware. Duplicate of #3982 and #973 .

closed 2017-11-02 by thinkyhead

rjcuomo1, 2017-11-02:
I may be able to help with that. I have a background in HVAC and building automation. What are you looking for?

thinkyhead, 2017-11-02:
Someone must…
- Add a new pin definition to all pins files specifying the thermistor input that will be used for the build chamber / ambient temp sensor. 
- In similar fashion, a pin should be assigned for the venting fan.
- The feature should be given a configuration switch to enable it (e.g., `AMBIENT_TEMP_CONTROL`), and it should be disabled by default.
- Finally, code must be written to monitor the chamber temperature and to activate the venting fan when the temperature goes above a set threshold, either for a period of time, or until the temperature has dropped sufficiently below the set threshold.

Easy-peasy, yes?!

rjcuomo1, 2017-11-03:
I'm not overly familiar with C+/Arduino itself but going through the
current firmware and configuration, it does appears that a lot of the
coding is done in different areas of Marlin 1.1.6. There are a lot phrases
and commands being used in the programs which overlap in the automation I
am familar with. So building the majority of the code would be simple
copying and pasting which someone with more advanced C+/Arduino skills than
I could do when instructed on what bits of program to use and how to
properly set up the sequence of operations. I can help someone that knows
how to code. A simple heated and ventlated chamber is ridiculously
rudimentary compared to the systems that controls buildings.

I know the RUMBA board does have the extra pins and outputs to control
another fan, thermistor, and heater with the aide of relays or MOSFETs.

Briofred, 2017-11-18:
Great idea for ABS printing!!!

Javier-Moreno-Blanc, 2018-01-15:
Finally the issue of adding the possibility that Marlin will include temperature control of the chamber like this?

thinkyhead, 2018-01-22:
Marlin could include it, or it could be handled by a separate Arduino. Climate control is also a good thing to research at Gardening Supply stores.

Javier-Moreno-Blanc, 2018-01-22:
Sorry, I was referring to if future versions of Marlin, they're going to have built-in this temperature control camera.
As a 3D printer manufacturer, I'm going to start designing a camera-controlled machine and I don't know whether to develop the code for this control (obviously adding a place with a MOSFET or waiting for the next versions of Marlin to include it. That's why I want to know if it's on the list.

thinkyhead, 2018-01-22:
There is a lot to developing and properly testing climate control, so I don't expect it to be done soon. Possibly some other fork has already gotten a start on it, but I'm not sure where to start looking.

mb300sd, 2018-12-19:
Is there any update on this? I've installed an enclosure heater on my printer, and will try to add support to Marlin. Thought I'd ask before spending a day or 2 digging into the temperature code.

InsanityAutomation, 2018-12-19:
There's a mostly working PR open with some cleanup needed. If the original dev doesn't finish it up soon, one of us may get to it.

chainreaction, 2019-03-05:
> There's a mostly working PR open with some cleanup needed. If the original dev doesn't finish it up soon, one of us may get to it.

Hello Insanity,
could you clarify what did u mean?

InsanityAutomation, 2019-03-05:
See #12201