MarlinFirmware/Marlin #8211: Buzzer code half written ?

karabas, 2017-11-02:
What is with buzzer.cpp status?
I tried to compile then look into code
It seems that it was half rewritten
        CRITICAL_SECTION_START;
          ::tone(BEEPER_PIN, state.tone.frequency, state.tone.duration);
        CRITICAL_SECTION_END;

---------------
Compiling .pioenvs\LPC1768\src\src\module\probe.o
Compiling .pioenvs\LPC1768\src\src\module\scara.o
Marlin\src\libs\buzzer.cpp: In static member function 'static void Buzzer::tick()':
Marlin\src\libs\buzzer.cpp:63:11: error: '::tone' has not been declared
::tone(BEEPER_PIN, state.tone.frequency, state.tone.duration);
^~
*** [.pioenvs\LPC1768\src\src\libs\buzzer.o] Error 1

tcm0116, 2017-11-05:
The tone function from Arduino hasn't been implemented in the LPC1768 HAL, which is why it doesn't compile. I briefly looked into it, and there's a reference implementation available, but I don't have a speaker to test with. 

@thinkyhead @p3p we should probably add some sanity checks to error out on Marlin features that are not yet supported in the LPC1768 HAL.

thinkyhead, 2018-02-09:
This is one that should definitely be added to the HAL, if not present. I leave it to those focused on the 32-bit side to apply needed patches for unimplemented features. I'll get more on the 32-bit team once I have the issue queue down to a manageable number… under 100. Currently have it whittled down to around 500.

vivian-ng, 2018-10-10:
Hi, what is the status of this issue? Is it a matter of writing a tone() implementation for LPC1768? If so, there seems to be a similar library available, maybe can use #ifdef statements to wrap references to tone() and use this library function for LPC1768.
[os.mbed.com: Piezo Buzzer](https://os.mbed.com/components/Piezo-Buzzer/)
The page has links to an example HelloWorld and the library source code.

thinkyhead, 2018-10-10:
I implemented `tone()` for the DUE HAL. It shouldn't be very hard to port it over to the LPC1768 HAL.

gloomyandy, 2018-10-10:
Out of interest what LPC1768/display setup has a speaker?

vivian-ng, 2018-10-10:
I am using a MKS Sbase with a Reprap full graphics smart controller (aka LCD12864).

AnHardt, 2018-10-10:
> Reprap full graphics smart controller

Where on the sound device usually has not to be pulsed, because it has its own resonator. Just set the input high and it will make a fixed frequency tone. By our definition it has a 'buzzer', not a 'speaker'.

gloomyandy, 2018-10-10:
Yep that's the same combination I have. No need for any fancy speaker code just leave SPEAKER undefined and it will buzz away happily.

thinkyhead, 2018-10-10:
Can you play variable frequency tones without `SPEAKER` on that device?

gloomyandy, 2018-10-10:
I think it is just a simple buzzer that plays a single tone when you enable it. It's the same display and sounder used on lots of systems I don't think they normally use it as anything other than a buzzer. I suppose you might be able to modulate it in some way, but not sure it would be worth it.

p3p, 2018-10-20:
@karabas The feature is now implemented in lpc176x framework.

closed 2018-10-26 by thinkyhead