MarlinFirmware/Marlin #8212: How to configure Software serial correctly?

karabas, 2017-11-02:
I need to control external mp3 player via serial 9600 Tx only.
On marlin 1xxx and mega2560 I use SoftwareSerial lib
Now I jump to Marlin2 and Sbase. As I have told there are no spare hardware serials on that chip. 
If it is correct I want set configure seftware serial for my usage, 
( I already done  it on smoothieware)

tcm0116, 2017-11-02:
If you're using the emulated serial port over the USB for normal host comms, then UART0 is available for use on the MKS SBASE. The recently updated HardwareSerial class in the LPC1768 HAL can be used to communicate over the UART.

karabas, 2017-11-03:
Thanx! Any examples of config?

tcm0116, 2017-11-03:
I assume you're working custom code to achieve this be function? If so, the instances of HardwareSerial are already created and just need to be initialled, and used. If you want to use UART1 @ 115200 baud, then just write `Serial1.begin(115200);`. You can then use the read, write, and print functions on Serial1 to read and write data.

karabas, 2017-11-07:
Thanx a big!  I need set correct pins for every SerialXXX. I will look into code to find out.

tcm0116, 2017-11-07:
The pins should already be setup correctly. Once you can begin on the
appropriate SerialX object, it will be mapped to the correct pins for that
port.

closed 2018-02-09 by thinkyhead