MarlinFirmware/Marlin #8217: ROXY-3D I have a job for you :-)

studiodyne, 2017-11-02:
Hi, Roxanne , you told me you contributed to create UBL then , Now i know you can create this

The  ' Z_PROBE_OFFSETTING  Menu ' 
With this all in one menu , we can set nozzle offsets in one click
If you don't want to make it , i can make it but not for the moment 
Thk you 

I explain the more clear as possible : To set a zprobe offset it's not easy , you must go on motion menu , increase the offset , return to z homing , go to move 0.1 , test , and to apply value , return to motion and many times to verify all ,  so ,  the job is to make a menu that simply all , and looks like this 

Step by Step
BUTTON  ' Make offset '
- Set z_probe_offset = 10  ' it's enough in all cases '
- Make a Z homing 
- Show the move menu z ' 0.1mm ' but with another message 
       ' Place paper Under Nozzle and touch it '
       ' Click when finish '
- After menu clic ---> Calculation of Z_probe_offset = 10 - current height  
                          ---> Apply  the value
 BUTTON  ' Test Offset '
- G28Z
- - Show the move menu z ' 0.1mm ' but with another message 
       ' Place paper Under Nozzle and touch it '
       ' Click when finish '

Easy to make
I want to make it , but , i need to take a pause , ' spool swap feature have taken time and my wife want to divorce ' 😄

pandel, 2017-11-02:
Why don't you just use babystepping in combination with setting z probe offset? See configuration_adv.h for details...

studiodyne, 2017-11-02:
Oh i never seen this 
I think you have solved my problem
Yeah !
i have only six monthes of marlin knowledge and i learn something everyday
GREAT !
I don't have tested this but i already know i will love this way , i have seen the defined in adv.h
I test now

Thksssssssssss

closed 2017-11-02 by studiodyne