MarlinFirmware/Marlin #8218: Beep before halting

token47, 2017-11-02:
My printer started behaving. Randomly (almost always when I'm not near it) it will freeze with thermal runaway problem. It's difficult to investigate because when I notice it's already happened.

How could I make it beep as soon as the temperatures start to deviate from the target? I have 3C configured as a threshold (so it freezes when it reaches 97C, which amost never happen, it oscilates between 100C and 99C normally when set to 100C), it could start beeping as crazy as soon as it reaches 98C for example.

I can try to hack the code if someone points me to a strategy.

Any other ideas how to catch the problem happening?

Thanks!

thinkyhead, 2017-11-02:
The best place to stick your logic is probably in the ultralcd code for the status screen, where it's displaying the target and current temperatures. You should add a `bool` flag that is only set when your machine successfully heats up to temperature the first time, and which is cleared when you set the temp back to zero. When that flag is set, compare the current and target temps, and if they're off by 2 degrees or more, do a beep. The beep will only occur about once per second since it will be in the lcd drawing routine.

thinkyhead, 2017-11-02:
As a general feature, I like it. I've added beeps in some reasonable places on my own machine — for example, when the laser turns on it seems friendly to give an audible warning.

token47, 2017-11-02:
Well, it seems that most of the logic for this is already in place in the temperature.cpp code, where it watches the hysteresis. It seems to be just a matter of adding a second if below the if that freezes the machine, but checking for a smaller value and beeping.

I'm a noob in C/C++ but I think I got somewhere, please see if you think this is ok: https://pastebin.com/trybBCZ8 (not sure how to paste code here).

Thank you!

token47, 2017-11-02:
It works... well, sorta:

- the if was inverted, changed >= to <= and the logic works now.
- it beeps toooo much. Don't know how to solve this one.

How to make it beep once every few seconds?

token47, 2017-11-02:
I took example from the M600 routines to beep in intervals. It works now.

- "works for me", needs revision.
- configuration altered to document new option
- sanity checks altered to check for options and limits

any other thing I need to care? Do I need to / How can I submit it for revision before sending a PR?

token47, 2017-11-02:
Please see the modifications here:

https://github.com/token47/Marlin/commit/79ce8143b7e6ee3da978c126dcf864ba12dff7ff

mikko70, 2017-11-03:
Yes, i have too thermal runaway problem, sometimes when i put preheat before i start working, then drive home state and open file from SD-card. When i come back after couple minutes. System is halted.
This never happen, if i only open file.

token47, 2017-11-06:
It has been working fine for the last days, no other apparent sideeffect and it has indeed saved me a print where it beeped and I had time to investigate the issues.

@thinkyhead et all: if you can give me your opinion about the code it would be very much appreciated.

token47, 2017-11-07:
In fact I noticed a little side-effect just now. And it's reproducible. At the end of a G29 (bi-linear with 4x4 grid, if that matters) it will make a short beep sometimes, not yet sure under what circumstances.

One way to reproduce:

- heat the bed to 100C
- do a G28 (no beeps)
- do a G29 (after all the probings it makes a short beep, just once).

It seems harmless but started after I modified the code. Any ideas?

thinkyhead, 2018-02-09:
Arrived here working backward from oldest open issues. What's the status?

token47, 2018-02-09:
I tried to implement this feature because of a problem I was having with my heated bed. It worked, and warned me a few times in real cases. But it also had a bug that would make it beep gratuitously during G29 in rare cases (probably a lack of my understanding of the impacts of my changes).

I have since changed completely my heated bed and the new one is behaving well. So, when I upgraded to latest (current) version I just left this patch out and I'm not using it anymore.

We can just forget about it completely or maybe get a review from an experienced developer and maybe merge since it still might be interesting to other people. And in this case, it could be improved. I don't have a strong opinion and I don't mind to just forget about it.

thinkyhead, 2018-02-17:
I'd like to add a more general feature to have lots of audio feedback for different states. This would be friendlier to visually impaired people, or just those of us who don't continually stare at our printers. For machines capable of producing different tones this is fairly easy. But for machines that can only make a single pitch it's trickier to make the sounds very distinctive.

token47, 2018-02-19:
Well, I'm all for it. It's a better way to implement the feature for sure. Anyway I didn't have any expectations of my code being merged without a bit of work, so don't mind it.

Thank you!