MarlinFirmware/Marlin #8219: MARLIN_ML

mirek140872, 2017-11-02:
//file Configuration.h Release:tags/1.1.6
//section machine
#define SERIAL_PORT 4
#define X_BED_SIZE 216
#define Y_BED_SIZE 260
#define Z_MAX_POS 260
//#define MAX_SOFTWARE_ENDSTOPS 
//section extruder
#define EXTRUDERS 2
#define SINGLENOZZLE 
#define MK2_MULTIPLEXER 
#define HOTEND_OFFSET_X {0.0, 0.0}
#define HOTEND_OFFSET_Y {0.0, 0.0}
#define INVERT_E1_DIR true
//section temperature
#define PREHEAT_1_TEMP_BED 80
#define PREHEAT_2_TEMP_BED 100
//section homing
#define X_MIN_ENDSTOP_INVERTING true
#define Y_MIN_ENDSTOP_INVERTING true
#define Z_MIN_ENDSTOP_INVERTING true
#define X_MAX_ENDSTOP_INVERTING true
#define Y_MAX_ENDSTOP_INVERTING true
#define Z_MAX_ENDSTOP_INVERTING true
#define Z_MIN_PROBE_ENDSTOP_INVERTING true
//section motion
#define DISTINCT_E_FACTORS 
#define DEFAULT_AXIS_STEPS_PER_UNIT { 80.22, 80.22, 400, 80.22 }
#define DEFAULT_MAX_FEEDRATE { 500, 500, 2.5, 250 }
#define DEFAULT_MAX_ACCELERATION { 1500, 1500, 100, 10000 }
#define DEFAULT_ACCELERATION 1500
#define DEFAULT_RETRACT_ACCELERATION 1500
#define DEFAULT_TRAVEL_ACCELERATION 1500
//section lcd
#define LCD_LANGUAGE pl
#define ULTRA_LCD 
#define DOGLCD 
#define SDSUPPORT 
#define SD_CHECK_AND_RETRY 
//#define ENCODER_STEPS_PER_MENU_ITEM 10
#define INDIVIDUAL_AXIS_HOMING_MENU 
#define SPEAKER 
#define ULTIPANEL 
#define REPRAP_DISCOUNT_SMART_CONTROLLER 
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER 

//file Configuration_adv.h Release:tags/1.1.6
//section temperature
#define BED_CHECK_INTERVAL 50000

psavva, 2017-11-02:
Hi,

Can you clarify what you're reporting here?

bobc, 2017-11-02:
Seems to be another cat on keyboard, Marlin seems popular with cats :)

psavva, 2017-11-03:
See #7969

I think both should just be closed

closed 2017-11-03 by thinkyhead