MarlinFirmware/Marlin #8222: Azteeg - Viki - Compile error

Printalect, 2017-11-02:
### Compile Error

Switched to using marlin 1.1.6 where I am now getting a compile error when defining the use of a Viki or miniViki. This does seem to be specific to Marlin 1.1.6 since this doesn't occur on previous versions. 

It is not a serious issue, yet since I do not have the skills to solve this, I wanted to make it known to those who possibly can.

```
Error is as follows:

fastio.h:79: error: 'DIODOGLCD_SCK_WPORT' was not declared in this scope

```
Using: 
Azteeg X3 Pro
Arduino 1.8.5

------------------------------------------------------------
Config:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1439181/Configuration.zip)

Full error spit out in arduino:
[Viki Compile Error Marlin1.1.6.zip](https://github.com/MarlinFirmware/Marlin/files/1439205/Viki.Compile.Error.Marlin1.1.6.zip)

------------------------------------------------------------

Printalect, 2017-11-03:
I am open to doing any testing needed. I know the marlin community has since shifted attention and resources to 2.0 and restructuring.  I can certainly help in areas that I am familiar. Built a few printers now, just can't program and debug code all that well.

If someone wants to reconfig a ".h" or ".cpp" file and have me test it, I'm all for it. 

Also, any other testing that's needed for other code involving any of the Punucatt boards, I can do that too. Have the Azteeg X3 Pro, Azteeg Full featured, and X5 GT. 

Lastly, I did go back to the Marlin 1.1.4, reinstituting the same changes there. 1.1.4 was fine. This makes me think there is an issue related to subsequent releases.

GMagician, 2017-11-06:
I think you have to update your configuration file first...
I get "SanityCheck.h:43: error: #error "You are using an old Configuration.h file, update it before building Marlin.""

GMagician, 2017-11-06:
by the way updating configuration I get the same errors...but fix goes beyond my knowledges

Printalect, 2017-11-09:
Yeah, I'm perplexed with this. I've been sitting on it for a while now. I'm out on a work trip at the moment, so I can't manage anything from here. I'll try updating as well when I get back.

I think that it will make its way into the rest of the community as others shift their attention. Everyone seems to be working super hard on restructuring this older marlin while getting the 32bit platform up and running.

Bob-the-Kuhn, 2017-11-09:
It's complaining that the pin **DOGLCD_SCK** hasn't been defined.  

Go back to your working setup/directory and copy the files **pins_RAMPS.h** and **pins_AZTEEG_X3_PRO.h** into your new setup/directory.  That'll take care of the problem.

**DOGLCD_SCK** isn't defined for VIKI2 in these two files in the Marlin releases. One of those two files had to have been modified after they were downloaded.  You just need to bring the customized file over to the new release.

WOOKIE111, 2017-11-11:
Hello i also recieve the same error. 

Most Current Marlin 1.1.x from marlinfw.org as of Nov 11 4:14am EST

even when the old pins_RAMPS.h and pins_AZTEEG_X3_PRO.h are pasted into the new directory. as well as the other ramps related files.

it seems there is something broken for the VIKI as just uncommenting it seems to break marlin

using hte most current marlin and not modifying ANYTHING and only enableing/commenting viki2 back in on line 1355 of configurtation.h throws this crazyness

Pastebin used as its rather long.

https://pastebin.com/zQggRL7f

but commenting viki2 back out. and recompiling clears it up. and its repeatable.

@Roxy-3D  @nophead perhaps one of you two can figure this out? or know someone whom does?

WOOKIE111, 2017-11-12:
maybe @Bob-the-Kuhn too?

WOOKIE111, 2017-11-12:
Use the bugfix of marlin

 the 1.1.6 version is missing a define  for the viki and a few other display types.

Printalect, 2017-11-16:
I think @Bob-the-Kuhn is looking in the right direction, yet sadly I've done a replace on the two files you mentioned. 

I'll look more into this today. I do see that there is a new file "ultralcd_st7565_u8glib_VIKI.h". Guessing that this is a good place to start.

Also, I'm not sure if there is a relation to issue "#7368" and "#7256".

Bob-the-Kuhn, 2017-11-17:
Try adding **#define DOGLCD_SCK 52** to one of the two files in the VIKI2 section.

Printalect, 2017-11-21:
That seems to be all it is. 

There is a variable translation in the 'ultralcd_st7565_u8glib_VIKI.h' file (as set below). The pins are then defined for 'DOGLCD_CS' and 'DOGLCD_A0' in the ramps pin file. Nothing is set for SCK or MOSI. Not sure why. I haven't cross-referenced the previous versions on this yet but could this just have been missed?

ultralcd_st7565_u8glib_VIKI.h:
#define ST7565_CLK_PIN  DOGLCD_SCK
#define ST7565_DAT_PIN  DOGLCD_MOSI
#define ST7565_CS_PIN   DOGLCD_CS 
#define ST7565_A0_PIN   DOGLCD_A0

Printalect, 2017-11-22:
In following the "The Marlin 2.0 32-bit thread" thread, there has been a great deal of reconstruction on the "ultralcd_impl_*" and other LCD files in order to change up the hierarchical structure where in Marlin 1.1.6 there was a great deal of reconstruction to "ultralcd_impl_DOGM.h" and the "ultralcd_st7565_u8glib_VIKI.h" file. 

I'm not savvy enough to deduce the connections, but something isn't computing. 

I simply added pin defs to the pins_RAMPS.h, which allows the code to compile, yet feel I might be voiding arguments in the other files.

Nonetheless, I'm sure this only effects the VIKI2, miniVIKI, and possibly the MAKRPANEL.

Bob-the-Kuhn, 2017-11-22:
You're in no danger of causing problems else where with your changes.  The use of those names is limited to LCD displays.

If your LCD is working correctly then you have it 100% correct.

eyal0, 2017-11-26:
I'm also seeing this problem.  I merged bugfix-1.1.x which solved the compilation issue but the LCD no longer works.

Tomorrow I will try to run a build before 381ebc043 to see if the LCD works.

Bob-the-Kuhn, 2017-11-27:
Another option you could try is bugfix-2.0.x.  It'll compile with Arduino.  You may have to use the Arduino beta version to get around a command line length issue.

2.0.x has re-arranged the directory structure so you'll need to do some hunting to find the location to copy your customized pins_XXXX.h file(s) into.

I don't remember if your config files can just be dropped into 2.0.x.  To be on the safe side just modify the default config with your motherboard and LCD info.  If that works then finish moving the config items over.

eyal0, 2017-11-27:
I put my pins.h changes on https://github.com/MarlinFirmware/Marlin/commit/c7e0a49c94c484559f75276ca6ed9a600bbae14f , which is right before the problematic https://github.com/MarlinFirmware/Marlin/commit/381ebc043fe0cba6f915521114afefe633e7b384 and it works.  So I've found the commit that causes my LCD to stop working.  I haven't yet tried 2.0.x, I'll try that next.  Would be nice for bugfix-1.1.x to work, too!

eyal0, 2017-11-27:
Okay, I got the screen to display something in bugfix-1.1.x.  What I did was modify the default screen contrast for VIKI2 to 40 and changed the max to 63 and the default to 40.  Then I was able to see things on the screen.  However, the screen was upside-down.  I tried messing with the LCD_ROTATION_180 parameter but that made is upside-down and mirrored, somehow!  Worse.

Worse still, the display wasn't responsive and when I turned the machine off and back on, the display stopped working.

Maybe I should just try 2.0.x...

thinkyhead, 2018-02-09:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch in case the issue is already fixed. If we already got it then you can close the issue. If you still see the bad behavior then we can investigate further.

eyal0, 2018-02-12:
I'll have to dig up the code and give it a shot, sure!

thinkyhead, 2018-02-20:
Buried here: https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x

eyal0, 2018-02-20:
I was able to build a few versions and try uploading them but none of them worked.  In all cases, the screen was blank.

I didn't try connecting to the printer or doing anything, however.  I'll try them again but this time connect and see if I can command the printer to move something.  Also, I'll try sending commands to adjust the screen contrast, which as far as I know should be 0-63 but in the latest versions of code was changed to be 0-255 for Viki2.

Do you see any troubles with the customizations that I have made to the bugfix branch?  The are here:

https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...eyal0:new_s3d_bugfix-2.0.x?expand=1

Similar changes tried on bugfix-1.0.x, didn't work:

https://github.com/MarlinFirmware/Marlin/compare/bugfix-1.1.x...eyal0:new_s3d_bugfix-1.1.x?expand=1

This one works fine:

https://github.com/MarlinFirmware/Marlin/compare/c7e0a49c9...eyal0:new_s3d_c7e0a49c9

All the diffs are pretty much the same, I think.

eyal0, 2018-02-21:
I also tried adjusting the contrast from the terminal.  It didn't make the screen work, either.

Bob-the-Kuhn, 2018-02-21:
I have a VIKI2 so I can do some testing.  Please ZIP & attach your Configuration.h, Configuration_adv.h & pins_YOUR_BOARD.h files.

Also, if you can diagram what wires are tied to which pin I can check your pin definitions.  What I need is something like:
```
    NAME ON 
  VIKI2 BOARD    color wire           pin
  SDCD           orange/white         AUX1-2
  BTN            dark blue            AUX3-6
  SDCS           dark purple
  LCS            green/white stripe
  SCK            orange
  GND            black
  +VIN           red
  MOSI           yellow
  A0             brown
  MISO           gray
  ENCA           dark green
  ENCB           white
```

eyal0, 2018-02-22:
Oh man, I'm not excited about tearing apart the printer.  It's a Something 3d Royal from a few years back, not a printer that I put together myself, so I'm unfamiliar with the wiring of the Viki2.  But maybe we can look at my patch which works vs my patch which doesn't and see something?

This works:

https://github.com/MarlinFirmware/Marlin/compare/c7e0a49c9...eyal0:new_s3d_c7e0a49c9

It's the Something3D modifications that need need to be patched on to `c7e0a49c9`.  If you look at the diffs, you'll see that most of the changes are just cosmetic, not affecting Viki2.  Only at the very bottom, in pins.h, I have some changes:

```
#define SDSS               53
+#define LED_PIN            13
+
+#define SDPOWER            -1
+#define HEATER_1_PIN       -1
+#undef SD_DETECT_PIN
+#define SD_DETECT_PIN      49 // Pin 49 if using display sd interface
+#undef KILL_PIN
+#define KILL_PIN           -1
```

I patched all those same changes to bugfix-2.0.x and they didn't work: https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...eyal0:new_s3d_bugfix-2.0.x?expand=1

I'm pretty sure that I got the patch right.  Maybe a default value for one of the Viki2 pins changed in `381ebc043`?  That is the first commit in which Viki2 is broken for me.  In fact, that one doesn't even compile for me.

It looks like there is a change from:

`U8GLIB_NHD_C12864_2X u8g(DOGLCD_CS, DOGLCD_A0); // 4 stripes  HWSPI`

to:

`U8GLIB_ST7565_64128n_2x_VIKI u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);  // using SW-SPI`

Maybe the switch from HWSPI to SW-SPI started to use two new pins, MOSI and SCK, and those aren't connected for me?  I'll try commenting out the SWSPI, use the HWAPI, and see if that works.  And then try to find where those two new pins are.

eyal0, 2018-02-22:
Okay, progress!  I got the display to show something.

If I load a working build and then load bugfix-2.0.x with my patches, it works.  The screen is way too bright at contrast 140, the default, so I have to put it to 40.  The screen is rotated 180 degrees without mirroring but it's usable.

However, if I then try to load a new image (even the same one) or if I turn the printer off and on, then the screen goes blank.  The printer still works and I can control it from the terminal but the screen is blank.

Have you heard of something like this, where loading a new firmware works but after rebooting, it doesn't work?  My previous version works reliably, both after uploading firmware and after reboot.

Also, I checked: DOGLCD_SCK == SCK_PIN and DOG_LCD_MOSI == MOSI_PIN, so my system is always choosing the HW-SPI.

Bob-the-Kuhn, 2018-02-22:
Agreed - tearing the printer apart to get the wiring details should be avoided if possible.

The good news is that you're getting a useable display.  That means all the logical pin connections/definitions are correct.  No need to look at the wiring.

My expectation is that the display is not getting initialized.  Once initialized it'll retain the initialization until it's re-initialized or it's power cycled.

Try using the soft SPI version of the VIKI2 driver. 
- In 2.0.x it's `U8GLIB_64128N_2X_HAL u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);  // using SW-SPI`.  
- In bugfix-1.1.x it's `U8GLIB_ST7565_64128n_2x_VIKI u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);  // using SW-SPI`.

Bob-the-Kuhn, 2018-02-22:
I think it's worth trying the latest bugfix-1.1.x or bugfix-2.0.x.  Both had changes last week that affect the VIKI2 display.

Use your current pins_AZTEEG_xxxx.h file with the new images.

eyal0, 2018-02-22:
I was on latest bugfix-2.0.x already.  I synced up around 5 hours ago for my testing.

I'll try the SW-SPI version.  I'll try writing `#if 0` here: https://github.com/eyal0/Marlin/blob/new_s3d_bugfix-2.0.x/Marlin/src/lcd/ultralcd_impl_DOGM.h#L190

I can try it on Sunday.

eyal0, 2018-02-22:
@Bob-the-Kuhn Any drawback to using SW-SPI vs HW-SPI?

Bob-the-Kuhn, 2018-02-25:
Are you up & running?

The nice thing about software SPIs is they don't interact with each other so there's no compatibility problems.  The bad things are they're slower and they may not work if they share pins with a hard SPI.

eyal0, 2018-02-26:
I tried the SW-SPI.  It didn't work.  Totally blank screen after rebooting and trying to mess with the contrast.

Rolling back to the old way worked:

`
--- a/Marlin/src/lcd/ultralcd_impl_DOGM.h
+++ b/Marlin/src/lcd/ultralcd_impl_DOGM.h
@@ -188,7 +188,8 @@
 #elif ENABLED(U8GLIB_ST7565_64128N)
   // The MaKrPanel, Mini Viki, and Viki 2.0, ST7565 controller
   #if DOGLCD_SCK == SCK_PIN && DOGLCD_MOSI == MOSI_PIN
-    U8GLIB_64128N_2X_HAL u8g(DOGLCD_CS, DOGLCD_A0);  // using HW-SPI
+    U8GLIB_NHD_C12864_2X u8g(DOGLCD_CS, DOGLCD_A0); // 4 stripes
+    //U8GLIB_64128N_2X_HAL u8g(DOGLCD_CS, DOGLCD_A0);  // using HW-SPI
   #else
     U8GLIB_64128N_2X_HAL u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);  // using SW-SPI
   #endif
`

Bob-the-Kuhn, 2018-03-06:
If it's working with **U8GLIB_NHD_C12864_2X** then it's not your hardware.

You may want to try the latest bugfix.  There was a change to the ST7565 code so the **U8GLIB_64128N_2X_HAL** should be working again.

eyal0, 2018-04-15:
Sorry to keep you waiting, @Bob-the-Kuhn .  I have merged the latest bugfix-2.0.x and switched from the old one `U8GLIB_NHD_C12864_2X` to the one in bugfix-2.0.x `U8GLIB_64128N_2X_HAL` and it works!  Hooray!

One thing, though, I needed to add `#undef  LCD_SCREEN_ROT_180` into `Marlin/src/pins/pins.h`.

Anyway, my problem solved.  You can close this issue as far as I'm concerned.

thinkyhead, 2018-04-17:
Is there anything we should do with `LCD_SCREEN_ROT_180`? I seem to remember this was needed for some VIKI displays, but maybe not all, and maybe it depends on how the specific unit is made.

eyal0, 2018-04-18:
I can say that I didn't change the hardware between the old u8glib and this
one but the display flipped. I don't know if both u8glib use the same
#define.

thinkyhead, 2018-04-20:
That's interesting! It probably has to do with the change in lib.

thinkyhead, 2018-04-30:
> As soon as I reboot the machine, the screen stops working.

I concur with Bob that there's most likely something funky going on with the init sequence.