MarlinFirmware/Marlin #8223: [FR] Unwanted Auto Cooldown

joeedh, 2017-11-02:
I just upgraded my printer's Marlin, and it's now automatically cooling down the printer if I do "stop print" from the menu.  I very, very much do not want this to happen.  From my reading of the code (in lcd_sdcard_stop()), there is no configuration option to turn this off.

Anyway, I propose encapsulating:

thermalManager.disable_all_heaters();

in an #ifdef, and an appropriate entry added to Configuration.h

I'm not sure if I should submit a pull request, since it's such a simple change and this seems like the sort of thing the developers will want to do themselves (to fit naming/doc conventions, etc).

Thanks.

psavva, 2017-11-02:
I suggest you create a pull request.  
Nothing like a PR to contribute :)

thinkyhead, 2017-11-03:
What do you think, @tcm0116 ? Does it make sense to revert this element of #6567?

tcm0116, 2017-11-03:
I don't think we should revert it, but rather make it a configuration option as @joeedh has suggested.

stuartmp, 2017-11-08:
I think it a great idea. you could possibly have some type of time out so if no other input is given within x seconds then the cool down could start. I would love to see a re-print option too. Just a double press of the control button like micro stepping works after you start a print. You could then select from a yes / no to re-print.  As I hate having to re browse my sd card to find the file I want. If for instance something goes wrong on the first layer.

Roxy-3D, 2017-11-08:
> I would love to see a re-print option too. Just a double press of the control button like micro stepping works after you start a print. You could then select from a yes / no to re-print. As I hate having to re browse my sd card to find the file I want.

It wasn't done as a Double Click like the Z-BabyStepping...   But if you go into Configuration_adv.h you can turn on 
```cpp
  /**
   * This option makes it easier to print the same SD Card file again.
   * On print completion the LCD Menu will open with the file selected.
   * You can just click to start the print, or navigate elsewhere.
   */
  //#define SD_REPRINT_LAST_SELECTED_FILE
```

stuartmp, 2017-11-08:
Nice... Thanks for the response. I will look into that option and what can be done to to improve it.
I will start my own issue, so not to hijack joeedh and his cooldown

thinkyhead, 2017-11-14:
I also never miss a change to pimp my latest feature.

stuartmp, 2017-11-14:
So are you able to point me in the right direction for the modifications or are you thinking of adding this feature to the work that you have done....

thinkyhead, 2017-11-14:
Trying to get 1.1.7 out. This is on the low priority list. It's not a difficult thing to do, just not vital for the upcoming release.

stuartmp, 2017-11-14:
Nice... keep up the awesome work.

joeedh, 2017-11-19:
Okay, I have created a pull request, feel free to redo it (it really is like three lines of code).  #8494

wikolii, 2018-12-02:
I can't find any updates on this issue, i just updated (too late, i know) to 1.1.9 and can't disable auto cooldown on print stop the way i used to on previous versions.

I used to comment out 1 line on Ultralcd.cpp and now that line doesn't existe anymore.
"thermalManager.disable_all_heaters();"

Thanks

boelle, 2019-02-19:
@thinkyhead since there is a PR could this one not be closed?

InsanityAutomation, 2019-06-22:
FYI this has been implemented in config adv

 // Leave the heaters on after Stop Print (not recommended!)
  //#define SD_ABORT_NO_COOLDOWN

So this can be closed.

boelle, 2019-07-21:
Closing as this has been implemented

closed 2019-07-21 by boelle