MarlinFirmware/Marlin #8224: M665 missing variables

ProduktTester, 2017-11-02:
Hi @ all,
i have bought my first Delta printer 4 weeks ago.
I want to calibrate the printer with G33. That worked very good in the first look.
The Measurements from the parts were not that good. I then wanted to use the diagonal rod trim settings.
But without any luck.

But I think that's caused by the firmware (1.1.5 or 1.1.6... 1.1.x)

I found that at a older post:

   * M665: Set delta configurations
   *    L = diagonal rod
   *    R = delta radius
   *    S = segments per second
   *    A = Alpha (Tower 1) diagonal rod trim
   *    B = Beta (Tower 2) diagonal rod trim
   *    C = Gamma (Tower 3) diagonal rod trim
   *    I = Alpha (Tower 1) radius trim
   *    J = Beta (Tower 2) radius trim
   *    K = Gamma (Tower 3) radius trim

I checked then the Marlin code (configuration_store)1.1.x:
(B) is wrong and (A,C,I,J,K) are missing. for the "delta calibration radius" should a new variable be found.

 * DELTA:                                                             40 bytes
 *  352  M666 XYZ  delta_endstop_adj                (float x3)
 *  364  M665 R    delta_radius                           (float)
 *  368  M665 L    delta_diagonal_rod                (float)
 *  372  M665 S    delta_segments_per_second  (float)
 *  376  M665 B    delta_calibration_radius         (float)
 *  380  M665 X    delta_tower_angle_trim[A]     (float)
 *  384  M665 Y    delta_tower_angle_trim[B]     (float)
 *  388  M665 Z    delta_tower_angle_trim[C]     (float)

The undefined variables and B are used below (Marlin _main):
(but B also used for DELTA_DIAGONAL_ROD_TRIM_TOWER B and for the delta_calibration_radius)

void recalc_delta_settings(float radius, float diagonal_rod, float tower_angle_trim[ABC]) {
    const float trt[ABC] = DELTA_RADIUS_TRIM_TOWER,
                drt[ABC] = DELTA_DIAGONAL_ROD_TRIM_TOWER;
    delta_tower[A_AXIS][X_AXIS] = cos(RADIANS(210 + tower_angle_trim[A_AXIS])) * (radius + trt[A_AXIS]); // front left tower
    delta_tower[A_AXIS][Y_AXIS] = sin(RADIANS(210 + tower_angle_trim[A_AXIS])) * (radius + trt[A_AXIS]);
    delta_tower[B_AXIS][X_AXIS] = cos(RADIANS(330 + tower_angle_trim[B_AXIS])) * (radius + trt[B_AXIS]); // front right tower
    delta_tower[B_AXIS][Y_AXIS] = sin(RADIANS(330 + tower_angle_trim[B_AXIS])) * (radius + trt[B_AXIS]);
    delta_tower[C_AXIS][X_AXIS] = cos(RADIANS( 90 + tower_angle_trim[C_AXIS])) * (radius + trt[C_AXIS]); // back middle tower
    delta_tower[C_AXIS][Y_AXIS] = sin(RADIANS( 90 + tower_angle_trim[C_AXIS])) * (radius + trt[C_AXIS]);
    delta_diagonal_rod_2_tower[A_AXIS] = sq(diagonal_rod + drt[A_AXIS]);
    delta_diagonal_rod_2_tower[B_AXIS] = sq(diagonal_rod + drt[B_AXIS]);
    delta_diagonal_rod_2_tower[C_AXIS] = sq(diagonal_rod + drt[C_AXIS]);

Sorry for my not perfect english.
I hope i am right, if i am wrong please give me help.

Thanks 
ProduktTester

thinkyhead, 2017-11-03:
@LVD-AC Can you provide some guidance?

I believe the old documentation for `M665` is out of date and no longer applies. Current documentation reads:
```cpp
  /**
   * M665: Set delta configurations
   *
   *    H = delta height
   *    L = diagonal rod
   *    R = delta radius
   *    S = segments per second
   *    B = delta calibration radius
   *    X = Alpha (Tower 1) angle trim
   *    Y = Beta (Tower 2) angle trim
   *    Z = Rotate A and B by this angle
   */
```

thinkyhead, 2017-11-03:
I haven't pushed the latest `G33` documentation up to the website. Will do that soon…

LVD-AC, 2017-11-03:
"diagonal rod trim" and "radius trim" can't be set by G-code and are to be set in Configuration.h (uncomment them if you want to use)

Phr3d13, 2019-06-10:
@thinkyhead Another one that needs to be closed? Sorry, I got bored.

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed? Sorry, I got bored.

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle