MarlinFirmware/Marlin #8226: No movement on any axis MKS SBASE

icthus86, 2017-11-02:
Motors turn on with no movement using pronterface. I can not move any of all the axis by hand so the steppers are engaging. LCD display is showing where it believes each axis is and changes when told to move.

This could be nothing more than my own stupidity and inexperience!
``
[Configuration(_adv).h.zip](https://github.com/MarlinFirmware/Marlin/files/1439576/Configuration._adv.h.zip)

maar1201gh, 2017-11-02:
I confirm the same fact. Motors turn on with no movement

Spawn32, 2017-11-02:
But operates normally with other programs ?

Spawn32, 2017-11-02:
you could try changing #define MINIMUM_STEPPER_PULSE 4 // (µs) The smallest stepper pulse allowed
in _adv.h

Test it with the double or trippel value, dont know why that would change using different programs, but worth a test :)

karabas, 2017-11-03:
I connect one motor to X and "Move axis" via display menu. It works as usually. Turns well.
I still did not setup on my printer as Sdsupport is broken for now.

icthus86, 2017-11-03:
I can't move from the LCD controls or pronterface. I'll have a look at MINIMUM_STEPPER_PULSE later on this afternoon.

karabas, 2017-11-03:
I use default Marlin2 Sbase settings. Have tried move X via pronterface - works.

icthus86, 2017-11-03:
@karabas it does work, however I have 1/32 microsteps and 0.9 steppers so I need to change the configuration to match my machine and it just doesn't seem to work.

I have now changed the pulse to 8 and 12 with no changes. I'm assuming this is something to do with my setup being odd?

karabas, 2017-11-03:
My Sbase jumper  is 1/32 on. Stepper is 1.8. I have not 0.9 steppers to check.

icthus86, 2017-11-03:
@karabas So have you not set the mircosteps in the configuration_adv.h?

Spawn32, 2017-11-03:
The examples for Sbase is setup for 1/32 and 1.8 steppers, dont have a 0.9 to test with...

maar1201gh, 2017-11-03:
@Spawn32 Configuration_adv.h has MINIMUM_STEPPER_PULSE 0.
I changed to MINIMUM_STEPPER_PULSE 4 and it work. 
Maybe is better to update Configuration_adv.h
Thanks

Spawn32, 2017-11-03:
@maar1201gh 
No, you have used the wrong example file, Marlin/src/config/examples/Mks/Sbase/Configuration_adv.h
has MINIMUM_STEPPER_PULSE 4, you should always use the example file if it exists :)

icthus86, 2017-11-04:
So I ended up starting from scratch again changing one thing at a time and uploading, It all works now and I believe it was my own stupidity! I changed the wrong setting for inverting axis. I used define X_ENABLE_ON 1 which is wrong in my setup and now everything is printing perfectly. A few changes I need to make to the settings such as bed centre and offsets but everything is working well. I was ready for chucking this board as I just couldn't get it to work with smoothieware at all!  

Thank you all!

closed 2017-11-04 by icthus86