MarlinFirmware/Marlin #8232: [FR] Disable M200 Feature Completely - AKA completely shut off volumetric extrusion option

houseofbugs, 2017-11-03:
I have a lot of people accidentally turning on the volumetric extrusion and it causes issues for them because its on. 

Can we have an option in the Configuration.h or Configuration_adv.h to disable this completely? Not many people use it and it would also be great to remove it to save space on the 1284P boards.

thinkyhead, 2017-11-03:
Sounds like a good idea. We'll look into it soon.

houseofbugs, 2017-11-03:
Thanks. I run a small maker space where we do a lot of support and work with marlin printers. We get at least 2-3 people a week that change the filament diameter and then it turns this on. They then go on a hunt to recalibrate their extruder and it's a mess.

houseofbugs, 2017-11-06:
In the mean time is there a way I can manually remove it?

houseofbugs, 2017-12-18:
Just following up. Was any progress made on this? I checked the 1.1.7 and it doesn't look like it is an option. If I'm correct I'll start working on stripping it out.

thinkyhead, 2017-12-20:
No, it hasn't been made strippable yet. It should be straightforward to do so.

thinkyhead, 2017-12-20:
Here you go. The above PRs allow volumetric extrusion to be completely disabled.

closed 2017-12-20 by thinkyhead