MarlinFirmware/Marlin #8233: language zh_CN get confusion code in LCD viewer

CrazyMouse, 2017-11-03:
### Bug Report
- Description: language zh_CN get confusion code in LCD viewer
- Expected behaviour: view chinese character
- Actual behaviour: view confusion code
- Steps to reproduce:
set language to zh_CN in Configuration.h

my lcd type is 'MKS  12864LCD_Marlin '

AnHardt, 2017-11-03:
Sorry. CN has not been updated for a long long time,
We are waiting for a capable Chinese developer / translator.
But even then it will be very difficult to do anything meaningful with only ~127 available symbols.

Do you know about a readable 12x12 pixel utf-font?

AnHardt, 2017-11-03:
The code in PR #5441 is not that old. Maybe Marlin in that state is usable for you.

CrazyMouse, 2017-11-06:
I really hope I can do that ,but I'm a java develop, and just only can understand a little C/C++ Code,I'm very sorry!

thinkyhead, 2017-11-14:
Java is the "FOX News" of programming languages.

Do you see the same problem in —for example— Marlin 1.1.4?

AnHardt, 2017-11-14:
Try 'cn'.
'zh_CN' was added in the hope of #5441. Currently it can't work.

CrazyMouse, 2017-11-15:
@thinkyhead I use Marlin 1.1.6 ,zh_CN has the same problem.I'm using back to 'cn' now ,It's works good enough

boelle, 2019-02-19:
@CrazyMouse is the problem still there with the latest bugfix 2.0?

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed? Sorry, I got bored.

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle