MarlinFirmware/Marlin #8235: UBL 1.1.x and 2.0.x don't match

thinkyhead, 2017-11-03:
One of these things is not like the others.
One of these things just doesn't belong.
One of these things is not like the others.
Can you tell which one it is…
Before I finish this song?

Somehow changes to UBL were made to one branch and not the other. This is especially obvious with `find_closest_mesh_point_of_type` but probably impacts other methods also. Please check it over and make a PR with the necessary updates to the appropriate branch.

Roxy-3D, 2017-11-03:
I will be putting some fixes into the bugfix-v1.1.x branch very soon.    The 
`
  inline bool prepare_move_to_destination_cartesian() {}
```
routine is doing the wrong thing for UBL and it is causing issues with the circle drawing for G26.    

I can get UBL identical across the two at the same time.     But the difference is only for the G29 P1 U option to space out the probes to assist with physical bed leveling.   The new code does a better job but the old code does work.    

With that said...   I'll do a 3-way merge in next day or two and get everything identical.  But I need to understand and resolve the G26 Circle problem first.

thinkyhead, 2017-11-14:
Pretty well matching now? I'd guess they are close. Try pasting code from each branch to replace the equivalent code on its counterpart and see what diffs show up in Github Desktop.

boelle, 2019-02-19:
do they match now?

closed 2019-02-19 by thinkyhead