MarlinFirmware/Marlin #8237: SD init fail on RUMBA SD card adapter (no LCD)

Diaoul, 2017-11-03:
I have [this](https://jelwek.pl/296-large_default/rumba-sd-card-reader.jpg) SD card adapter plugged in my RUMBA board but I cannot get it to work. I constantly have `SD init fail` messages.

Here is what I've tried so far:

- SDSUPPORT enabled (even tried without)
- The adapter is correctly plugged in
- Various SPI_SPEED settings
- SD_CHECK_AND_RETRY enabled or not
- REPRAP_DISCOUNT_SMART_CONTROLLER enabled or not (as I cannot find anything with only SD support)
- 4 different SD cards (class 4, class 10, 8GB and 16GB)

Is this SD card adapter supported? If so, what am I doing wrong? If not, how can I help to have it working (pointers to the source code appreciated).

karabas, 2017-11-03:
I am not able to get ANY sdcard reader to work. I tried two different displays.
I think, something wrong with SPI pins assignment

thinkyhead, 2017-11-04:
Which plug is the SD adapter plugged into on the RUMBA? I don't think the RUMBA uses non-standard (non-default) SPI pins, or we'd get a lot more complaints. I presume —unlike @karabas— that you can get SD to work using the external LCD controller…?

Diaoul, 2017-11-04:
We'll see, I've ordered one on eBay so it'll take a while...

karabas, 2017-11-07:
Sorry, I use Sbase 32bit board not arduino based.

boelle, 2019-02-19:
@Diaoul is the problem still there with latest bugfix 2.0?

Diaoul, 2019-02-20:
I need to check but I don't think I'm going to make the switch to 2.0 just yet given that I just print from time to time. You can close this if you want and I'll reopen after testing with 2.0 if needed.
It seems I'm the only one using this SD adapter anyways.

closed 2019-02-20 by Diaoul