MarlinFirmware/Marlin #8240: Nozzle clean math issues

agreen, 2017-11-03:
### Bug Report
- Description: Nozzle clean calculates the math wrong if the start position is smaller than the end position. I have my nozzle clean set for start  X -7 Y56, End Y-14 Y86. It winds up cleaning right next to where it's supposed to, on the X axis. When I changed the start/end positions around, I got the problem on the Y axis instead.

Also, My nozzle clean is at Z12, and NOZZLE_CLEAN_GOBACK is defined. The GOBACK position is lower on the bed, and the problem is, it plunges down before moving over. This is probably meant for when the brush is lower than the bed, so that the nozzle will raise up before it moves over, but in my case, it crashes into my nozzle clean area. It should make sure it moves up if the move raises the Z, and if the move doesn't raise Z, move over first, then down.

thinkyhead, 2017-11-03:
@jbrazio is the original author of this feature. I'd best consult with him before I break it.

thinkyhead, 2017-11-04:
I'm pretty confident that #8248 will fix your Z movement issue, but I didn't see any obvious reason why it isn't honoring the negative coordinates, but I'll look at it closely and add onto #8248 when I find a solution.

Roxy-3D, 2017-11-04:
@jbrazio    You would have fun helping out with Marlin!!!       Come on
back for a while!

jbrazio, 2017-11-04:
@agreen is right the feature was only designed to use cleaning areas lower than the bed; the move first to X, Y then move Z is to deal with clean areas where some sort of container may be present i.e. think something like a sponge inside a box, the sponge has 3mm height but the container holding the sponge has 3.5mm height, if we do a three axis sync move like in Scott's #8249 the nozzle may crash against the container.

@agreen the negative math bug is present on all cleaning types ?
It should work, there is no reason why negative coordinates should not be used, specially on deltas where the (0, 0) is at the center of the table.

agreen, 2017-11-04:
@jbrazio My apology I meant to specify. Stroke is OK, Zig Zag is not. I didn't test circular.

thinkyhead, 2017-11-04:
I'll check the zig-zag code more closely…

thinkyhead, 2017-11-04:
> if we do a three axis sync move like in Scott's #8248 and #8249 the nozzle may crash against the container.

Read the code of `do_blocking_move_to`.

thinkyhead, 2017-11-04:
@agreen I can't find any obvious cause for nozzle clean failing to work in negative coordinate space, but I have done additional cleanup to #8248 and that should have a positive effect.

I'm assuming you don't have any `M206` or `G92` offset applied, because in that case the whole clean operation would be in the wrong spot, not just the start position. Anyway, that issue should also be cleared up soon as we overhaul workspaces.

jbrazio, 2017-11-04:
> Read the code of do_blocking_move_to.

Things do change around here.
👍

agreen, 2017-11-05:
@thinkyhead I don't have M206 nor G92 offsets. I pulled your PR and it fixes the Z issue. I also added some serial output, and I can 100% confirm it does NOT fix the stated issue.
in my case (my coordinates listed in original post):
zigpos = 7
start.x = -7
-7 + 7 = 0 (should be -14).  FABS is important to get absolute length, but you need to know the sign when doing the addition else you get the wrong data.

thinkyhead, 2017-11-05:
> FABS is important to get absolute length, but you need to know the sign when doing the addition else you get the wrong data.

Aha, yes. That makes sense. You could work around the issue by swapping the `start` and `end` sides so that the zigzag starts on the opposite side of the rectangular area. But, I totally see where the bug lies and will apply a patch shortly!

thinkyhead, 2017-11-05:
Ok, I've re-written the `zigzag` method so it respects the passed `start` and `end` points instead of referring back to the config points. To accomplish the zigzag motion it simply moves to one side or the other (i.e., `start.y` or `end.y`) on alternating strokes. Should work fine now!

closed 2017-11-06 by thinkyhead