MarlinFirmware/Marlin #8242: "initialize SD" menu missing in 1.1.6 for Creality CR-10S

Tcoton, 2017-11-03:
### Bug Report

- Description: 
SD Initialize menu is not available in 1.1.6 for the CR-10S, the printer must be powered on with the SD card inserted in order to list files. If the SD card is removed then reinserted, it is impossible to list anything and the sub menu "Print from SD card" becomes sluggish with nothing to list. If the printer is powered on with no card inserted, the menu becomes "No SD card" and its sub-menu displays a useless folder icon. 

- Expected behaviour: When a SD Card is not detected, going into "No SD card" should initialize the SD Card and list the files
- Actual behaviour: When a SD card is inserted after power on, there is no other possibility to initialize it than power cycle.

- Steps to reproduce:

  - Start the printer with an SD card inserted and click once to show the menu, click on "print from SD card" to list files, they should be listed.
  - Return to status screen and remove the SD card
  - Return to the menu and click on "Print from SD card", files are not listed
 
  - Start the printer with no SD card inserted and click once to show the menu, click on "No SD card", there is just a folder icon
  - Insert a SD card and click on that folder icon, nothing happens
  - Return to status screen and go back to the menu "No SD card", still nothing possible to initialize the SD card.
  - Switch the printer off then on, the SD card content can be listed in "Print from SD card".

[Config_files.zip](https://github.com/MarlinFirmware/Marlin/files/1441880/Config_files.zip)

Attached a ZIP of `Configuration.h` and `Configuration_adv.h`.

thinkyhead, 2017-11-03:
Try doing a low-level format on your SD card. Very slow SD cards can confuse Marlin.

Tcoton, 2017-11-04:
I did a low level format on an Class 10 sd card and it does not fix the issue. The SD card is only detected if inserted before power on.

Tcoton, 2017-11-04:
Found it, there is a typo in "ultralcd.cpp" lines 977 and 985, it should be: #if PIN_EXISTS(SD_DETECT)
 instead of #if !PIN_EXISTS(SD_DETECT). Now I can initialize the SD card even after removing it. There is a "Change SD card" menu to clear the buffer in between 2 SD cards.

AnHardt, 2017-11-04:
No. This menues are only needed when the is no SD_DETECT-pin. Otherwise it should be done in:
https://github.com/MarlinFirmware/Marlin/blob/f6f881f5ed954bc2f44ca64895b6af99e74265a2/Marlin/ultralcd.cpp#L4514

Tcoton, 2017-11-04:
Maybe, all I know is that I can use my SD card on the Creality CR-10S with this fix. Other users with a Creality CR-10S have the same issue with the SD card only detected once after power on. I wish I did not have to touch anything at all!

isilcala, 2017-11-06:
I have the exact same issue.

Tcoton, 2017-11-16:
It seems that for the CReality CR-10S, configuration_adv.h must have "#define SD_DETECT_INVERTED" disabled and pins_RAMPS.h "#define SD_DETECT_PIN" on line 343 set to -1 in order to regain control of the SD cardreader. So, no typo and no need to edit the "ultralcd.cpp".

configuration_adv.h

`
#if ENABLED(SDSUPPORT)

  // Some RAMPS and other boards don't detect when an SD card is inserted. You can work
  // around this by connecting a push button or single throw switch to the pin defined
  // as SD_DETECT_PIN in your board's pins definitions.
  // This setting should be disabled unless you are using a push button, pulling the pin to ground.
  // Note: This is always disabled for ULTIPANEL (except ELB_FULL_GRAPHIC_CONTROLLER).
  //#define SD_DETECT_INVERTED
`

pins_RAMPS.h
` // LCD Display input pins
  //
  #if ENABLED(NEWPANEL)

    #if ENABLED(REPRAP_DISCOUNT_SMART_CONTROLLER)

      #define BEEPER_PIN        37

      #if ENABLED(CR10_STOCKDISPLAY)
        #define BTN_EN1         17
        #define BTN_EN2         23
      #else
        #define BTN_EN1         31
        #define BTN_EN2         33
      #endif

      #define BTN_ENC           35
      #define SD_DETECT_PIN     -1 // origin value = 49
      #define KILL_PIN          41`

closed 2018-02-09 by thinkyhead