MarlinFirmware/Marlin #8254: Need dev's / mod's approval for use og logo...

Spawn32, 2017-11-04:
Will delete this "Issue" as soon as i get a answer.
Board going to PCB gab on monday, is the use og the Marlin logo ok ?
The board is OSH and will be published on github as soon as i am finished and have cleaned up the files.
Board will be black but haven't derided on whit or yellow logo / text yet.

[image]
[image]

Roxy-3D, 2017-11-04:
I would feel more comfortable if the board was tested and working well before the Marlin logo was put on it.  

But with that said...   That does look like a nice board.    I like the Aux Stepper Motor connector.  A person can do Dual Z motors or a 3rd extruder.

Tannoo, 2017-11-04:
No provision for dual Z motors without using the Aux?

Tannoo, 2017-11-04:
>I would feel more comfortable if the board was tested and working well before the Marlin logo was put on it.

I agree.

Roxy-3D, 2017-11-04:
> No provision for dual Z motors without using the Aux?

Would it make sense to put a second connector on the Z motor?   If a person has multiple Z motors and wants to drive them from the same chip, that would make it easier.

Spawn32, 2017-11-04:
Yes, i thought about that, but had to decide not to do it this time, the PCB footprint around a TMC2660 is pretty big, and i actually have run out of space for a thing like that on only 4 layers :/
Only way of doing that would be to make the pcb a bit bigger, will consider that on a new revision :)

You could use a simple splitter, but i always prefer running a a second Z motor from it own stepper driver, the TMC2660 will probably be running pretty hot with 2, and we do have 6 stepper drivers :)

Regarding the logo, i will just order a few pcb this time and they will only end up with people helping me getting a HAL working, probably one of you guy's :)

I am just a private person, not a company, i started this just because i am so tired of all wires going to all extra boards, wanted something with everything included and the MCU power that will be very fast for several years to come.

When i get this working and if people wants the board i could produce them, but the price of the quality steppers and the STM32F765ZGT6 (1MB x8 flash, 512x8 kb ram, 216mhz, 462 DMIPS/2.14 DMIPS/MHz and a FPU) makes this a high cost board. (can almost run linux on this puppy).

It's only a short time before someone else brings out a STM32F7 board so a HAL would be very nice 😄 
I have ST's HAL for it, have no idea how much work it would be to adapt it to Marlin ?

Tannoo, 2017-11-05:
Is the RGB 5V or 12V? Selectable?

Spawn32, 2017-11-05:
@Tannoo yes, it has a dedicated buck converter board for the RGB that can be adjusted from 5-28V :)

Spawn32, 2017-11-05:
Sendt the gerber files to the PCB fab 5 hours ago, got a messaged now that it passed the tests and was in production, efficient people those Chinese, 72hour service 😄

RowanMeara, 2017-11-06:
I have been following your posts about the board and I think it looks awesome!  I can't wait to see how it works.

Spawn32, 2017-11-25:
[image]

On the test bench 😄

stigjoergensen, 2017-11-25:
looks awsome...

aby idea about what the pricetag on one of these will be, if... ;)

Spawn32, 2017-11-25:
Need som debug help..
Most stuff is now working, usb, temp sensors, fans, hotends, bed.
cs to TMC2660 is ok and spi to TMC2660 seems to be working (looks like it on the scope) set the spi clk to 3.2mhz, 4mhz is max for TMC2660..

I can lock/ unlock stepper motor (enable / disable) but i can not make it move, i have come to a point where i cant do anything more if i cant get the debug stuff in TMC26XStepperSTM32.cpp to print to the host usb or some other serial, have tried every trick in the book, can t get i to pint the debug info in Marlin.
At one point i managed to compile it directing it to the main usb host port, but it crashed the usb port at startup :/
Also tested with @tcm0116 multihost branch to use the second serial for debug, but it seems that it cant be compiled with arduino yet ? 

So would love if someone can fix it so i can get some debug info, have included the stm32f7 hal, this hal is mostly work done by Chriss Barr (for the F4) but i adapted it to the F7...

Please someone, need help :)
@Roxy-3D with friends :)

Roxy-3D, 2017-11-25:
How are you locking and unlocking the stepper motors?  You are not doing that with PronterFace, right?

> At one point i managed to compile it directing it to the main usb host port, but it crashed the usb port at startup :/

This is the path I would explore.   Because if this can be made to work...  Then a host program can be used to talk to the board.    The USB port work for Serial on the F4 boards ?   If so...   It probably isn't that much to get it to work on the F7 boards.   (But of course, that is just uninformed opinion!)

Spawn32, 2017-11-25:
Not pronto but simplify3d :)
The hal uses STM32Generic library so serial should work even if i haven't really tested it, will make a smal prog and test..

@tcm0116 can you confirm that the multihost can't be compiled yet with arduino ? typical that i diden't save the compile errors, but if you now it should work i can have another go at it ?

Spawn32, 2017-11-25:
stigjoergensen, thanks for those nice words ;)
Have no ider yet, this first one is hand build, you typical hav max 3 hours to place the components before you get in trouble whit the solder past, i used 3.5 hours, 402 parts, 0.7mm :)

I started this for fun and because wanted one, but it looks nice and i have all bit's and pieces for a pick and place machine on the way to me :)

It probably going to be a enthusiast board in the higher ends of duet and smoothie board regarding price.. expensive components :/ but to early to say exact.

Roxy-3D, 2017-11-25:
> Not pronto but simplify3d :)
> The hal uses STM32Generic library so serial should work even if i haven't really tested it, will make a smal prog and test..

OK!   If you can talk to the board at all....   We can kludge up some debug stuff for you.   And seriously...  I use MAX7219 8x8 LED matrix's to debug things.   That support is merged in with basic sample examples that show you the depth of the Read and Write command queues.    

https://www.ebay.com/itm/MAX7219-Dot-Matrix-8x8-Led-Display-Module-MCU-Control-For-Arduino-TB-PL/222719373894?epid=9009430186&hash=item33db1bf246:g:rQEAAOSwOtdYTge9

You should order a couple of those because you can tell it to output what ever you want on the matrix.

stigjoergensen, 2017-11-25:
A pick'n'place machine?

Converting a 3d printer to do that?... i had that thought... and might do
at some point...

Yeaa - i got it would be on the high side... but would rather support an
individual than some knockoff or commercial business... it means more
development... and most likely a direct access to developer if i might have
ideas for improvement/expansion

Spawn32, 2017-11-25:
Nice, order'd 2 now from UK, faster shipping to me :)

Yes, Marlin is working and communicating like it should over USB, so if i could only stick another serial port output in there i would be fine, on the AUX port you can see in the picture i have UART5 available, but Marlin is being noty and want allow me to use it :)

Spawn32, 2017-11-25:
A 3d printer would work, but the problem is space, you need room for it to pick components so i am building a cross between this 2 projects :)
http://openpnp.org/hardware/
https://www.liteplacer.com/

Will be using my MKS Sbase for that and the The Borg for my 3d printer 😄

Roxy-3D, 2017-11-25:
This is a super cool project!

Can this solder the parts have it picks them up and places them?

Spawn32, 2017-11-25:
yes it's cool, it cant solder the parts, but does all the hard work:

1. Order a solder past stencil when you order your pcb about 16usd extra
2. apply soler paste using the stencil to the pcb.
3. use a pick an place file generated by you pcb cad, load it in openpnp (running marlin or smoothie on controller)
4. watch your machine do the dirty work.
5. cook it in an ir oven an you are done :) , a cheap T-962 with this modified firmware is a great oven.

https://github.com/UnifiedEngineering/T-962-improvements/wiki

Spawn32, 2017-11-25:
btw, when i try to stepp in simplify3d, marlin starts / locks the motor an goes in to a READ: echo:busy: processing busy: processing loop..

teemuatlut, 2017-11-25:
I'm getting my TMC2660 drivers on Monday from Panucatt so I can start doing testing them on the AVR platform at first. The drivers or the upcoming library only need a functional SPI library.

The reason you can't drive the motors is likely because you haven't configured the off time (TOFF) value.

tcm0116, 2017-11-25:
>can you confirm that the multihost can't be compiled yet with arduino ?

@Spawn32 With the default configuration, it compiles just fine on my MacBook. I'll need to know more info on your configuration in order to look into your issue.

teemuatlut, 2017-11-25:
Look at the TMC2660 datasheet on page 21. Chopconf register bits 0..3.
A setting of 0b0000 means "Driver disable, all bridges off".
This is basically the only setting that is mandatory to run the TMC steppers outside of standalone mode.

The library will not be a priority for me as I have quite a lot of other things I need to do but I'll try to get something basic so that they at least spin a stepper.

Spawn32, 2017-11-25:
Ok, i was sure it had turned the bridge on since it locks / holds the stepper motor.

I volunteer to be a guinea pig when you finds time to write it 😄

Spawn32, 2017-11-25:
@tcm0116 i will clean up my files an transfer my latest changes from the branch i am using over to multihost and do another test compile later to night, will send you the compile errors when done :)

teemuatlut, 2017-11-25:
It's actually a good sign that they lock up. We will see if there's more that they need when I get to working with the chip.

Spawn32, 2017-11-25:
the TOFF looks to be ok, starts off / init's at 0 but is changed when you issue a move, think i need the debug info to figure this one out...

teemuatlut, 2017-11-25:
The TOFF value is changed only when you change it specifically. It won't automatically change when you start a stepping pulse.

Spawn32, 2017-11-25:
it's set to 0 in init, changed in the start rutine and changed again in the un-start :)
actually a lot of code to get this thing running, probably want start if it get's the resistor part wrong so it can calculate the current draw.. :)

Spawn32, 2017-11-25:
@tcm0116 
Before i test again, the physical serialport i need on my board is UART5, i would then use:
#define SERIAL_PORT_2 5 ? (0 is usb)

and what exactly would i use insted of: 
`Serial.println("TMC26X stepper library");
Serial.print("CS pin: ");`

and do i need to include something extra in the cpp and h file ?

Spawn32, 2017-11-25:
@tcm0116 
The Marlin-bugfix-2.0.x branch i use for testing was downloaded on the 15-Nov and compiles fine.
i applied the same patches to you multihost branch an end up with this errors, haven't enabled debug in the TMC26x.cpp yet, included my pins config :

`n file included from c:\users\familien\appdata\local\temp\arduino_build_859897\sketch\src\module\planner.h:35:0,

                 from c:\users\familien\appdata\local\temp\arduino_build_859897\sketch\src\module\stepper.h:53,

                 from C:\Users\Familien\AppData\Local\Temp\arduino_build_859897\sketch\src\gcode\host\M114.cpp:27:

c:\users\familien\appdata\local\temp\arduino_build_859897\sketch\src\marlin.h:219:29: error: 'pin_t' does not name a type

 bool pin_is_protected(const pin_t pin);

                             ^~~~~

In file included from C:\Users\Familien\AppData\Local\Temp\arduino_build_859897\sketch\src\gcode\host\M114.cpp:27:0:

c:\users\familien\appdata\local\temp\arduino_build_859897\sketch\src\module\stepper.h:115:12: error: 'hal_timer_t' does not name a type

     static hal_timer_t acc_step_rate; // needed for deceleration start point

            ^~~~~~~~~~~

c:\users\familien\appdata\local\temp\arduino_build_859897\sketch\src\module\stepper.h:117:12: error: 'hal_timer_t' does not name a type

     static hal_timer_t OCR1A_nominal;

            ^~~~~~~~~~~

In file included from C:\Users\Familien\AppData\Local\Temp\arduino_build_859897\sketch\src\gcode\host\M114.cpp:27:0:

c:\users\familien\appdata\local\temp\arduino_build_859897\sketch\src\module\stepper.h:280:25: error: 'hal_timer_t' does not name a type

     static FORCE_INLINE hal_timer_t calc_timer(hal_timer_t step_rate) {

                         ^~~~~~~~~~~

c:\users\familien\appdata\local\temp\arduino_build_859897\sketch\src\module\stepper.h: In static member function 'static void Stepper::trapezoid_generator_reset()':

c:\users\familien\appdata\local\temp\arduino_build_859897\sketch\src\module\stepper.h:361:7: error: 'OCR1A_nominal' was not declared in this scope

       OCR1A_nominal = calc_timer(current_block->nominal_rate);

       ^~~~~~~~~~~~~

c:\users\familien\appdata\local\temp\arduino_build_859897\sketch\src\module\stepper.h:361:61: error: 'calc_timer' was not declared in this scope

       OCR1A_nominal = calc_timer(current_block->nominal_rate);

                                                             ^

c:\users\familien\appdata\local\temp\arduino_build_859897\sketch\src\module\stepper.h:364:7: error: 'acc_step_rate' was not declared in this scope

       acc_step_rate = current_block->initial_rate;
[pins_THE_BORG.zip](https://github.com/MarlinFirmware/Marlin/files/1503446/pins_THE_BORG.zip)
`

tcm0116, 2017-11-25:
@Spawn32 there have probably been updates to the HAL you're using since I
last re-based my multi host branch. I'd suggest checking out my multi host
branch and then merging in the latest bugfix-2.0.x branch.

Spawn32, 2017-11-25:
the hal i am using from the stm32F4 witch i adapted to F7 have not been updated for a long time, but i think it even compiles in the very lates Marlin-bugfix-2.0.x branch, will test that now.

You multihost branch i tested here is the latest you have on you github.. might there be som pin scheme difference in you multihost that is not merged with the current Marlin-bugfix-2.0.x branch that messes this up ? if you look in my borg pin def, it's not exactly marlin standard, lol

tcm0116, 2017-11-25:
@Spawn32 I just re-based by branch to the latest bugfix-2.0.x, so you'll need to pull it back down. If you want to get the multi-host working with your HAL, you'll need to duplicate the code from the LPC1768 HAL.h file that defines `MYSERIAL0` and `MYSERIAL1` in your HAL. If you have both set and `NUM_SERIAL` defined as 2, then, the multi-host code should work.

Spawn32, 2017-11-25:
Nice, thanks :)
Almost finished patching the latest  bugfix-2.0.x, will test compile that first and du the same to you new multihost after, we should get multihost merged, that function is really needed in so many ways :)

Spawn32, 2017-11-26:
arg..
SET_BIT messes up compiling my hal again, i used to have to do this to make it compile.
in macros.h:
#define SET_BIT(n,b,value) (n) ^= ((-value)^(n)) & (_BV(b))
//#define SEB(n,b,value) (n) ^= ((-value)^(n)) & (_BV(b))

now that dosen work anymore, i know something was merged concerning SET_BIT a few days back that probably did this, but don't know how to fix it :/
Cant find the merge, do i member correctly that it was you that did something @Roxy-3D ? 😸 

the compiler error looks the same before and after my fix now:
`C:\Users\Familien\AppData\Local\Temp\arduino_build_411229\sketch\src\module\endstops.cpp:441:48: error: macro "SET_BIT" passed 3 arguments, but takes just 2

           if (X_MIN_TEST) UPDATE_ENDSTOP(X, MIN);

                                                ^

C:\Users\Familien\AppData\Local\Temp\arduino_build_411229\sketch\src\module\endstops.cpp:456:48: error: macro "SET_BIT" passed 3 arguments, but takes just 2

           if (X_MIN_TEST) UPDATE_ENDSTOP(X, MAX);

                                                ^

C:\Users\Familien\AppData\Local\Temp\arduino_build_411229\sketch\src\module\endstops.cpp:475:32: error: macro "SET_BIT" passed 3 arguments, but takes just 2

           UPDATE_ENDSTOP(Y, MIN);

                                ^

C:\Users\Familien\AppData\Local\Temp\arduino_build_411229\sketch\src\module\endstops.cpp:490:32: error: macro "SET_BIT" passed 3 arguments, but takes just 2

           UPDATE_ENDSTOP(Y, MAX);

                                ^

C:\Users\Familien\AppData\Local\Temp\arduino_build_411229\sketch\src\module\endstops.cpp:511:34: error: macro "SET_BIT" passed 3 arguments, but takes just 2

             UPDATE_ENDSTOP(Z, MIN);

                                  ^

C:\Users\Familien\AppData\Local\Temp\arduino_build_411229\sketch\src\module\endstops.cpp:519:38: error: macro "SET_BIT" passed 3 arguments, but takes just 2

           UPDATE_ENDSTOP(Z, MIN_PROBE);

                                      ^

C:\Users\Familien\AppData\Local\Temp\arduino_build_411229\sketch\src\module\endstops.cpp:538:32: error: macro "SET_BIT" passed 3 arguments, but takes just 2

           UPDATE_ENDSTOP(Z, MAX);

                                ^

C:\Users\Familien\AppData\Local\Temp\arduino_build_411229\sketch\src\module\endstops.cpp: In static member function 'static void Endstops::update()':

C:\Users\Familien\AppData\Local\Temp\arduino_build_411229\sketch\src\module\endstops.cpp:321:44: error: 'SET_BIT' was not declared in this scope

   #define UPDATE_ENDSTOP_BIT(AXIS, MINMAX) SET_BIT(current_endstop_bits, _ENDSTOP(AXIS, MINMAX), (READ(_ENDSTOP_PIN(AXIS, MINMAX)) != _ENDSTOP_INVERTING(AXIS, MINMAX)))

                                            ^

C:\Users\Familien\AppData\Local\Temp\arduino_build_411229\sketch\src\module\endstops.cpp:326:7: note: in expansion of macro 'UPDATE_ENDSTOP_BIT'

       UPDATE_ENDSTOP_BIT(AXIS, MINMAX); \

       ^~~~~~~~~~~~~~~~~~

C:\Users\Familien\AppData\Local\Temp\arduino_build_411229\sketch\src\module\endstops.cpp:441:27: note: in expansion of macro 'UPDATE_ENDSTOP'

           if (X_MIN_TEST) UPDATE_ENDSTOP(X, MIN);

[... 85 lines cut ...]

C:\Users\Familien\AppData\Local\Temp\arduino_build_411229\sketch\src\module\endstops.cpp:519:11: note: in expansion of macro 'UPDATE_ENDSTOP'

           UPDATE_ENDSTOP(Z, MIN_PROBE);

           ^~~~~~~~~~~~~~

C:\Users\Familien\AppData\Local\Temp\arduino_build_411229\sketch\src\module\endstops.cpp:321:44: error: 'SET_BIT' was not declared in this scope

   #define UPDATE_ENDSTOP_BIT(AXIS, MINMAX) SET_BIT(current_endstop_bits, _ENDSTOP(AXIS, MINMAX), (READ(_ENDSTOP_PIN(AXIS, MINMAX)) != _ENDSTOP_INVERTING(AXIS, MINMAX)))

                                            ^

C:\Users\Familien\AppData\Local\Temp\arduino_build_411229\sketch\src\module\endstops.cpp:326:7: note: in expansion of macro 'UPDATE_ENDSTOP_BIT'

       UPDATE_ENDSTOP_BIT(AXIS, MINMAX); \

       ^~~~~~~~~~~~~~~~~~

C:\Users\Familien\AppData\Local\Temp\arduino_build_411229\sketch\src\module\endstops.cpp:538:11: note: in expansion of macro 'UPDATE_ENDSTOP'

           UPDATE_ENDSTOP(Z, MAX);

           ^~~~~~~~~~~~~~
`

Roxy-3D, 2017-11-26:
> Cant find the merge, do i remember correctly that it was you that did something @Roxy-3D ? 

Yes...  And if we need to make changes to eleminate a name conflict...  Probably the names in bitmap_flags.cpp should change.    The bit map functions were pulled out of UBL to make them available to anything that wanted to pull in G26.   

If you just want to change the names in G26, G29 and bitmap_flags.cpp, that will resolve the conflict.   I can do that tomorrow if you want.   But if you do it now...  I'll merge it as soon as I see it show up.

Spawn32, 2017-11-26:
Probably best if you do it so i dont break something else, lol

If you could just mention the merge number here afterwards i can do the same in my fresh F7 adapted bugfix-2.0.x and multihost branches :)

Roxy-3D, 2017-11-26:
> Probably best if you do it so i dont break something else, lol

I'm waiting for Travis right now...  

https://github.com/MarlinFirmware/Marlin/pull/8565

When Travis says OK...   I'll merge.

Spawn32, 2017-11-26:
@Roxy-3D 
Applied your merge to my files and it didn't change anything, still the same error's. tested both in macros.h:
//#define SET_BIT(n,b,value) (n) ^= ((-value)^(n)) & (_BV(b))
#define SEB(n,b,value) (n) ^= ((-value)^(n)) & (_BV(b))

error's:`C:\Users\Familien\AppData\Local\Temp\arduino_build_879911\sketch\src\module\endstops.cpp:441:48: error: macro "SET_BIT" passed 3 arguments, but takes just 2

           if (X_MIN_TEST) UPDATE_ENDSTOP(X, MIN);

                                                ^

C:\Users\Familien\AppData\Local\Temp\arduino_build_879911\sketch\src\module\endstops.cpp:456:48: error: macro "SET_BIT" passed 3 arguments, but takes just 2

           if (X_MIN_TEST) UPDATE_ENDSTOP(X, MAX);

                                                ^

C:\Users\Familien\AppData\Local\Temp\arduino_build_879911\sketch\src\module\endstops.cpp:475:32: error: macro "SET_BIT" passed 3 arguments, but takes just 2

           UPDATE_ENDSTOP(Y, MIN);

                                ^

C:\Users\Familien\AppData\Local\Temp\arduino_build_879911\sketch\src\module\endstops.cpp:490:32: error: macro "SET_BIT" passed 3 arguments, but takes just 2

           UPDATE_ENDSTOP(Y, MAX);

                                ^

C:\Users\Familien\AppData\Local\Temp\arduino_build_879911\sketch\src\module\endstops.cpp:511:34: error: macro "SET_BIT" passed 3 arguments, but takes just 2

             UPDATE_ENDSTOP(Z, MIN);

                                  ^

C:\Users\Familien\AppData\Local\Temp\arduino_build_879911\sketch\src\module\endstops.cpp:519:38: error: macro "SET_BIT" passed 3 arguments, but takes just 2

           UPDATE_ENDSTOP(Z, MIN_PROBE);

                                      ^
`

Roxy-3D, 2017-11-26:
aaargh...   Yet another definition of SET_BIT() except this one is all capital letters.  Can you cut and paste your current error messages without your patch?

Spawn32, 2017-11-26:
removed it from macros and endstops, probaly would be more errors if i enabled bedleveling etc. but i want to keep it as minimal as possible until the board is fully working :)
If you ad SEB you will have the STM32GENERIC library cover'd.

Errors: `In file included from C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\cores\arduino/stm32_HAL/stm32XXxx_ll_spi.h:19:0,

                 from C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\libraries\SPI\src/SPI.h:7,

                 from C:\Users\Familien\Desktop\Marlin-bugfix-2.0.x\Marlin\Marlin.ino:52:

C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\system/STM32F7/HAL_Inc/stm32f7xx_ll_spi.h:379:33: error: macro "SET_BIT" requires 3 arguments, but only 2 given

   SET_BIT(SPIx->CR1, SPI_CR1_SPE);

                                 ^

C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\system/STM32F7/HAL_Inc/stm32f7xx_ll_spi.h:719:35: error: macro "SET_BIT" requires 3 arguments, but only 2 given

   SET_BIT(SPIx->CR1, SPI_CR1_CRCEN);

                                   ^

C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\system/STM32F7/HAL_Inc/stm32f7xx_ll_spi.h:783:37: error: macro "SET_BIT" requires 3 arguments, but only 2 given

   SET_BIT(SPIx->CR1, SPI_CR1_CRCNEXT);

                                     ^

C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\system/STM32F7/HAL_Inc/stm32f7xx_ll_spi.h:883:34: error: macro "SET_BIT" requires 3 arguments, but only 2 given

   SET_BIT(SPIx->CR2, SPI_CR2_NSSP);

                                  ^

C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\system/STM32F7/HAL_Inc/stm32f7xx_ll_spi.h:1108:35: error: macro "SET_BIT" requires 3 arguments, but only 2 given

   SET_BIT(SPIx->CR2, SPI_CR2_ERRIE);

                                   ^

C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\system/STM32F7/HAL_Inc/stm32f7xx_ll_spi.h:1119:36: error: macro "SET_BIT" requires 3 arguments, but only 2 given

   SET_BIT(SPIx->CR2, SPI_CR2_RXNEIE);

                                    ^

C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\system/STM32F7/HAL_Inc/stm32f7xx_ll_spi.h:1130:35: error: macro "SET_BIT" requires 3 arguments, but only 2 given

   SET_BIT(SPIx->CR2, SPI_CR2_TXEIE);

                                   ^

C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\system/STM32F7/HAL_Inc/stm32f7xx_ll_spi.h:1216:37: error: macro "SET_BIT" requires 3 arguments, but only 2 given

   SET_BIT(SPIx->CR2, SPI_CR2_RXDMAEN);

                                     ^

C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\system/STM32F7/HAL_Inc/stm32f7xx_ll_spi.h:1249:37: error: macro "SET_BIT" requires 3 arguments, but only 2 given

   SET_BIT(SPIx->CR2, SPI_CR2_TXDMAEN);

                                     ^

C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\system/STM32F7/HAL_Inc/stm32f7xx_ll_spi.h:1648:63: error: macro "SET_BIT" requires 3 arguments, but only 2 given

   SET_BIT(SPIx->I2SCFGR, SPI_I2SCFGR_I2SMOD | SPI_I2SCFGR_I2SE);

                                                               ^

C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\system/STM32F7/HAL_Inc/stm32f7xx_ll_spi.h:1717:39: error: macro "SET_BIT" requires 3 arguments, but only 2 given

   SET_BIT(SPIx->I2SCFGR, ClockPolarity);

                                       ^

C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\system/STM32F7/HAL_Inc/stm32f7xx_ll_spi.h:1857:39: error: macro "SET_BIT" requires 3 arguments, but only 2 given

   SET_BIT(SPIx->I2SPR, SPI_I2SPR_MCKOE);

                                       ^

C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\system/STM32F7/HAL_Inc/stm32f7xx_ll_spi.h:1891:45: error: macro "SET_BIT" requires 3 arguments, but only 2 given

   SET_BIT(SPIx->I2SCFGR, SPI_I2SCFGR_ASTRTEN);

                                             ^

Using library SPI at version 1.0 in folder: C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\libraries\SPI 
Using library stm32_dma at version 1.0 in folder: C:\Program Files (x86)\arduino190B\hardware\STM32GEN\STM32\libraries\stm32_dma 
exit status 1
Error compiling for board The Borg.`

Spawn32, 2017-11-26:
@tcm0116 
Finally got the second port working :)

#define SERIAL_PORT 0
#define SERIAL_PORT_2 5

and in my hal:

`#if !WITHIN(SERIAL_PORT, 0, 6)
  #error "SERIAL_PORT must be from 0 to 6"
#endif
#if SERIAL_PORT == 0
  #define MYSERIAL0 SerialUSB	// was SerialUSB // usb_serial
#elif SERIAL_PORT == 1
  #define MYSERIAL0 SerialUART1
#elif SERIAL_PORT == 2
  #define MYSERIAL0 SerialUART2
#elif SERIAL_PORT == 3
  #define MYSERIAL0 SerialUART3
#elif SERIAL_PORT == 4
  #define MYSERIAL0 SerialUART4
#elif SERIAL_PORT == 5
  #define MYSERIAL0 SerialUART5
#elif SERIAL_PORT == 6
  #define MYSERIAL0 SerialUART6
#endif

#ifdef SERIAL_PORT_2
  #if !WITHIN(SERIAL_PORT_2, 0, 6)
    #error "SERIAL_PORT_2 must be from 0 to 6"
  #elif SERIAL_PORT_2 == SERIAL_PORT
    #error "SERIAL_PORT_2 must be different than SERIAL_PORT"
  #endif
  #define NUM_SERIAL 2
  #if SERIAL_PORT_2 == 0
    #define MYSERIAL1 SerialUSB
  #elif SERIAL_PORT_2 == 1
    #define MYSERIAL1 SerialUART1
  #elif SERIAL_PORT_2 == 2
    #define MYSERIAL1 SerialUART2
  #elif SERIAL_PORT_2 == 3
    #define MYSERIAL1 SerialUART3
  #elif SERIAL_PORT_2 == 4
    #define MYSERIAL1 SerialUART4
  #elif SERIAL_PORT_2 == 5
    #define MYSERIAL1 SerialUART5
  #elif SERIAL_PORT_2 == 6
    #define MYSERIAL1 SerialUART6
  #endif
#else
  #define NUM_SERIAL 1
#endif`

was struggling for a long time, got nothing on the port, the i realized that all the uart's can be on 4 different places on this cpu, so had to pull a dirty one in the STM32GENERIC:

`const stm32_af_pin_list_type chip_af_usart_rx [] = {
//UART4
    { UART4 , GPIOA, GPIO_PIN_1  , GPIO_AF8_UART4 }, 
    { UART4 , GPIOA, GPIO_PIN_11 , GPIO_AF6_UART4 }, 
    { UART4 , GPIOC, GPIO_PIN_11 , GPIO_AF8_UART4 }, 
    { UART4 , GPIOD, GPIO_PIN_0  , GPIO_AF8_UART4 }, 
//UART5
    //{ UART5 , GPIOB, GPIO_PIN_5  , GPIO_AF1_UART5 }, 
    //{ UART5 , GPIOB, GPIO_PIN_8  , GPIO_AF7_UART5 }, 
    //{ UART5 , GPIOB, GPIO_PIN_12 , GPIO_AF8_UART5 }, 
    { UART5 , GPIOD, GPIO_PIN_2  , GPIO_AF8_UART5 }, 
//UART7
    { UART7 , GPIOA, GPIO_PIN_8  , GPIO_AF12_UART7}, 
    { UART7 , GPIOB, GPIO_PIN_3  , GPIO_AF12_UART7}, 
    { UART7 , GPIOE, GPIO_PIN_7  , GPIO_AF8_UART7 }, 
    { UART7 , GPIOF, GPIO_PIN_6  , GPIO_AF8_UART7 }, 
//UART8
    { UART8 , GPIOE, GPIO_PIN_0  , GPIO_AF8_UART8 }, 
//USART1
    { USART1, GPIOA, GPIO_PIN_10 , GPIO_AF7_USART1}, 
    { USART1, GPIOB, GPIO_PIN_7  , GPIO_AF7_USART1}, 
    { USART1, GPIOB, GPIO_PIN_15 , GPIO_AF4_USART1}, 
//USART2
    { USART2, GPIOA, GPIO_PIN_3  , GPIO_AF7_USART2}, 
    { USART2, GPIOD, GPIO_PIN_6  , GPIO_AF7_USART2}, 
//USART3
    { USART3, GPIOB, GPIO_PIN_11 , GPIO_AF7_USART3}, 
    { USART3, GPIOC, GPIO_PIN_11 , GPIO_AF7_USART3}, 
    { USART3, GPIOD, GPIO_PIN_9  , GPIO_AF7_USART3}, 
//USART6
    { USART6, GPIOC, GPIO_PIN_7  , GPIO_AF8_USART6}, 
    { USART6, GPIOG, GPIO_PIN_9  , GPIO_AF8_USART6}, 
}; 

const stm32_af_pin_list_type chip_af_usart_tx [] = {
//UART4
    { UART4 , GPIOA, GPIO_PIN_0  , GPIO_AF8_UART4 }, 
    { UART4 , GPIOA, GPIO_PIN_12 , GPIO_AF6_UART4 }, 
    { UART4 , GPIOC, GPIO_PIN_10 , GPIO_AF8_UART4 }, 
    { UART4 , GPIOD, GPIO_PIN_1  , GPIO_AF8_UART4 }, 
//UART5
    //{ UART5 , GPIOB, GPIO_PIN_6  , GPIO_AF1_UART5 }, 
    //{ UART5 , GPIOB, GPIO_PIN_9  , GPIO_AF7_UART5 }, 
    //{ UART5 , GPIOB, GPIO_PIN_13 , GPIO_AF8_UART5 }, 
    { UART5 , GPIOC, GPIO_PIN_12 , GPIO_AF8_UART5 }, 
//UART7
    { UART7 , GPIOA, GPIO_PIN_15 , GPIO_AF12_UART7}, 
    { UART7 , GPIOB, GPIO_PIN_4  , GPIO_AF12_UART7}, 
    { UART7 , GPIOE, GPIO_PIN_8  , GPIO_AF8_UART7 }, 
    { UART7 , GPIOF, GPIO_PIN_7  , GPIO_AF8_UART7 }, 
//UART8
    { UART8 , GPIOE, GPIO_PIN_1  , GPIO_AF8_UART8 }, 
//USART1
    { USART1, GPIOA, GPIO_PIN_9  , GPIO_AF7_USART1}, 
    { USART1, GPIOB, GPIO_PIN_6  , GPIO_AF7_USART1}, 
    { USART1, GPIOB, GPIO_PIN_14 , GPIO_AF4_USART1}, 
//USART2
    { USART2, GPIOA, GPIO_PIN_2  , GPIO_AF7_USART2}, 
    { USART2, GPIOD, GPIO_PIN_5  , GPIO_AF7_USART2}, 
//USART3
    { USART3, GPIOB, GPIO_PIN_10 , GPIO_AF7_USART3}, 
    { USART3, GPIOC, GPIO_PIN_10 , GPIO_AF7_USART3}, 
    { USART3, GPIOD, GPIO_PIN_8  , GPIO_AF7_USART3}, 
//USART6
    { USART6, GPIOC, GPIO_PIN_6  , GPIO_AF8_USART6}, 
    { USART6, GPIOG, GPIO_PIN_14 , GPIO_AF8_USART6}, 
}; `

Have no ider where or how to tell it to use the correct one, ant chance you could have a look ?
the fix i did is not a good solution :)

here is the link to the original STM32GENERIC if you could have a quick look:

https://github.com/danieleff/STM32GENERIC

Roxy-3D, 2017-11-26:
> removed it from macros and endstops, probaly would be more errors if i enabled bedleveling etc. but i want to keep it as minimal as possible until the board is fully working :)

Yeah...   I don't really like the Marlin definition of SET_BIT().   It is overly complex.   For sure, other libraries are going to use identifiers like set_bit().   And our SET_BIT is only used in one place.   My vote would be we get rid of the macro definition.

Spawn32, 2017-11-26:
@tcm0116 
So back to the debug output from TMC26XStepperSTM32.cpp.
If i enable debug and change the serial print to:
`#ifdef DEBUG	
	MYSERIAL1.println("TMC26X stepper library");
	MYSERIAL1.print("CS pin: ");
	MYSERIAL1.println(cs_pin);
	MYSERIAL1.print("DIR pin: ");
	MYSERIAL1.println(dir_pin);
	MYSERIAL1.print("STEP pin: ");
	MYSERIAL1.println(step_pin);
	MYSERIAL1.print("current scaling: ");
	MYSERIAL1.println(current_scaling,DEC);`

it compiles without error's but during restet / boot i get no usb port anymore and no response if i try to send anything to UART5, is there something more i need to do or include in TMC26XStepperSTM32.cpp ?

Feeling puzzled 😄

tcm0116, 2017-11-26:
@Spawn32 I'm not really familiar with that HAL. What file did you make
those modifications to?

You might consider making the USB Serial port be -1 to match the other
HALs.

Instead of using MYSERIAL1 directly, use the wrapper macros in serial.h,
such as SERIAL_PRINTF.

Spawn32, 2017-11-26:
For the STM32F765ZG it uses STM32GEN\STM32\system\STM32F7\stm32_chip\stm32_USART_variant_8.h

Yes will change it to -1 when things are working and i do the clean up of the files :)

Spawn32, 2017-11-26:
@tcm0116 
This is how the debug stuff looks now, stil stops or crashes the usb / cpu, no usb after reset / boot 😞 
as soon as i do //#define TMC_DEBUG everything is back to normal..

[TMC26XStepperSTM32.zip](https://github.com/MarlinFirmware/Marlin/files/1504179/TMC26XStepperSTM32.zip)

Spawn32, 2017-11-26:
added this at the beginning but ended up with errors:

#ifdef TMC_DEBUG	
  #if NUM_SERIAL > 0
    MYSERIAL0.begin(BAUDRATE);
    #if NUM_SERIAL > 1
      MYSERIAL1.begin(BAUDRATE);
    #endif
    SERIAL_PRINTF("TMC26x Debug running:");
    SERIAL_FLUSHTX();
  #endif
#endif

`In file included from C:\Users\Familien\AppData\Local\Temp\arduino_build_812354\sketch\src\HAL\HAL_STM32F7\TMC26XStepper\TMC26XStepperSTM32.cpp:36:0:

c:\users\familien\appdata\local\temp\arduino_build_812354\sketch\src\hal\hal_stm32f7\hal_stm32f7.h:61:21: error: 'SerialUSB' does not name a type

   #define MYSERIAL0 SerialUSB // was SerialUSB // usb_serial

                     ^

C:\Users\Familien\AppData\Local\Temp\arduino_build_812354\sketch\src\HAL\HAL_STM32F7\TMC26XStepper\TMC26XStepperSTM32.cpp:125:5: note: in expansion of macro 'MYSERIAL0'

     MYSERIAL0.begin(BAUDRATE);

     ^~~~~~~~~

c:\users\familien\appdata\local\temp\arduino_build_812354\sketch\src\hal\hal_stm32f7\hal_stm32f7.h:94:23: error: 'SerialUART5' does not name a type

     #define MYSERIAL1 SerialUART5

                       ^

C:\Users\Familien\AppData\Local\Temp\arduino_build_812354\sketch\src\HAL\HAL_STM32F7\TMC26XStepper\TMC26XStepperSTM32.cpp:127:7: note: in expansion of macro 'MYSERIAL1'

       MYSERIAL1.begin(BAUDRATE);

       ^~~~~~~~~

c:\users\familien\appdata\local\temp\arduino_build_812354\sketch\src\hal\hal_stm32f7\hal_stm32f7.h:61:21: error: 'SerialUSB' does not name a type

   #define MYSERIAL0 SerialUSB // was SerialUSB // usb_serial

                     ^

c:\users\familien\appdata\local\temp\arduino_build_812354\sketch\src\core\serial.h:158:39: note: in expansion of macro 'MYSERIAL0'

   #define SERIAL_PRINTF(args...)      MYSERIAL0.printf(args)

                                       ^~~~~~~~~

C:\Users\Familien\AppData\Local\Temp\arduino_build_812354\sketch\src\HAL\HAL_STM32F7\TMC26XStepper\TMC26XStepperSTM32.cpp:129:5: note: in expansion of macro 'SERIAL_PRINTF'

     SERIAL_PRINTF("TMC26x Debug running:");

     ^~~~~~~~~~~~~`

tcm0116, 2017-11-26:
Your hal_stm32f7.h file needs to #include whatever file has USBSerial
declared in it.

Spawn32, 2017-11-26:
USBSerial is declared inside STM32GENERIC version of arduino.h and automatically included, the hal works fine with multihost, if i dont turn on TMC_DEBUG i have perfectly working hosts from simplyfy3d and my uart5 at the same time, trouble starts when i want to use serial from TMC26XStepperSTM32.cpp, it's like serial magically just works inside Marlins core functions / files, LOL

tcm0116, 2017-11-26:
@Spawn32 I'll see if I can find some time to look at it. Would it be
possible for you to push your entire working folder to a branch in your
fork of Marlin on GitHub so I can look at exactly what you're working with?
I can then submit a PR entity any changes I might have.

By the way, how much did it cost to build one of your boards? It might be
fun to have one to play around with, if it's not too expensive to get one
to me in the U.S.

Spawn32, 2017-11-27:
OMG, worst hour of my life, been without internet, never happen'd before, just got i back LOL

I can upload my Marlin branch on my github, reason i haven't before is that it's full of stupid test remarks, guess it's not that important, will upload it later and tell you.

But you will also need the STM32GENERIC with my board and mods, that's the problem, it's 4700 files and 76mb, github dont allow more than 100 files and 10mb, github is full of projects bigger than that, but don't know how people get it uploaded :/

you must also install the latest 2017-q2-update/bin/arm-none-eabi and change the tool path to the new one, the internal in arduino only goes up to a STM32F4 :)

I think i mentioned before that if i get help i will  of course supply a card for free, problem is that they are to hard to hand build, looks like i will be getting most of the parts for my pick & place machine tomorrow so i can start building that and get some more cards build :)

Spawn32, 2017-11-27:
Hmm... now i have the same problem with Marlin, can upload it to github, to many files and to big, is this a a restriction on the free account ?
Until i can solve this i will zip marlin and STM32GENERIC and past the links to my onedrive here :)

Spawn32, 2017-11-27:
Borg Marlin MultiHost:  https://1drv.ms/u/s!Asf9doI7WS5djekLOfiqPaK9rW089Q
STM32GEN (goes into you hardware folder on arduino) :   https://1drv.ms/u/s!Asf9doI7WS5djekM16bRBsV0DJBFfw

after that you should have The_Borge in the boards section of arduino :)
this one compiles and multihost works, but locks up steppers and no TMC_DEBUG...

tcm0116, 2017-11-27:
Any reason you're not using the official stm32duino STM32 HAL implementation? It looks like it can directly install into the Arduino IDE:

https://github.com/stm32duino/wiki/wiki/Boards-Manager

Spawn32, 2017-11-27:
I think it's harder to add your own board,it has some development boards in it that's all i think, i actually tried to install it now to have a look at it, following the guide i cant even find it...

I started out my design looking at Chris Barr's picoprint , but after that i decided on another faste MCU and adding a lot of options, changing the powe supply, etc.. 
But he already had a F4 hal with the TCM26x driver almost finished but in a very old release of the 32-Bit-RCBugFix branch.

Her is the official forum thread for STM32Generic, has a lot of more posts than the other one:
http://stm32duino.com/viewforum.php?f=51

Bob-the-Kuhn, 2017-11-27:
Would a ST Nucleo F7xx board be useful for code development & testing?  Just wondering if there was a quick/easy way for others to help.

[20 more replies]

closed 2018-02-11 by thinkyhead