MarlinFirmware/Marlin #8259: X-axis / X-bed size limit question

playlet, 2017-11-05:
Hi guys, I have a question regarding the bed leveling and bed placement setup.

I've put together a modified prusa i3 printer which has stretched X axis (original bed size), stretched only at the end (not symmetrical).
I'm trying to setup a proximity sensor that would be able to probe entire bed in auto mode (nozzle has to go beyond bed size on x axis, and can since it's stretched on far end).

X axis can go 235mm total, bed ends at 197mm.
Now, how do I limit printing area to 0-197mm and still have 197-235mm for probing only?

Is it enough to only set 197mm as bed X size in slic3r or do I need to do something else/more?

Thanks and best regards,

Ivan

bruce356, 2017-11-05:
Good Question does not work for me in version 1.1.5 either, limited by the bed size and the sensor can not reach all of the bed for probing. Sanity check stops me doing it also.
SO I loaded Repetier firmware that way I can probe all of my bed 250x320  no problem, probe offset is 35mm

JimBrown, 2017-11-05:
I had a similar situation, except on the Y axis. To allow the probe to reach the entire bed, I changed Y_MAX_POS to be the size of the bed plus the extra distance it could move (my probe is in front of the nozzle). I left Y_BED_SIZE set to the actual size of the bed. I then had to play around with the various probe points to get it to work. If your probe is left of the nozzle, I'm sure you could do the same with X_MAX_POS. If it is to the right of the nozzle, then setting X_MIN_POS to a negative number should do it. Oh, and leave the bed size set to the actual bed size in your slicer. Any host program can still move the nozzle outside the bed, but the slicer should not try to print anything outside the bed. But don't quote me on that. ;-)

playlet, 2017-11-05:
Damn, I totally missed that x_max_pos and x_bed_size are not the same thing. Thanks JimBrown!!!! :)

closed 2017-11-05 by playlet