MarlinFirmware/Marlin #8263: BABY_STEP ? in 2.0

cdedwards, 2017-11-05:
Is babystepping working in 2.0? I have changed the value of increase to 100 in an attempt to see and when I attempted to use it, nothing really seemed to be working. neither up or down

Tannoo, 2017-11-05:
Configs?

Roxy-3D, 2017-11-05:
There was a problem where the Baby Step multiplier got set to 100.  It is back at a more reasonable number of 1 right now.    For my machines, 3 or 4 works well.

```cpp
  #define BABYSTEP_MULTIPLICATOR 1 // Babysteps are very small. Increase for faster motion.
```

cdedwards, 2017-11-06:
I have mine set at 10. doesn't seem to move at all though.

Tannoo, 2017-11-06:
I cannot open that zip file.

Try disabling `BABYSTEP_Z_PROBE_OFFSET` in `Configuration_adv.h`.

cdedwards, 2017-11-06:
it's a renamed rar file. sorry

cdedwards, 2017-11-06:
from Marlin\src\HAL\HAL_AVR\watchdog_AVR.cpp:25:
Marlin\src\HAL\HAL_AVR\../../inc/SanityCheck.h:365:6: error: #error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a BABYSTEP_ZPROBE_OFFSET."
#error "BABYSTEP_ZPROBE_GFX_OVERLAY requires a BABYSTEP_ZPROBE_OFFSET."

Roxy-3D, 2017-11-06:
Aaaauurrgghhhh....   

Please turn off the BABYSTEP_ZPROBE_GFX_OVERLAY  until we get that problem cleared up.

cdedwards, 2017-11-06:
yep. problem disappeared once GFX_OVERLAY was disabled.

closed 2017-11-17 by cdedwards