MarlinFirmware/Marlin #8264: Switching extruder seems br0ken after Native Machine Coordinates

alexxy, 2017-11-05:
Hi!

Seems like switching extruder is br0ken after recent changes in bugfix-2.0.x branch. Possibly because of Native machine coordinates patch

GMagician, 2017-11-05:
I installed new 2.0.x and I noticed something really wrong. My cnf is bl-touch and dondolo  and it seems that commands for dondolo servo are redirected to bltouch. I'm analizyng code to understand why.
At the moment nothing strange with code, but I', novice to firmware so don't expect anything

alexxy, 2017-11-05:
@GMagician  can you share you're bltouch holder for dondolo? =) my bltouch is comming next week.

If i remove Native machine coordinates patch from tree, dondolo works again... Seems like something wrong with servo and e1 e2 directions...

GMagician, 2017-11-05:
Of course I can, here it is
[BLTouch-support 2.zip](https://github.com/MarlinFirmware/Marlin/files/1444241/BLTouch-support.2.zip)

note that my printer is not fully working. I've done some homing and it works. Not tested yet repeatibility
You need a M3 square nut and a M3 x18 screw

GMagician, 2017-11-05:
Now I had fixed a issue with Italian accents letters and I stopped to check this issues.
I noticed that setting bltouch as servo 1 and dondolo as servo 0 it works perfectly.
Theoretically native system coordinates as nothing to do whit servos. I know for sure that correct function is called (MOVE_SERVO) with proper index (I wrote 1 instead of define name)...still investigating

GMagician, 2017-11-05:
Found. It was my fault. I continuosly update my configuration to current bugfix version and I forgot to enable NUM_SERVOS with a value of 2 and to set SERVO_DELAY to 500 for extruder.
From LCD I can normally change extruder and it seems to work.
What is your issue?

thinkyhead, 2017-11-08:
@alexxy Did your test remove _only_ the native workspace patch or did it revert to _before_ the native workspace patch?

alexxy, 2017-11-09:
in that case i test by removing only native workspace patch. I'll test again with latest bugfix-2.0.x  and report back.

thinkyhead, 2017-11-11:
Thanks for continuing to report such issue! Be sure to check the latest `bugfix-1.1.x` on a daily basis, as we continue to debug and patch for the native workspace overhaul. Sooner or later we're bound to have all the kinks worked out.

closed 2017-11-12 by thinkyhead