MarlinFirmware/Marlin #8271: Reboot on print start when printing from sd card

Zwaubel, 2017-11-05:
I'm currently having lot's of reboots when printing from sd card. When choosing a file to print, my printer does the bed leveling routine and a bit of preparation gcode (nozzle priming etc.) as it is supposed to but then, when the actual print should start, marlin crashes and reboots. 
This problem seems to be depending mostly on two things:
1. The size of the model: Big models tend to be causing reboots more frequently then smaller ones
2. The slicer: With Cura 15.04 only few gcode files cause the firmware to reboot, how ever with Cura 3.0.4 almost every gcode file causes marlin to reboot.

Either way, the firmware currently behaves very unstable for me, when printing from the sd card.
I'd appreciate any help!

Hardware details: 
Custom cartesian printer wiht RAMPS + Arduino Mega, RRD full graphics display and drv8825 drivers. 

Configuration files here:
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1444188/Config.zip)

Zwaubel, 2017-11-06:
Just checked Marlin version bugfix-1.1.x with similiar results: crash and reboot after the bed leveling routine has finished and the actual print should start. Config files here:
[Config_bugfix_1_1_x.zip](https://github.com/MarlinFirmware/Marlin/files/1444585/Config_bugfix_1_1_x.zip)

It seems to me as if something goes badly wrong causing the watchdog timer to kick in and resetting the controller.

Zwaubel, 2017-11-06:
MarlinKimbra and Repetier FW are both working fine (just tested) with my printer setup... so I guess, it must be a bad configuration or a bug.

thinkyhead, 2017-11-08:
The usual reason for a watchdog reset is trying to move an axis faster than the firmware can handle, but it could also be due to a buffer overflow. See if lowering your max speeds makes any difference.

Are you seeing this crash also with Marlin 1.1.6?

Zwaubel, 2017-11-08:
Interesting, I lowered the print speed to 40 mm/s and merged the last bugfix-2.0.x commit 004fc55eaf5c3dd8ac1278d17cd29e29453867ba and now the printer isn't even moving. It's doing something, since the fan control does seem to work (it behaves as if the printer would be printing) but the print head stays at the last probe point and no filament is beeing extruded.

Marlin 1.1.6 seems to work just fine, even with normal and high speeds.

Side note to Marlin 1.1.6: 
The delay functions of the neo pixel self test on start up should be switched to safe_delay functions, otherwise the printer will end up in a boot loop because the watchdog is constantly resetting the controller. 
And I think the delay it self could be way shorter (I set it to 300ms per color) since this is just a quick test and shouldn't stop the printer from starting up. Just my personal opinion though.

Zwaubel, 2017-11-13:
Well, Cura 3.0.4 still seems to be incompatible with Marlin 1.1.6. 
Version 1.1.6 is running very stable but objects printed of a gcode file that does come from Cura 3.0.4 do have a slight z height problem. It seems like the z coordinates are slightly scaled down, so that those objects end up being printed a bit to flat. 
However models sliced with Cura 15.04 print just fine.
I double checked my z steps per unit to be correct. The problem has to be somewhere else...

thinkyhead, 2018-02-09:
Presuming solved.

closed 2018-02-09 by thinkyhead