MarlinFirmware/Marlin #8276: 2.0.x Archim2 Compile error

Allted, 2017-11-05:
I am not sure where the best place is for this information. I only know enough to find an error but not enough, yet, to fix it or I would submit a pull request.

When using the Archim2 and setting the configuration.h to
 `#define SERIAL_PORT -1`
This knocks out the marlin serial_due.cpp
`// Based on selected port, use the proper configuration
#if SERIAL_PORT == 0
  #define HWUART UART
  #define HWUART_IRQ UART_IRQn
  #define HWUART_IRQ_ID ID_UART
#elif SERIAL_PORT == 1
  #define HWUART USART0
  #define HWUART_IRQ USART0_IRQn
  #define HWUART_IRQ_ID ID_USART0
#elif SERIAL_PORT == 2
  #define HWUART USART1
  #define HWUART_IRQ USART1_IRQn
  #define HWUART_IRQ_ID ID_USART1
#elif SERIAL_PORT == 3
  #define HWUART USART3
  #define HWUART_IRQ USART3_IRQn
  #define HWUART_IRQ_ID ID_USART3
#endif`

I understand a little of what is going on but I am not sure how a -1 would play with the uarts.

The next issue is the Archim2_pins needs to get the SDSS moved out of the LCD conditional or it won't compile unless an LCD is selected.

I am willing to help and test but this doesn't really seem like the best place, or is it, or should I just give it some time??

teemuatlut, 2017-11-05:
I'm linking here the previous places these have been mentioned. In short, changes are needed to the Due HAL and pin defs.
[HWUART issue mentioned earlier](https://github.com/MarlinFirmware/Marlin/pull/8190#issuecomment-340888073)
[SDSS issue mentioned earlier](https://github.com/MarlinFirmware/Marlin/pull/8190#discussion_r148879097)

Allted, 2017-11-05:
Okay, I'll keep poking around to see if I can get it closer.

closed 2017-11-05 by Allted